volkszaehler/mbmd #25: unable to make build

Markus2801A, 2019-08-24:
Console output (last ones):
# github.com/volkszaehler/mbmd/cmd
cmd/root.go:82:16: undefined: os.UserHomeDir
make: *** [Makefile:31: build] Error 2

Via:
`sudo git clone https://github.com/volkszaehler/mbmd.git
cd mbmd
sudo make install
sudo make build`

I was able to build latest version of mbmd on Raspberry Pi 4B yesterday.
Since changing root.go, I received the above errors!
[image]

andig, 2019-08-24:
Please upgrade to go 1.12 

Viele Grüße,
Andreas

andig, 2019-08-25:
Updated docs ed985ea83792959567ea5808993c45bd1ab5ebd5

closed 2019-08-25 by andig

Markus2801A, 2019-08-25:
Hello, sorry but unable to upgrade to go 1.12.9 on raspberry pi 4B!
there seems to be issues with this go version.
I`m not able to install it (go1.12.9.linux-arm64.tar.gz) regardless if I choose the right ARMv8 version or I even tried ARMv6l version of newest go 1.12.9 packages.

andig, 2019-08-25:
Try the armv6 version, its working fine for me on Pi3:

````
Linux uv 4.14.58-v7+ #1130 SMP Thu Jul 26 14:30:50 BST 2018 armv7l
Last login: Sat Aug 17 12:00:02 2019 from 192.168.0.22
pi@uv:~$ go version
go version go1.12.7 linux/arm
````

Markus2801A, 2019-08-25:
Thanks for your help, but I already tried, nothing is working? Maybe because Go has issues with on ARMv8 of raspberry 4B?
Installed AMRv6:

> pi@raspberrypi:/usr/local/go/bin $ ./go version
> go version go1.12.9 linux/arm
after that:

> pi@raspberrypi:~/mbmd $ sudo make install
> go install golang.org/x/tools/cmd/stringer
> make: go: Command not found
> make: *** [Makefile:15: install] Error 127

andig, 2019-08-25:
> pi@raspberrypi:/usr/local/go/bin $ ./go version

and
 
> make: go: Command not found

Just add your go/bin folder to the path!

Markus2801A, 2019-08-25:
in progress: but dunno if it's correct?
pi@raspberrypi:/usr/local/go/bin $ export PATH=$PATH:/usr/local/go/bin/

after make install und make build last three rows:

github.com/volkszaehler/mbmd
go build github.com/volkszaehler/mbmd: open mbmd: permission denied
make: *** [Makefile:31: build] Error 1

andig, 2019-08-25:
Dunno. Did you checkout as sudo maybe?

Markus2801A, 2019-08-25:
sudo make install or sudo make build gives immediate error message.
without "sudo" process starts but than I get error after make build:
```
go build github.com/volkszaehler/mbmd: open mbmd: permission denied
make: *** [Makefile:31: build] Error 1
```

andig, 2019-08-25:
You have a permission problem which has nothing to do with mbmd most likely.

Markus2801A, 2019-08-25:
interesting, I always start from scratch
Write Image with galena etcher to sd card
connection vis SSH and setting new pwd
then
sudo apt update && sudo apt upgrade -y
after those initial steps I tried installing golang, and followed your suggestions
so system should be "clean" without any older golang installs or other "faulty" dependencies etc.
it does not matter what I try, ARMv6 or ARMv8 even tried -> https://golang.org/doc/install/source
always same error.
But why permission problem?

2days ago everything worked, it was a coincidence that I wanted to setup the whole system because Im new and played with some other packages on RPi but wanted to make new clean system for monitoring SDM630.

Markus2801A, 2019-08-26:
chmod 777 mbmd resolved the issue
know 777 is not ideal set it back to 775 after making build
Thanks for your help!
Keep up your great work on this tool!