volkszaehler/mbmd #33: Topic ID in Homie MQTT entspricht nicht der Convention

chrostek, 2019-09-30:
Ich habe versucht MBMD mit Homie MQTT in Verbindung mit openHAB 2 zu nutzen. Leider meldet mir openHAB einen Fehler:

ID segment 'mbmd_sdm1.1_meter' contains invalid characters. Each segment of the ID must match the pattern [A-Za-z0-9_-]*.

Das ist auch korrekt so, laut Homie Convention:
https://homieiot.github.io/specification/#topic-ids
darf da kein Punkt drin sein. Das Problem liegt also bei "sdm1.1".
Meine Config sieht so aus:

```
# mqtt config
mqtt:
  broker: localhost:1883
  topic:
  user: 
  password: 
  clientid: mbmd
  qos: 0
  clean: true
  homie: mbmd

# adapters are referenced by device
adapters:
- device: /dev/ttyUSB0
  baudrate: 9600
  comset: 8N1

# list of devices
devices:
- type: sdm
  id: 1
  adapter: /dev/ttyUSB0
- type: sdm
  id: 2
  adapter: /dev/ttyUSB0
- type: sdm
  id: 3
  adapter: /dev/ttyUSB0
```

Ich habe also aktuell 3 SDM Geräte, die entsprechend durchnumeriert werden.
Im einfachsten Falle müsste für Homie einfach ein "." durch ein "-" ersetzt werden. 

Danke

Viele Grüße
Sebastian

andig, 2019-09-30:
Siehe https://github.com/volkszaehler/mbmd/pull/34. Leider hab ich gerade Probleme mit dem Linter so dass ich es aktuell nicht mergen kann.

chrostek, 2019-09-30:
Danke. Bringt mich leider nur einen Schritt weiter, jetzt meckert openHAB 2 das hier an:

> Did not receive mandatory topic value: mbmd/sdm1-1/meter/$properties

Kann es sein, dass es $properties in jeder Node geben muss?

In den Logs (Verbose) sieht das $properties publishing sowieso etwas seltsam aus:

> mqtt: publish mbmd/sdm1-1/$properties, message: voltagel1
> mqtt: publish mbmd/sdm1-1/$properties, message: voltagel1,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: importl3,voltagel1,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: exportl3,importl3,voltagel1,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: exportl3,frequency,importl3,voltagel1,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: currentl1,exportl3,frequency,importl3,voltagel1,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: currentl1,exportl3,frequency,importl1,importl3,voltagel1,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: currentl1,exportl1,exportl3,frequency,importl1,importl3,voltagel1,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: currentl1,exportl1,exportl2,exportl3,frequency,importl1,importl3,voltagel1,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: currentl1,export,exportl1,exportl2,exportl3,frequency,importl1,importl3,voltagel1,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil2,currentl1,export,exportl1,exportl2,exportl3,frequency,importl1,importl3,voltagel1,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil2,currentl1,export,exportl1,exportl2,exportl3,frequency,importl1,importl3,thdl1,voltagel1,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil2,currentl1,export,exportl1,exportl2,exportl3,frequency,importl1,importl3,thdl1,voltagel1,voltagel2,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil2,currentl1,currentl2,export,exportl1,exportl2,exportl3,frequency,importl1,importl3,thdl1,voltagel1,voltagel2,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil2,currentl1,currentl2,export,exportl1,exportl2,exportl3,frequency,importl1,importl3,powerl1,thdl1,voltagel1,voltagel2,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil2,currentl1,currentl2,export,exportl1,exportl2,exportl3,frequency,importl1,importl3,powerl1,powerl3,thdl1,voltagel1,voltagel2,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil2,currentl1,currentl2,export,exportl1,exportl2,exportl3,frequency,importl1,importl2,importl3,powerl1,powerl3,thdl1,voltagel1,voltagel2,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil2,currentl1,currentl2,export,exportl1,exportl2,exportl3,frequency,import,importl1,importl2,importl3,powerl1,powerl3,thdl1,voltagel1,voltagel2,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil2,cosphil3,currentl1,currentl2,export,exportl1,exportl2,exportl3,frequency,import,importl1,importl2,importl3,powerl1,powerl3,thdl1,voltagel1,voltagel2,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil2,cosphil3,currentl1,currentl2,currentl3,export,exportl1,exportl2,exportl3,frequency,import,importl1,importl2,importl3,powerl1,powerl3,thdl1,voltagel1,voltagel2,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil2,cosphil3,currentl1,currentl2,currentl3,export,exportl1,exportl2,exportl3,frequency,import,importl1,importl2,importl3,powerl1,powerl2,powerl3,thdl1,voltagel1,voltagel2,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil1,cosphil2,cosphil3,currentl1,currentl2,currentl3,export,exportl1,exportl2,exportl3,frequency,import,importl1,importl2,importl3,powerl1,powerl2,powerl3,thdl1,voltagel1,voltagel2,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil1,cosphil2,cosphil3,currentl1,currentl2,currentl3,export,exportl1,exportl2,exportl3,frequency,import,importl1,importl2,importl3,powerl1,powerl2,powerl3,thdl1,thdl2,voltagel1,voltagel2,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil1,cosphil2,cosphil3,currentl1,currentl2,currentl3,export,exportl1,exportl2,exportl3,frequency,import,importl1,importl2,importl3,powerl1,powerl2,powerl3,thdl1,thdl2,thdl3,voltagel1,voltagel2,voltagel3
> mqtt: publish mbmd/sdm1-1/$properties, message: cosphil1,cosphil2,cosphil3,currentl1,currentl2,currentl3,export,exportl1,exportl2,exportl3,frequency,import,importl1,importl2,importl3,powerl1,powerl2,powerl3,thd,thdl1,thdl2,thdl3,voltagel1,voltagel2,voltagel3
> 

Er scheint die Liste der Properties in einer Schleife zu ergänzen, scheint aber jeden Schleifendurchlauf dabei zu publishen ... ist das beabsichtigt?

andig, 2019-09-30:
> Er scheint die Liste der Properties in einer Schleife zu ergänzen, scheint aber jeden Schleifendurchlauf dabei zu publishen ... ist das beabsichtigt?

Ja- das erspart vorab zu wissen welche der Properties beim Zähler wirklich existieren. Mir schien das normkonform zu sein?

> Kann es sein, dass es $properties in jeder Node geben muss?

Wahrscheinlich ja, schaue ich mir an. Um das Pingpong abzukürzen: kannst Du mir einen Tip geben wir ich das selber in OpenHAB testen kann?

chrostek, 2019-09-30:
Normkonform ist das bestimmt, erscheint mir nur etwas unperformant ;-)

In openHAB (erst ab 2.4 möglich!):

MQTT Binding installieren
Inbox -> "+" drücken -> "MQTT Binding" -> "manually add thing" -> "MQTT Broker"
-> den entsprechend konfigurieren

Dann das Spiel nochmal:
Inbox -> "+" drücken -> "MQTT Binding" -> "manually add thing" -> "Homie MQTT device"
-> als Bridge den eben angelegten Broker auswählen, Device ID: "sdm1-1" in meinem Fall und "MQTT base prefix" "mbmd" (bei default Config "homie")

dabei /var/log/openhab2/openhab.log beobachten.
Wir können aber auch Pingpong machen, ich teste gerne :-)

andig, 2019-09-30:
`$properties` ist auch behoben, jetzt hänge ich an:

    Status: OFFLINE - COMMUNICATION_ERROR java.lang.NullPointerException

chrostek, 2019-09-30:
Ich habe jetzt:

> Did not receive mandatory topic value: mbmd/sdm1-1//$name

:-)

andig, 2019-09-30:
Sieht aus wie dieses Problem: https://github.com/homieiot/convention/issues/183. Ich verstehe die Ursache nicht.

andig, 2019-09-30:
> 
      Did not receive mandatory topic value: mbmd/sdm1-1//$name

Das ist ziemlich seltsam. Eigentlich müsste das sein:

    mbmd/sdm1-1/meter/$name

Es scheint als würden das `$nodes=meter` Property ignoriert werden? Blöderweise sehe ich den Fehler bei mir nicht :/

chrostek, 2019-09-30:
achje ... manchmal regt mich openhab schon auf ... Dienst neu gestartet -> Fehler weg ... sorry!
Aktuell scheint alles zu passen, auch wenn der Validation Error, den du verlinkt hast für mich auch nicht nachvollziehbar ist. Aber bei mir erkennt er jetzt wunderbar die Channels in openHAB. Vielen Dank!

andig, 2019-09-30:
> achje ... manchmal regt mich openhab schon auf ... Dienst neu gestartet -> Fehler weg ... sorry!

Gar kein Problem. Ich bin mir grad nicht sicher wer Schuld ist. Aktuell schiebe ich "NULL" in die retained topics um den device tree einmal zu löschen (statt einfach `$state=disconnected` zu setzen). Das mag OpenHAB nicht- diverse Exceptions im Log.

Wenn ich jetzt nur den state setze würde ich ja trotzdem gerne die retained topics abräumen. Das wiederum scheint aber die Exceptions anzuheizen. Mir fehlt gerade eine richtig gute Idee wie ich alle die `node` und `property` topics abräumen kann, die ein Device nach Neustart vielleicht nicht mehr hat?

chrostek, 2019-09-30:
Wieso sollte ein Device die nicht mehr haben? Wer wechselt denn so oft Devices, die vom selben Hersteller aber anderen Modell sind, dass das wahrscheinlich sein könnte?
Selbst wenn dem so wäre, würde über die neuen node und property Werte der Verweis ja nicht mehr existieren - die würden dann verwaisen - nicht schön, aber auch unschädlich.

openHAB läuft bei mir auf einem anderen Server als MBMD - wenn der Raspberry mit MDMD mal neu startet, openHAB aber nicht, will ich ja eigentlich nicht, dass die Werte geNULLt werden. Da soll der letzte Wert stehen bleiben, bis wieder aktuellere kommen. Das bringt sonst ggf. auch Graphen (Influx, etc) durcheinander, wenn da NULL Werte rein laufen und openHAB den Wert verliert...

Alternativ ne neue Config Option einführen, mit der man den Devices Namen für MQTT (und die WebUI?) geben kann. Das macht es unwahrscheinlicher, dass ein Nutzer den Zähler wechselt und ihn gleich nennt :-)
Und man könnte die Zähler besser auseinander halten (eher ein Feature Request als eine Lösung deines Problems ;-) )

chrostek, 2019-09-30:
Noch eine Frage, auch wenns nur bedingt hier hin gehört:

In der README steht:

> MQTT API
> Another option for receiving client updates is by using the built-in MQTT publisher. By default, readings are published at /mbmd/<unique id>/<reading>. Rate limiting is possible.

> Rate limiting is possible.

Übersehe ich da was? Ich habe keine Option gefunden, die das steuern würde. Aktuell wird doch jeder Wert direkt ins MQTT geschrieben, oder?

andig, 2019-09-30:
> Wieso sollte ein Device die nicht mehr haben? Wer wechselt denn so oft Devices, die vom selben Hersteller aber anderen Modell sind, dass das wahrscheinlich sein könnte?

Vielleicht nicht standard aber möglich. Z.B. wenn ein neues Device unter altem Namen auftaucht- sobald Namen selbst vergeben werden können wird das passieren. 

> openHAB läuft bei mir auf einem anderen Server als MBMD - wenn der Raspberry mit MDMD mal neu startet, openHAB aber nicht, will ich ja eigentlich nicht, dass die Werte geNULLt werden. Da soll der letzte Wert stehen bleiben, bis wieder aktuellere kommen.

Genauso isses jetzt. Genullt wird trotzdem- immer dann wenn ein Property oder Device nicht existiert. Bei Neustart kann das also kurz passieren, dann kommen sie ja aber- falls die Dinger existieren auch sofort wieder.

Ich hab den PR gerade rundum überholt, jetzt sehe ich gar keine Exceptions mehr im OpenHAB Log- schaust Du auch nochmal?

> Übersehe ich da was? Ich habe keine Option gefunden, die das steuern würde. Aktuell wird doch jeder Wert direkt ins MQTT geschrieben, oder?

Steht noch im Todo nach dem Rewrite, siehe https://github.com/volkszaehler/mbmd/issues/22

andig, 2019-09-30:
... jetzt fehlt nur noch auch für mehrere Devices des Last Will zu setzen. Das erfordert leider separate MQTT Connections. Würde ich aber- falls Du soviel Geduld hast- noch in den PR mit einbauen.

chrostek, 2019-09-30:
hm, ich hab jetzt Fehler im MBMD Log:

```
mbmd  ()
config: using /etc/mbmd.yaml
config: declared device SDM:2
config: declared device SDM:3
config: declared device SDM:1
httpd: starting api at 0.0.0.0:8888
mqtt: connecting at [tcp://172.17.2.21:1883]
mqtt: connecting at [tcp://172.17.2.21:1883]
initialized device SDM1.1: {SDM Eastron SDM meters   }
http: response.Write on hijacked connection
mqtt: error: write tcp 172.30.2.1:37294->172.17.2.21:1883: use of closed network connection
mqtt: error: write tcp 172.30.2.1:37298->172.17.2.21:1883: use of closed network connection
mqtt: error: write tcp 172.30.2.1:37530->172.17.2.21:1883: use of closed network connection
mqtt: error: write tcp 172.30.2.1:37534->172.17.2.21:1883: use of closed network connection
initialized device SDM1.2: {SDM Eastron SDM meters   }
mqtt: error: write tcp 172.30.2.1:39398->172.17.2.21:1883: use of closed network connection
mqtt: error: write tcp 172.30.2.1:39674->172.17.2.21:1883: use of closed network connection
mqtt: error: write tcp 172.30.2.1:40300->172.17.2.21:1883: use of closed network connection
mqtt: error: write tcp 172.30.2.1:40304->172.17.2.21:1883: use of closed network connection
initialized device SDM1.3: {SDM Eastron SDM meters   }
mqtt: error: write tcp 172.30.2.1:42336->172.17.2.21:1883: use of closed network connection

```

und komischerweise während des MBMD Starts wieder den bekannten Fehler von openHAB, den wir nicht verstehen:

```
[WARN ] [.internal.convention.homie300.Device] - Could not subscribe
java.util.concurrent.CompletionException: java.lang.Exception: Did not receive mandatory topic value: mbmd/sdm1-3/meter/$name
```

Dafür kommt aber beim stoppen von MBMD keine openHAB Exception mehr :-)

mach das mit dem Last Will ruhig noch, ich kann mir ja aus den Teilen deines PRs eine für mich funktionierende Version bauen - also keine Eile.

andig, 2019-10-01:
Hab Die Ursache gefunden- da mehrere Mqtt Verbindungen müssen die unterschiedliche ClientIDs haben. Kein triviales Protokoll....
Der Rest ist ggf. Folgefehler, ohne „normales“ Mqtt sehe ich den nicht.

chrostek, 2019-10-01:
Mit dem Code aus deinem neusten Commit habe ich MBMD jetzt drei mal neu gestartet und bekomme weder in MBMD noch in openHAB mehr irgendwelche Fehler/Exceptions angezeigt. 
Vielen Dank!

andig, 2019-10-01:
> Vielen Dank!

Ich habe zu danken! Wenn Du weitere Issues findest gerne melden!

closed 2019-10-01 by andig

andig, 2019-10-02:
Ps.: Du kannst den mbmd Prozess jetzt auch hart killen, dann zeigt Homie „lost“ anstatt „disconnected“. Was jetzt ebenfalls noch fehlt ist nicht nur den Status des Prozesses dort zu reflektieren, sondern auch den der Devices. Kommt auf die Todo Liste.

andig, 2019-10-04:
@chrostek siehe https://github.com/volkszaehler/mbmd/pull/37 - das sollte jetzt nochmal runder sein.

PR #34 (2019-10-01): Make mqtt topic compatible with Homie
Fix #33