volkszaehler/mbmd #48: SBC ALE3 Problems

kugelkopf123, 2019-11-21:
I own a SBC ALE3 Modbus. Unfortunately, I do not manage to read this Meter. I tried it both: via Tasmota (SmartMeter) on a Wemos D1 Mini and via the MbMd software on a Raspberry Pi3b. On the Raspi both, via internal UART interface and via a USB2UART converter. I get no response from the SBC back on my RS485 converter. ([https://hacktronics.co.in/rs232485-usb-ttl-converters/scm-ttl-to-rs485-converter-485-to-uart-serial-port-automatic-flow-control-module](https://hacktronics.co.in/rs232485-usb-ttl-converters/scm-ttl-to-rs485-converter-485-to-uart-serial-port-automatic-flow-control-module)) 
I have already swapped the wires, tried at both ends with and without 120ohm terminator, as well as simply swapped TX / RX between RS485adapter and Pi.

Has anyone ever had experience with this meter?

The supply line is about 1m Shield TP.

andig, 2019-11-21:
I cannot comment on possible electrical problems, but some things to double-check are:

- modbus id
- data rate and coms parameters
- different adapter (I'm using a dirt-cheap rs485 USB adapter)

The `mbmd` implementation is totally untested afaik as Saia wasn't willing to provide any help or test sample. 
If you want I could try remotely if you want to provide access via pm (cpuidle@gmx.de).

andig, 2019-11-26:
Solved with different adapter.

closed 2019-11-26 by andig

andig, 2020-02-23:
@kugelkopf123 seems I have a similar problem now. Device not responding to by rs485/tcp converter. Which one did you end up using?

kugelkopf123, 2020-02-24:
@andig Sometimes the ALE3 is a bit bitchy.
I have the feeling that if you fill him up with data / requests that he doesn't understand, he simply won't answer anymore. At least this was my initial experience, if you leave him alone for a short time and the baud rate and parity are correct, he replied properly.

Further Information:
[Link Howto](https://docs.google.com/document/d/1olSGZcaE_vkdNXzN_p0zGee61xKCcK3HJe3Al00xj2g/edit)
[Link Tasmota SML](https://tasmota.github.io/docs/#/peripherals/Smart-Meter-Interface?id=sbc-ale3-modbus)

andig, 2020-02-24:
@kugelkopf123 the ALE should auto-detect baud rate and parity. I've tried with 9600 and 8E1 and 8N1 both, no luck. Can you confirm these are working for you?

kugelkopf123, 2020-02-24:
>@kugelkopf123 the ALE should auto-detect baud rate and parity. I've tried with 9600 and 8E1 and 8N1 both, no luck. Can you confirm these are working for you?

@andig Yes. It works with 9600 8E1. Are you shure that you use the right ID within the request messages?

andig, 2020-02-25:
Yes, id is correct. Guess I‘ll need to recheck the wires installation.