volkszaehler/mbmd #50: API documentation errors

bikeymouse, 2019-11-26:
Hi,
 
I have tried mbmd and found that the documentation on the API is not correct.

It says:
_mbmd provides a convenient REST API. Supported endpoints under /api are:
/last/{ID} latest data for device
/minuteavg/{ID} averaged data over last minute
/status daemon status_

This however should be:
_mbmd provides a convenient REST API. Supported endpoints under /api are:
/last/{ID} latest data for a specific device or /last for all devices
/avg/{ID} averaged data over last minute for  a specific device or /avg for all devices
/status daemon status_

andig, 2019-11-26:
The next sentence in the docs is:

> Both device APIs can also be called without the device id to return data for all connected devices.

closed 2019-11-26 by andig

bikeymouse, 2019-11-26:
You are right, I overlooked that, sorry.
However the /minuteavg does not exist. It should be /avg