volkszaehler/mbmd #52: feature request: variable polling frequency and registers

bikeymouse, 2019-11-26:
Great program, now that I have it working. However I notice that it looks like it polls my meter like crazy (every second?) and requests all parameters from my SDM630 meter.

However I only would like query my meter every minute and want only with a subset of all possible parameters, so as not to produce too much overhead and pushing useless data to Influx (or MQTT).
Is there a place in the code where I can adjust this? Or even better: would you consider to make this configurable? Perhaps via the input parameters or even better using the mbmd.yaml so this can be configured specifically for each meter.

andig, 2019-11-26:
I'm not concerned with the overhead but please see https://github.com/volkszaehler/mbmd/issues/22 for open roadmap topics- this includes rate limiting. For limiting the data pushed I could suggest https://github.com/andig/ingress.

bikeymouse, 2019-11-26:
Thanks for your quick reaction. I think even on my Raspberry Pi2b this can be handled by Influx although this would wear the SSD additionally while perhaps the amounts of write for my use case could be reduced by more then 60-fold (from 1/sec to 1/min) if I could reduce the polling frequency. And then again by 2 or 3 times by not writing all kind of parameters I will never look at.

While this rate-limiting is on the roadmap, could I change this somewhere in the code?

And would it work If I edit mdm.go and comment out all the op-codes that I don't need?

andig, 2019-12-25:
See https://github.com/volkszaehler/mbmd/pull/59

bikeymouse, 2019-12-26:
Tried it and it works great, thanks for the update!

Perhaps good to at this in the documentation as well, I struggled a bit to find out that the --rate flags needs not only a value but also a unit parameter, e.g. "--rate 30s"

bilbolodz, 2020-01-03:
Could you please add more documentation about "rate limit feature"? Is it possible to put these parameters into config file?

andig, 2020-01-03:
Right now, `--rate|-r` can only be set globally, on command line and as global setting in the yaml. I will add per-device rate limit to the todo list.

bilbolodz, 2020-01-03:
OK but please give some details about --rate parameter. Maybe a little example with some explanation?

andig, 2020-01-03:
Just add it:

```
# REST api, use 127.0.0.1 to restrict to localhost
api: 127.1:8081
rate: 30s
...
```

bilbolodz, 2020-01-03:
OK but "rate: 30s" what it means?

andig, 2020-01-03:
Not sure I follow. It means that the device will be polled every 30 seconds?!

bilbolodz, 2020-01-03:
OK, I see. Ideal solution (for me of course) is to keep frequency of pooling device BUT throttle frequency of Influxdb updates.Maybe it also could be good idea (for some parameters) send data to Influxdb ONLY on value change.

andig, 2020-01-03:
Please open separate issue for influx throttling. I'm not sure thats very benefitial but we can keep track of it.

bilbolodz, 2020-01-03:
Done.

andig, 2026-09-30:
Sorry for letting this sit for so long without a resolution. The request is valid, but nobody has picked it up in the meantime and there are no plans to implement it from the maintainer side, so I am closing this to keep the tracker honest. If it is still relevant, a PR is very welcome and we will happily reopen.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig