volkszaehler/mbmd #63: SDM630, TCP communication via USR-410

FollowTheWizard, 2020-01-01:
Hi, 
I tired to collect data from my two SDM630 meters meters with alpha11&alpha12 over TCP.
I used the precompiled software "mbmd_0.8-alpha.12_linux_armv6.tar.gz" that runs on RaspberryPi -> "USR-TCP232-410S" -> SDM:ID1 & SDM:ID2.
Ìt looks like I get a response from the meters but they are not detected correclty.
Currenlty I am using Node-Red ( where I can pull data).

What am I missing?

terminal copy:
pi@raspberrypi:/usr/local/bin $ ./mbmd scan --raw -a 192.168.178.31:26
2020/01/01 09:49:59 config: creating TCP connection for 192.168.178.31:26
2020/01/01 09:49:59 starting bus scan on 192.168.178.31:26
2020/01/01 09:49:59 modbus: send 00 01 00 00 00 06 01 03 9c 40 00 02
2020/01/01 09:50:00 modbus: recv 00 01 00 00 00 07 01 03 04 00 00 00 00
2020/01/01 09:50:00 modbus: send 00 02 00 00 00 06 01 03 c3 50 00 02
2020/01/01 09:50:00 modbus: recv 00 02 00 00 00 07 01 03 04 00 00 00 00
2020/01/01 09:50:00 modbus: send 00 03 00 00 00 06 01 03 00 00 00 02
2020/01/01 09:50:00 modbus: recv 00 03 00 00 00 07 01 03 04 42 70 00 00
2020/01/01 09:50:00 device 1: n/a
2020/01/01 09:50:00 modbus: send 00 04 00 00 00 06 02 03 9c 40 00 02
2020/01/01 09:50:00 modbus: recv 00 04 00 00 00 07 02 03 04 00 00 00 00
2020/01/01 09:50:00 modbus: send 00 05 00 00 00 06 02 03 c3 50 00 02
2020/01/01 09:50:00 modbus: recv 00 05 00 00 00 07 02 03 04 00 00 00 00
2020/01/01 09:50:00 modbus: send 00 06 00 00 00 06 02 03 00 00 00 02
2020/01/01 09:50:00 modbus: recv 00 06 00 00 00 07 02 03 04 42 70 00 00
2020/01/01 09:50:00 device 2: n/a
2020/01/01 09:50:00 modbus: send 00 07 00 00 00 06 03 03 9c 40 00 02
2020/01/01 09:50:01 modbus: send 00 08 00 00 00 06 03 03 c3 50 00 02
2020/01/01 09:50:02 modbus: send 00 09 00 00 00 06 03 03 00 00 00 02
2020/01/01 09:50:03 device 3: n/a

FollowTheWizard, 2020-01-01:
I also ran the scan with the option --rtu due to this line in the help:
Typically used with RS485 to Ethernet adapters that don't perform protocol conversion (e.g. USR-TCP232)

terminal copy:
pi@raspberrypi:/usr/local/bin $ ./mbmd scan --rtu --raw -a 192.168.178.31:26
2020/01/01 10:17:52 config: creating RTU over TCP connection for 192.168.178.31:26
2020/01/01 10:17:52 starting bus scan on 192.168.178.31:26
2020/01/01 10:17:52 modbus: send 01 03 4a 38 00 02 53 de
2020/01/01 10:17:53 modbus: send 01 04 00 08 00 02 f0 09
2020/01/01 10:17:54 modbus: send 01 03 00 23 00 01 75 c0
2020/01/01 10:17:55 modbus: send 01 04 00 00 00 02 71 cb
2020/01/01 10:17:56 modbus: send 01 03 5b 00 00 02 d7 2f
2020/01/01 10:17:57 modbus: send 01 03 00 04 00 02 85 ca
2020/01/01 10:17:58 modbus: send 01 03 50 02 00 02 74 cb
2020/01/01 10:17:59 device 1: n/a
2020/01/01 10:17:59 modbus: send 02 03 4a 38 00 02 53 ed
2020/01/01 10:17:59 modbus: recv 00 35 00 00 00 57 01 04 54 43 64 ba c1 43 5e f0 cd 43 60 3f ef 00 00 00 00 3f b8 e3 da 3e 77 10 80 00 00 00 00 43 8a 83 32 00 00 00 00 41 10 2d 79 43 9e f9 de 42 53 6c 68 c1 0c 4c 8b 43 1c 0c 11 c2 51 e6 f5 3f 80 00 00 3f 5f 0c 23 3d f5 3d 97 00 00 00 00 41 eb 28 10 c2 a6 41 df
2020/01/01 10:17:59 modbus: send 02 04 00 08 00 02 f0 3a
2020/01/01 10:17:59 modbus: recv 00 36 00 00 00 5f 01 04 5c 43 61 54 cd 00 00 00 00 3f 0d c7 d0 3f d4 ab b8 00 00 00 00 43 87 5c 5e 00 00 00 00 43 92 ed 3a 00 00 00 00 42 bb 9c 96 3f 72 9a d4 00 00 00 00 41 94 f1 fb 00 00 00 00 42 47 c9 77 46 3b 4f f7 3d 0f 5c 28 44 b1 a5 2f 44 22 bb b6 46 3e 17 24 47 34 d6 83 44 8f 43 7a 46 04 e7 af
2020/01/01 10:17:59 modbus: send 02 03 00 23 00 01 75 f3
2020/01/01 10:18:00 modbus: recv 00 37 00 00 00 13 01 04 10 44 90 72 aa 46 04 ec 1a 40 81 0a a9 41 56 8a d4
2020/01/01 10:18:00 modbus: send 02 04 00 00 00 02 71 f8
2020/01/01 10:18:00 modbus: recv 00 38 00 00 00 37 01 04 34 43 c3 9c d5 43 c1 af 97 43 c4 28 88 43 c3 26 fc 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3f a9 3c a2
2020/01/01 10:18:00 modbus: send 02 03 5b 00 00 02 d7 1c
2020/01/01 10:18:00 modbus: recv 00 39 00 00 00 4b 01 04 48 40 09 35 dd 40 2f 65 a8 40 0f 3d 4d 00 00 00 00 41 36 d2 3b 00 00 00 00 00 00 00 00 40 17 f2 f0 00 00 00 00 00 00 00 00 41 96 ba ce 00 00 00 00 3f 48 29 26 40 82 2a 92 3f 04 69 91 41 9e 32 7b 41 77 aa 6b 41 39 30 cd
2020/01/01 10:18:00 modbus: send 02 03 00 04 00 02 85 f9
2020/01/01 10:18:00 modbus: recv 00 3a 00 00 00 67 01 04 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 46 3b 50 1b 45 01 81 85 45 1d 9c bc 45 a9 06 e5 45 7d 95 58 3d 0b 43 95 00 00 00 00 3a 83 12 6e 45 1d 9d 47 45 a9 06 e5 45 7d 95 5c 43 ef 6f 7c 44 1e a4 49 43 99 dc ac 42 87 44 9b 43 9b 32 2d 43 88 74 18 44 08 a0 51 44 6c 3d 60 44 11 28 62 00 00 00 00
2020/01/01 10:18:00 modbus: send 02 03 50 02 00 02 74 f8
2020/01/01 10:18:01 device 2: n/a
2020/01/01 10:18:01 modbus: send 03 03 4a 38 00 02 52 3c
2020/01/01 10:18:02 modbus: send 03 04 00 08 00 02 f1 eb
2020/01/01 10:18:03 modbus: send 03 03 00 23 00 01 74 22
2020/01/01 10:18:04 modbus: send 03 04 00 00 00 02 70 29
2020/01/01 10:18:05 modbus: send 03 03 5b 00 00 02 d6 cd
2020/01/01 10:18:06 modbus: send 03 03 00 04 00 02 84 28
2020/01/01 10:18:07 modbus: send 03 03 50 02 00 02 75 29
2020/01/01 10:18:08 device 3: n/a
2020/01/01 10:18:08 modbus: send 04 03 4a 38 00 02 53 8b
2020/01/01 10:18:09 modbus: recv 00 3b 00 00 00 57 01 04 54 43 64 cd e5 43 5f 3d c8 43 60 16 e6 3d 4e a0 d4 3f b5 1d d5 3e 74 59 6f 00 00 00 00 43 87 0f 45 00 00 00 00 41 0c ff 31 43 9b ea 58 42 50 12 0f c1 08 cd 6e 43 1b cd 21 c2 4e c9 be 3f 80 00 00 3f 5d c1 af 3d e3 09 8a c2 97 fd 8a 41 ef d2 ee c2 a7 48 63
2020/01/01 10:18:09 modbus: send 04 04 00 08 00 02 f0 5c
2020/01/01 10:18:09 modbus: recv 00 3c 00 00 00 5f 01 04 5c 43 61 67 dc 00 00 00 00 3f 0d 97 2a 3f d4 62 c0 00 00 00 00 43 87 2a 5e 00 00 00 00 43 93 0f bd 00 00 00 00 42 bd 39 2d 3f 72 64 cc 00 00 00 00 41 96 1f e0 00 00 00 00 42 47 b0 09 46 3b 4f f8 3d 0f 5c 28 44 b1 a5 2f 44 22 bb b6 46 3e 17 25 47 34 d6 83 44 8f 43 7a 46 04 e7 af
2020/01/01 10:18:09 modbus: send 04 03 00 23 00 01 75 95
2020/01/01 10:18:10 modbus: recv 00 3d 00 00 00 13 01 04 10 44 90 72 aa 46 04 ec 1a 40 81 0a a9 41 56 8a d4
2020/01/01 10:18:10 modbus: send 04 04 00 00 00 02 71 9e
2020/01/01 10:18:10 modbus: recv 00 3e 00 00 00 37 01 04 34 43 c3 bc 7d 43 c1 c5 04 43 c4 14 28 43 c3 31 e3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3f a8 95 68
2020/01/01 10:18:10 modbus: send 04 03 5b 00 00 02 d7 7a
2020/01/01 10:18:10 modbus: recv 00 3f 00 00 00 4b 01 04 48 00 00 00 00 40 0c 7f 04 3f 4f 2a 76 00 00 00 00 41 30 dc 5c 00 00 00 00 00 00 00 00 3f 80 31 16 00 00 00 00 00 00 00 00 41 99 5c c0 00 00 00 00 3f 48 29 26 40 82 2a 92 3f 04 69 91 41 9e 32 7b 41 77 aa 6b 41 39 30 cd
2020/01/01 10:18:10 modbus: send 04 03 00 04 00 02 85 9f
2020/01/01 10:18:10 modbus: recv 00 00 00 00 00 67 01 04 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 46 3b 50 1c 45 01 81 85 45 1d 9c bc 45 a9 06 e7 45 7d 95 58 3d 0b 43 95 00 00 00 00 3a 83 12 6e 45 1d 9d 47 45 a9 06 e7 45 7d 95 5c 43 ef 6f 7c 44 1e a4 49 43 99 dc ac 42 87 44 9b 43 9b 32 2d 43 88 74 18 44 08 a0 51 44 6c 3d 60 44 11 28 62 00 00 00 00
2020/01/01 10:18:10 modbus: send 04 03 50 02 00 02 74 9e
2020/01/01 10:18:11 device 4: n/a

andig, 2020-01-01:
Pleaseuse current docker/git version that supports the —rtu switch. See https://github.com/volkszaehler/mbmd/pull/57 for discussion.

andig, 2020-01-02:
> Please use current docker/git version that supports the —rtu switch. See #57 for discussion.

I just realize you've already done that. Rechecked with my 2 SDM:

    ❯ gor main.go scan -a rs485.fritz.box:23 --rtu --raw                                                       
    2020/01/02 10:21:47 config: creating RTU over TCP connection for rs485.fritz.box:23
    2020/01/02 10:21:47 starting bus scan on rs485.fritz.box:23
    2020/01/02 10:21:47 modbus: send 01 03 4a 38 00 02 53 de
    2020/01/02 10:21:48 modbus: recv 01 03 04 00 00 00 00 fa 33
    2020/01/02 10:21:48 modbus: send 01 04 00 08 00 02 f0 09
    2020/01/02 10:21:48 modbus: recv 01 04 04 40 88 fa 31 ed 1a
    2020/01/02 10:21:48 modbus: send 01 03 00 23 00 01 75 c0
    2020/01/02 10:21:48 modbus: recv 01 03 02 00 00 b8 44
    2020/01/02 10:21:48 modbus: send 01 04 00 00 00 02 71 cb
    2020/01/02 10:21:48 modbus: recv 01 04 04 43 67 af 55 e2 10
    2020/01/02 10:21:48 device 1: SDM type device found, VoltageL1: 231.68
    2020/01/02 10:21:48 modbus: send 02 03 4a 38 00 02 53 ed
    2020/01/02 10:21:48 modbus: recv 02 03 04 00 00 00 00 c9 33
    2020/01/02 10:21:48 modbus: send 02 04 00 08 00 02 f0 3a
    2020/01/02 10:21:48 modbus: recv 02 04 04 40 8b 6a 89 42 68
    2020/01/02 10:21:48 modbus: send 02 03 00 23 00 01 75 f3
    2020/01/02 10:21:48 modbus: recv 02 03 02 00 00 fc 44
    2020/01/02 10:21:48 modbus: send 02 04 00 00 00 02 71 f8
    2020/01/02 10:21:48 modbus: recv 02 04 04 43 67 9a cd c7 ea
    2020/01/02 10:21:48 device 2: SDM type device found, VoltageL1: 231.60
    2020/01/02 10:21:48 modbus: send 03 03 4a 38 00 02 52 3c
    2020/01/02 10:21:48 modbus: recv 01 04 04 3f b4 5b e5 4d 8d
    2020/01/02 10:21:48 modbus: send 03 04 00 08 00 02 f1 eb
    2020/01/02 10:21:49 modbus: send 03 03 00 23 00 01 74 22
    2020/01/02 10:21:49 modbus: recv de df 92 83 63 7b 9e
    2020/01/02 10:21:49 modbus: send 03 04 00 00 00 02 70 29
    2020/01/02 10:21:50 modbus: send 03 03 5b 00 00 02 d6 cd
    2020/01/02 10:21:50 modbus: recv 01 04 04 3f b8 01 9f 36 4d
    2020/01/02 10:21:50 modbus: send 03 03 00 04 00 02 84 28

Your modbus responses look strage- more data than a simple error or "none" reply. Could you double-check your USR settings, i.e. data/stop bits, and tcp server mode?

andig, 2020-01-03:
ping @FollowTheWizard do you need further help?

FollowTheWizard, 2020-01-03:
Thank you for your help.
I was able to collect data with mbmd. I changed a few settings in the USR-410S. 
(Still do not know why this was necassery. I was able to collect data via node-red & "node-red-contrib-modbus" previously) 
I only noticed that the UI of the meters get reset (SDM630 reboots) by frequently polling them.
I think I'll stick with node-red.

andig, 2020-01-03:
> I was able to collect data with mbmd. I changed a few settings in the USR-410S.

Very good. Would you mind- for the benefit of other users- explain which settings you've changed?

> I only noticed that the UI of the meters get reset (SDM630 reboots) by frequently polling them.

Interesting. I have two SDMs and will try to verify that. Did you check at which polling rate this occurs? By default, mbmd polls once per second- was that enough to trigger the problem? Would it also happen when you polled on SDM only?

andig, 2020-01-05:
> Interesting. I have two SDMs and will try to verify that. 

@FollowTheWizard I've tried to hit my two SDMs hard, no delay, from 2 different PCs and could not reproduce any rebooting. Any help how to provoke and reproduce this would be appreciated.

FollowTheWizard, 2020-01-05:
Yes, polling with once per second was enough. 
This was happening by polling only one meters as well with two.
Polling with 30s has the same effect.
Firmware on both meters: 1.3022017
A/B Lines should be connect correct, because I can talk to the device via the node-red module.

USR settings (RS485 section): 
baudrate: 9600,  (same as on SDM630)
DataSize: 8, (same as on SDM630)
Parity to None (Does mbmd support odd?) (same as on SDM630)
Stop bits 1 (same as on SDM630)
Flow Control None
UART Packet Time&Length 0
Sync Baudrate on
Hearbeat off

Socket A Parameters: TCP Server / ModbusTCP
ModbusTCP Poll on timeout 200ms (I changed this paramter from off to on -> received data)
Rest is off/none

bikeymouse, 2020-01-05:
As you are both using the SDM630 like I do, another question: does anybody know what the difference is between:
Power ("Total System Power" according to the SDM modbus manual) at address 0x0034
and
ImportPower ("Total system power demand" according to the SDM manual) at address 0x0054?

From the measurements I get it looks like the "power demand" shows an integrated version (over 5 or 10 minutes) of the system power.

andig, 2020-01-05:
I have mapped them as:

		Power:         0x0034,
		ImportPower:   0x0054,

I can double-check tomorrow what these show.

FollowTheWizard, 2020-01-05:
By default the demand values are integrated over 60 minutes (default value).
Can be changed with Register 40003 (page5: https://bg-etech.de/download/manual/SDM630Register1-5.pdf)
The other ones like "Total System Power" are the current values.
I also use a twisted shielded cable and a 120 Ohm Termination resistor if I remember correct.

andig, 2020-01-05:
Good idea- need to add the 120Ohm. Should be on both sides of the bus actually?

bikeymouse, 2020-01-05:
> By default the demand values are integrated over 60 minutes (default value).
> Can be changed with Register 40003 (page5: https://bg-etech.de/download/manual/SDM630Register1-5.pdf)
> The other ones like [...]

Thanks, just as I thought, although the manual suggests that the "Demand Integration Time" (that you can also set in the set-up menu), is for the calculation of the Max Power and Current values.
I'll experiment with this time to see how it changes the output of register 0x0054.

bikeymouse, 2020-01-05:
> Good idea- need to add the 120Ohm. Should be on both sides of the bus actually?

It should be done on both ends. However, as I understand it is actually only needed on longer wires in combination with higher bitrates (>200 feet and > 9600 baud), so perhaps I'll have to see if it helps with the time-out errors if I actually remove them. See also [here](http://ltxfaq.custhelp.com/app/answers/detail/a_id/1382/~/rs485-%26-rs422---when-is-termination-required%3F).

andig, 2020-02-05:
I feel this has been answered. Please feel free to open new issues if needed.

closed 2020-02-05 by andig