volkszaehler/mbmd #65: Roadmap to 1.0

andig, 2020-01-03:
These things that should be addressed until 1.0:

- [ ] allow custom device names using config file
- [ ] rethink the device naming approach (maybe even allow backwards compatibility)
- [ ] add read/write commands to support modbus "fiddling" (#56)

mivola, 2021-05-11:
I really like the idea to allow custom names for devices, I even asked for it b/c I didnt know about this roadmap: https://www.photovoltaikforum.com/thread/134974-mbmd-neues-werkzeug-zum-auslesen-von-wechselrichtern-und-z%C3%A4hlern/?postID=2274375#post2274375

mivola, 2022-03-29:
@andig are there any plans to implement the "allow custom names" feature in the near future? I'm asking b/c I added another inverter to my PV system and therefore added a new adapter & device to my mbmd.yaml. The result is that the new device now uses the name of the old inverter (SMA2.126) and the old inverter got a new name (SMA3.126) ... This is a bit misleading and I'd need to change the REST URL in my automation systems accordingly...

andig, 2022-03-29:
I don't have plans but would be happy to accept a PR. It seems the entire device handling could need an update for more flexibility.

mivola, 2022-03-29:
> I don't have plans but would be happy to accept a PR. It seems the entire device handling could need an update for more flexibility.

Unfortunately I dont have any go knowledge so I cant really help here :-(

Furthermore I tried to work around the problem by using different naming for the devices and ordering of devices and adapters but nothing helped. So maybe you can tell me how the code determines the order in which device names are created?

glfp, 2023-01-10:
+1 for "allow custom names" ! I'll try a PR soon ! I'm restarting all my old project with Influxdb2 !