volkszaehler/mbmd #69: Fronius Inverter support

bilbolodz, 2020-01-04:
I'm trying to add support for my Fronius Symo Inverter, could you please tell how should I start? It looks that it's "SunSpec specification" compatible.

https://www.fronius.com/~/downloads/Solar%20Energy/Firmware/SE_FW_Fronius_Modbus_Card_Register_DE-EN.pdf

andig, 2020-01-04:
If it's Sunspec it already is supported- just use FRONIUS:id and the TCP address for the adapter.

bilbolodz, 2020-01-04:
What's "id"? In inverter I can set "String Control Adress-Offset". Now it's set to "101", after:

mbmd run --raw -a Fronius_IP -d FRONIUS:101

2020/01/04 16:41:54 config: creating TCP connection for X.X.X.X:502
2020/01/04 16:44:20 config: declared device FRONIUS:101
2020/01/04 16:44:20 httpd: starting api at 0.0.0.0:8080
2020/01/04 16:44:20 modbus: send 00 01 00 00 00 06 65 03 9c 40 00 02
2020/01/04 16:44:20 modbus: recv 00 01 00 00 00 03 65 83 0b
2020/01/04 16:44:20 modbus: send 00 02 00 00 00 06 65 03 c3 50 00 02
2020/01/04 16:44:20 modbus: recv 00 02 00 00 00 03 65 83 0b
2020/01/04 16:44:20 modbus: send 00 03 00 00 00 06 65 03 00 00 00 02
2020/01/04 16:44:20 modbus: recv 00 03 00 00 00 03 65 83 0b
2020/01/04 16:44:20 initializing device FRONIUS1.101 failed: not a SunSpec device

with:
mbmd run --raw -a Fronius_IP:502 -d FRONIUS:1 -v

2020/01/04 16:48:24 mbmd  ()
2020/01/04 16:48:24 config: creating TCP connection for X.X.X.X:502
2020/01/04 16:48:24 config: declared device FRONIUS:1
2020/01/04 16:48:24 httpd: starting api at 0.0.0.0:8080
2020/01/04 16:48:25 modbus: send 00 01 00 00 00 06 01 03 9c 40 00 02
2020/01/04 16:48:25 modbus: recv 00 01 00 00 00 07 01 03 04 53 75 6e 53
2020/01/04 16:48:25 modbus: send 00 02 00 00 00 06 01 03 9c 42 00 02
2020/01/04 16:48:25 modbus: recv 00 02 00 00 00 07 01 03 04 00 01 00 41
2020/01/04 16:48:25 modbus: send 00 03 00 00 00 06 01 03 9c 85 00 02
2020/01/04 16:48:25 modbus: recv 00 03 00 00 00 07 01 03 04 00 71 00 3c
2020/01/04 16:48:25 modbus: send 00 04 00 00 00 06 01 03 9c c3 00 02
2020/01/04 16:48:26 modbus: recv 00 04 00 00 00 07 01 03 04 00 78 00 1a
2020/01/04 16:48:26 modbus: send 00 05 00 00 00 06 01 03 9c df 00 02
2020/01/04 16:48:26 modbus: recv 00 05 00 00 00 07 01 03 04 00 79 00 1e
2020/01/04 16:48:26 modbus: send 00 06 00 00 00 06 01 03 9c ff 00 02
2020/01/04 16:48:26 modbus: recv 00 06 00 00 00 07 01 03 04 00 7a 00 2c
2020/01/04 16:48:26 modbus: send 00 07 00 00 00 06 01 03 9d 2d 00 02
2020/01/04 16:48:26 modbus: recv 00 07 00 00 00 07 01 03 04 00 7b 00 18
2020/01/04 16:48:26 modbus: send 00 08 00 00 00 06 01 03 9d 47 00 02
2020/01/04 16:48:27 modbus: recv 00 08 00 00 00 07 01 03 04 00 a0 00 30
2020/01/04 16:48:27 modbus: send 00 09 00 00 00 06 01 03 9d 79 00 02
2020/01/04 16:48:27 modbus: recv 00 09 00 00 00 07 01 03 04 ff ff 00 00
2020/01/04 16:48:27 modbus: send 00 0a 00 00 00 06 01 03 9c 44 00 42
2020/01/04 16:48:27 modbus: recv 00 0a 00 00 00 87 01 03 84 46 72 6f 6e 69 75 73 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 53 79 6d 6f 20 37 2e 30 2d 33 2d 4d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 33 2e 31 34 2e 31 2d 31 30 00 00 00 00 00 00 00 30 2e 33 2e 32 33 2e 30 00 00 00 00 00 00 00 00 33 30 31 39 38 30 30 37 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 71
2020/01/04 16:48:27 initializing device FRONIUS1.1 failed: sunspec: could not find supported model

andig, 2020-01-04:
The modbus slaveid you‘re inverter is listening to.

bilbolodz, 2020-01-04:
My number is "1" but I've got: "initializing device FRONIUS1.1 failed: sunspec: could not find supported model" (logs above). Any ideas?

bilbolodz, 2020-01-05:
After adding models: 120,121,122 (and some one "dummy" parameter from these models) and recompile, program is "working". "Working" it means running without error. Inverter IS detected:

initialized device Fronius1.1: {Fronius Symo 7.0-3-M 3.14.1-1 0.3.23.0 XXXXXXX}

but at web page there are no any measurements. Any idea how to deal with it?

bilbolodz, 2020-01-05:
OK, I'd to switch Modbus "Sunspec Model Type" from "float" to "int + SF" and now I can see some measurements. Question:
Models 120,121,122 are not quite interesting. How to "blacklist"(?) these models?
Now I'm using something like these:

 var opcodeMap = map[string]meters.Measurement{
[..]
        model120.ARtg:  meters.BatteryVoltage, 
        model121.WMax:       meters.BatteryVoltage,
        model122.ActVAh:        meters.BatteryVoltage,
}

But it's UGLY fix (used without any "deeper study of code").

andig, 2020-01-05:
@bilbolodz its really hard for me to follow what your actual problem and goal is. 

> OK, I'd to switch Modbus "Sunspec Model Type" from "float" to "int + SF" and added models: 

What do you mean? To solve which problem? As I've said, the inverter should be supported without any changes. What is missing?

> Models 120,121,122 are not quite interesting. How to "blacklist"(?) these models?

Why did you add them in the first place?

Could you please- before we continue- show the output of `mbmd inspect` for your device to understand which models it supports?

I can also suggest to switch to German if that would be easier.

bilbolodz, 2020-01-05:
> What do you mean? Switch what? 

"Sunspec Model Type" it's setting in Fronius Modbus card. You can chose which registers are "active" (float vs int + sf model)

>To solve which problem? 

I've put some comments before but then deleted because I've found sollution my self.

> As I've said, the inverter should be supported without any changes. What is missing?

 Main problem was that your "original code" ended with error: "sunspec: could not find supported model"

>Why did you add them in the first place?

To get mbmd working I'd to add these models to source and recompile. Now inverter is detected and working .

>I can also suggest to switch to German if that would be easier.

Sorry I don't speak german but if you can we can switch to Polish ;-)

>Could you please- before we continue- show the output of mbmd inspect for your device to understand which models it supports?

https://pastebin.com/VrCn1HN9

I'd like to get some information (values for string1 and string2) from model160.

bilbolodz, 2020-01-05:
> As I've said, the inverter should be supported without any changes. What is missing?
Sorry for trouble!!!! I've checked again and it turned out that it's working on your ORGINAL code!
I was stupid and tired yesterday!

But still question "how to get values" from model160 is active.

closed 2020-01-05 by bilbolodz

andig, 2020-01-05:
> "Sunspec Model Type" it's setting in Fronius Modbus card. You can chose which registers are "active" (float vs int + sf model)

That's really confusing. Sunspec clearly defines the encoding of each register- not sure why Fronius allows to chose anything.

> Sorry I don't speak german but if you can we can switch to Polish ;-)

Wish I could :/

> Sorry for trouble!!!! I've checked again and it turned out that it's working on your ORGINAL code!
I was stupid and tired yesterday!

You're totally welcome, glad it works.

> But still question "how to get values for MPPT" from model160 is active.

See my comment in #70