volkszaehler/mbmd #72: Wrong power factor reported on Fronius Inverter

bilbolodz, 2020-01-06:
On (my) Fronius inverter power factor is reported by mbdm as: -96.91.

andig, 2020-01-06:
See https://pastebin.com/VrCn1HN9, https://github.com/volkszaehler/mbmd/pull/61#issuecomment-570081618 and https://github.com/volkszaehler/mbmd/issues/69#issuecomment-570917405. Kostal has bugs in its Sunspec implementation, Fronius might as well. 

Please contact their support.

andig, 2020-01-06:
Idea: we might maintain a list of "known issues" of Sunspec violations by the various implementers. If we can confirm this it would be the third violation I know of.

andig, 2020-01-06:
Actually, I think PF can be between -1...1. They might treat it as % value, so it could be correct. Why are you raising this as issue- is it wrong?

bilbolodz, 2020-01-06:
If you treat "as %" thats fine. In "real math" as you said sin/cos is between -1 to 1. But you can be right maybe its "feature not a bug" but it looks a little strange when on the same web page (3 devices monitored) two devices reports "0.93" but one "-96.91". I thought that is a bug but probably you are right it's Fronius issue.

closed 2020-01-06 by bilbolodz