volkszaehler/mbmd #84: It seems that ABB meters show too much current and voltage by factor 10

klauskn, 2020-02-05:
Hello,

It seems that ABB meters show too much current and voltage by factor 10
For example, if I switch on a 500 watt halogen lamp, the power increases by 5000 watts.

According to the manual the voltage is output with a resolution of 01. But current and power with 001.

I have tried to adapt the code locally, unfortunately without success. With my local modification no more measured values are displayed. I get the following error message: device ABB1.2 did not respond (1/3)

Best regards,

Klaus

--- a/meters/rs485/abb.go
+++ b/meters/rs485/abb.go
@@ -143,11 +143,16 @@ func (p *ABBProducer) Probe() Operation {
 func (p *ABBProducer) Produce() (res []Operation) {
        for _, op := range []Measurement{
                VoltageL1, VoltageL2, VoltageL3,
-               CurrentL1, CurrentL2, CurrentL3,
        } {
                res = append(res, p.snip32u(op, 10))
        }

+       for _, op := range []Measurement{
+               CurrentL1, CurrentL2, CurrentL3,
+       } {
+               res = append(res, p.snip32u(op, 100))
+       }
+
        for _, op := range []Measurement{
                Cosphi, CosphiL1, CosphiL2, CosphiL3,
        } {
@@ -163,7 +168,7 @@ func (p *ABBProducer) Produce() (res []Operation) {
        for _, op := range []Measurement{
                Power, PowerL1, PowerL2, PowerL3,
        } {
-               res = append(res, p.snip32i(op, 10))
+               res = append(res, p.snip32i(op, 100))
        }

        for _, op := range []Measurement{

Caibuk, 2020-02-05:
I've noticed this nuisance after installing my ABB B23 212-100 Modbus meter.
I could correct the values in my home assistant visualization. But correct handling of it in the first place would be much appreciated.

andig, 2020-02-05:
Thanks for bringing this up. Fixed in 85.

Unfortunately ABB didn't want to work with me (or rather, mangle me through their partner certification programm and an unusable website) and I had to develop blindly without any chance for testing.

If anybody (ABB, are you out there?) want's to contribute an ABB meter for testing that would be most appreciated.

PR #85 (2020-02-05): Fix ABB scalers
Fix #84
closed 2020-02-05 by andig