volkszaehler/mbmd #87: After building unfortunately no data

klauskn, 2020-02-05:
When I try to build on my Raspberry 4, I get no error message but unfortunately no data in the web interface. 
On the console they are printed with -v.  
Anyone have an idea?

[image]

andig, 2020-02-05:
Whats in the browser console?

klauskn, 2020-02-05:
Source-Map-Fehler: Error: request failed with status 404
Ressourcen-Adresse: http://volkszaehler:8088/css/bootstrap.min.css
Source-Map-Adresse: bootstrap.min.css.map

Die Verbindung zu ws://volkszaehler:8088/ws wurde unterbrochen, während die Seite geladen wurde.

klauskn, 2020-02-05:
Excuse me, I just realized that 
data are displayed correctly on another PC.
After I have deleted the cache on the 1st PC, I see data there as well. So everything works as expected.

closed 2020-02-05 by klauskn