volkszaehler/mbmd #90: Trying to compile from the source -> invalid/unknown revision

wisekki, 2020-02-15:
I'm trying to compile MBMD from the source but I get this odd error that I can't get my head around with google :)

> go: github.com/andig/gridx-modbus@v0.0.0-20191223181555-ff0a69a1ad97: invalid version: unknown revision ff0a69a1ad97

In go.mod there's a line:

> replace github.com/grid-x/modbus v0.0.0-20191105145357-867898f52408 => github.com/andig/gridx-modbus v0.0.0-20191223181555-ff0a69a1ad97

andig, 2020-02-15:
You have an old version- replace is no longer needed, see https://github.com/volkszaehler/mbmd/blob/master/go.mod

closed 2020-02-15 by andig

wisekki, 2020-02-15:
Okay thnx.. I just downloaded sourcecode.zip from the releases.

andig, 2020-02-15:
Better clone from git. The missing revision is a temporary one when I needed a fix to be merged upstream...