volkszaehler/mbmd #92: error: "* '' has invalid keys: mqtt-clean"

bikeymouse, 2020-02-21:
I just did a docker build with the provided docker which ran fine, but if I start up the docker-container I get the following log with error:
```
2020/02/21 22:00:22 mbmd  ()
2020/02/21 22:00:22 config: using /etc/mbmd.yaml
2020/02/21 22:00:22 config: failed parsing config file /etc/mbmd.yaml: 1 error(s) decoding:
  * '' has invalid keys: mqtt-clean
```
This happens with both my own mdmd.yaml file as with the provided example file.
As I see there have been some changes with the mutt-clean flag this might be related.

andig, 2020-02-22:
Yes, the flags needs be removed as it worked in unexpected ways. I‘ll need to rework the config and make some additional changes to logging.

andig, 2020-02-22:
Actually, this has already been removed from the example: https://github.com/volkszaehler/mbmd/pull/88/files#diff-97b4cb71830459e3e0f9edc687a65357L12. Where do you still see it?

bikeymouse, 2020-02-22:
That's the strange thing, it is not present in my mdmd.yaml file (or the example) at all!

However if I search the code I can find references to "mqtt-clean"  in [run.go](https://github.com/volkszaehler/mbmd/blob/master/cmd/run.go) could that be the issue?

It is also still mentioned in the documentation in [mbmd_run.md](https://github.com/volkszaehler/mbmd/blob/master/docs/mbmd_run.md)

andig, 2020-02-22:
> That's the strange thing, it is not present in my mdmd.yaml

Not really since that file is not part of repo- you have created it yourself ;)

> However if I search the code I can find references to "mqtt-clean" in run.go

Fixed in https://github.com/volkszaehler/mbmd/pull/93

closed 2020-02-22 by andig

bikeymouse, 2020-02-22:
Thanks! It works fine now ;-)