volkszaehler/mbmd #95: Rest API not working with release binary

mdk2412, 2020-03-01:
Since updating sdm630 0.6.0 to mbmd release the rest API does not work anymore. Calling http://****:8080/last/ (same with avg, status) gives "404 page not found", ws gives "bad request".

The main page http://****:8080 shows the correct data.

What am I doing wroing?

andig, 2020-03-02:
Are you tying the api to 0.0.0.0?

mdk2412, 2020-03-02:
systemd status states "httpd: starting api at 0.0.0.0:8080". Accessing http://****:8080/ in firefox shows normal realtime stats. http://****:8080/last gives "404 page not found". http://****:8080/ws gives "Bad Request". Same with "curl -s".

andig, 2020-03-02:
Ahh, sorry. I forgot and the docs are not clear enough- all REST endpoints are below `/api` and `/ws` won't work if you try to http-connect to it.

PR #96 (2020-03-02): Document /api endpoint throughout
Fix #95
closed 2020-03-02 by andig