volkszaehler/mbmd #98: No readings on model 106 on ABB inverter

W-B, 2020-03-15:
I can't seem to get readings from model 160 data in web server or InfluxDB database. I only get partial information from model 103 data.

When I run mbmd inspect command, I see all the data I need, however, when using run, only selected data gets stored.

Is this by design?

andig, 2020-03-15:
It depends. You can see all models/blocks that we're reading here: https://github.com/volkszaehler/mbmd/blob/master/meters/sunspec/models.go#L15. What exactly are you missing?

W-B, 2020-03-15:
Yes, I did look at that file, however I was not able to trace it to logging code (no time & experience in Go lang). What gets logged is basically this:

```
{
   "results":[
      {
         "statement_id":0,
         "series":[
            {
               "name":"mbmd",
               "columns":[
                  "time",
                  "device",
                  "type",
                  "value"
               ],
               "values":[
                  [
                     "2020-03-15T14:17:01Z",
                     "Power-One1.247",
                     "Current",
                     5.2
                  ],
                  [
                     "2020-03-15T14:17:01Z",
                     "Power-One1.247",
                     "CurrentL1",
                     5.2
                  ],
                  [
                     "2020-03-15T14:17:01Z",
                     "Power-One1.247",
                     "CurrentL2",
                     5.2
                  ],
                  [
                     "2020-03-15T14:17:01Z",
                     "Power-One1.247",
                     "CurrentL3",
                     5.2
                  ],
                  [
                     "2020-03-15T14:17:01Z",
                     "Power-One1.247",
                     "DCPower",
                     3790
                  ],
                  [
                     "2020-03-15T14:17:01Z",
                     "Power-One1.247",
                     "Export",
                     2773.401
                  ],
                  [
                     "2020-03-15T14:17:01Z",
                     "Power-One1.247",
                     "Frequency",
                     50.01
                  ],
                  [
                     "2020-03-15T14:17:01Z",
                     "Power-One1.247",
                     "HeatSinkTemp",
                     31.8
                  ],
                  [
                     "2020-03-15T14:17:01Z",
                     "Power-One1.247",
                     "Power",
                     3676
[... 90 lines cut ...]
-- Block 1 --                                                                                                                            
ID                 1.00    uint16                                                                                                        
IDStr                      string                                                                                                        
DCA              416.00    uint16                                                                                                        
DCV             6814.00    uint16                                                                                                        
DCW             2858.00    uint16                                                                                                        
DCWH               0.00    acc32                                                                                                         
Tms                 n/a    uint32                                                                                                        
Tmp                 n/a    int16                                                                                                         
DCSt                n/a    enum16                                                                                                        
DCEvt               n/a    bitfield32                                                                                                    
-- Block 2 --                                                                                                                            
ID                 2.00    uint16                                                                                                        
IDStr                      string                                                                                                        
DCA               97.00    uint16                                                                                                        
DCV             3672.00    uint16                                                                                                        
DCW              361.00    uint16                                                                                                        
DCWH               0.00    acc32                                                                                                         
Tms                 n/a    uint32                                                                                                        
Tmp                 n/a    int16                                                                                                         
DCSt                n/a    enum16                                                                                                        
DCEvt               n/a    bitfield32
```   

Perhaps there are some settings that govern what gets written into the DB (and output into the web server)?

andig, 2020-03-15:
> What exactly are you missing?

Would still like to know so I can can add it ;)

W-B, 2020-03-16:
Fair enough :).

I'm missing:
- per MPPT line DC readings V, A, W
- Total AC energy (WH) produced by the plant

andig, 2020-03-16:
> I'm missing per MPPT line DC readings: voltage, current, power

Indeed, these should be available if you've compiled from git for yourself: https://github.com/volkszaehler/mbmd/blob/master/meters/sunspec/models.go#L56.

If not working please PM be for SSH remote access so I can take a look.

W-B, 2020-03-17:
It's there when compiling from source. Many thanks - great work with this tool - it's exactly what I was looking for (although I did manage to get a PhD on SunSpec before I found it)!

A suggestion for improvement, might be useful to some - add an option to specify (ideally through a config file) list of values to be dumped into DB.

Another one - add a SunSpec tag to your repo - it might be easier to find this nice piece of software :).

andig, 2020-03-17:
> A suggestion for improvement, might be useful to some - add an option to specify (ideally through a config file) list of values to be dumped into DB.

Sounds like #66 or do you mean something different?

> Another one - add a SunSpec tag to your repo - it might be easier to find this nice piece of software :).

Done :)

W-B, 2020-03-17:
> Sounds like #52 and #66 or do you mean something different?

Yes, it's like #52, exactly. Thanks!

closed 2020-03-17 by W-B

andig, 2020-03-17:
Closing here, will use #52 for tracking.