volkszaehler/mbmd #105: Program version is not available on command line

agerhard0, 2020-04-01:
With the command `mbmd version` the result is `version: unknown version` only instead of "version: 0.9".

Note: I just upgraded manually - at least I guess so - from 0.8 to 0.9. But due to the above I cannot verify on command line.

andig, 2020-04-02:
Should be fixed using https://github.com/volkszaehler/mbmd/releases/tag/0.9.1, can you confirm?

agerhard0, 2020-04-02:
Replacing 0.8 with 0.9.1 (incl. stopping/starting system service) does not solve the issue. Result is still `version: unknown version`. Also 0.9.1 still shows garbled output data on :8080. Going back to 0.8 solves at least this issue again.

LaForge112, 2020-04-02:
Same here "unknown version"
Also theres no Output for me  http: response.Write on hijacked connection from compress/gzip.(*Writer).Write (gzip.go:168)

Is this another Bug or is it related

andig, 2020-04-02:
Here's how it works for me:

    ❯ docker run -it volkszaehler/mbmd run -v
    2020/04/02 16:16:30 mbmd 0.9.1 (4f5e348)
    2020/04/02 16:16:30 config: no devices found - terminating

agerhard0, 2020-04-02:
@andig: What output does "mbmd version" create on your side?

With me on commandline with "mbmd run" only:
`2020/04/02 23:13:39 mbmd unknown version (unknown commit)`
`2020/04/02 23:13:39 config: using /etc/mbmd.yaml`

andig, 2020-04-03:
How did you get the software? How are you sure its 0.9.1?

agerhard0, 2020-04-03:
Pulling the tar.gz via wget directly to /local/usr/bin. Stopping service, removing all existing files in /local/usr/bin, unpacking archive, starting service (ok), "mbmd version".

andig, 2020-04-03:
Ahh, I can reproduce it now, will check.

andig, 2020-04-03:
Should be fixed in https://github.com/volkszaehler/mbmd/releases/tag/0.9.2. Closing in good faith.

closed 2020-04-03 by andig

andig, 2020-04-03:
Thanks for helping to find the root cause.