volkszaehler/mbmd #106: Version 0.9 garbles API-output on :8080 vs version 0.8

agerhard0, 2020-04-01:
Data output on :8080 is not ok with version 0.9, although itwas with 0.8 (s. att).

[image]

Reference:
https://www.photovoltaikforum.com/thread/134974-mbmd-neues-werkzeug-zum-auslesen-von-wechselrichtern-und-z%C3%A4hlern/?postID=1968266#post1968266

andig, 2020-04-02:
Fixed: https://github.com/volkszaehler/mbmd/releases/tag/0.9.1

closed 2020-04-02 by andig

agerhard0, 2020-04-02:
Replacing 0.8 with 0.9.1 (incl. stopping/starting system service) does not solve the issue. Result is still  garbled output data on :8080. Going back to 0.8 solves this issue again.

andig, 2020-04-02:
Please make sure to force-reload the web page to refresh the javascripts. If in doubt, clear browser cache.

agerhard0, 2020-04-02:
Clearing browser cache did the trick. 0.9.1 output on :8080 is fine here now.