volkszaehler/mbmd #118: Config file for 2 TCP adapters

bilbolodz, 2020-04-30:
My current setup is:

- Fronius Inverter (TCP connection: 192.168.105.252:502)
- 2 x SDM630 meters connected over "Modbus TCP converter"  (HF5111B: http://www.hi-flying.com/hf5111b) (TCP connection: 192.168.105.4:502)
Everything is fine when I'm using one of TCP connection (it means only Fronius or only SDM meters. It's working correctly also when I've Fronius connected via TCP and SDM meters only USB to RS485 converter). No I'm trying to connect all my devices via TCP and got strange error:

mbmd -c /etc/mbmd.yaml run
---
2020/04/30 13:30:11 config: using /etc/mbmd.yaml
2020/04/30 13:30:11 config: creating TCP connection for 192.168.105.252:502
2020/04/30 13:30:11 config: creating TCP connection for 192.168.105.4:502
2020/04/30 13:30:11 config: declared device SUNSPEC:1
2020/04/30 13:30:11 config: declared device SDM:1
2020/04/30 13:30:11 config: declared device SDM:2
2020/04/30 13:30:11 httpd: starting api at 0.0.0.0:8081
2020/04/30 13:30:11 initializing device SDM2.1 failed: not a SunSpec device
2020/04/30 13:30:14 initialized device Fronius1.1: {Fronius Symo 7.0-3-M 3.14.1-1 0.3.23.0 30198007}
2020/04/30 13:30:15 initializing device SDM2.2 failed: not a SunSpec device
---

My config file is:

---
# REST api, use 127.0.0.1 to restrict to localhost
api: 0.0.0.0:8081
rate: 60s

# adapters are referenced by device
adapters:
#- device: /dev/ttyUSB-modbus
#  baudrate: 38400
#  comset: 8N1 # "8E1" needs be quoted as string or will error
- device: 192.168.105.252:502
  rtu: false
- device: 192.168.105.4:502
  rtu: false

# list of devices
devices:
- name: Licznik wejscie
  type: sdm
  id: 1
  adapter: 192.168.105.4:502

- name: Licznik dom
  type: sdm
  id: 2
  adapter: 192.168.105.4:502

- name: Inverter Fronius
  type: sunspec
  id: 1
  adapter: 192.168.105.252:502
---

How should I get it working?

andig, 2020-04-30:
You are apparently trying to read the SDM meter using a TCP connection. The SDM are RTU meters. If your adapter does not perform protocol conversion (the cheap ones don't) you need `rtu: true`.

bilbolodz, 2020-04-30:
When I've changed rtu: to true I've got an error:

> 2020/04/30 14:42:25 device SDM2.1 did not respond (1/3)
> 2020/04/30 14:42:27 device SDM2.1 did not respond (2/3)
> 2020/04/30 14:42:29 device SDM2.1 did not respond (3/3)
It's not a problem of rtu setting.
From some reason mbmd (when has configured 2 TCP adapters) treats SDM meteres as "SunSpec device".

andig, 2020-04-30:
It seems to mix up the connections. Would you pm me at cpuidle@gmx.de for discussion if I could have a look into your network?

bilbolodz, 2020-04-30:
Yes of course.

andig, 2020-04-30:
Fixed in https://github.com/volkszaehler/mbmd/releases/tag/0.10

closed 2020-04-30 by andig