volkszaehler/mbmd #119: Docker not starting, throws error

dimagoltsman, 2020-05-02:
i couldnt get docker to start.  its throwing error probably some wrong arguments/

```
dima@dima-desktop:~/docker-mounts/compose$ sudo docker run -p 8080:8080  --device=/dev/ttyUSB0 volkszaehler/mbmd -a /dev/ttyUSB0 -u 0.0.0.0:8080 -d sdm:1
Error: unknown shorthand flag: 'u' in -u
unknown shorthand flag: 'u' in -u
Usage:
  mbmd [command]

Available Commands:
  help        Help about any command
  inspect     Inspect SunSpec device models and implemented values
  read        Read register (EXPERIMENTAL)
  run         Read and publish measurements from all configured devices
  scan        Scan for attached devices
  version     Show MBMD version
  write       Write register (EXPERIMENTAL)

Flags:
  -a, --adapter string   Default MODBUS adapter. This option can be used if all devices are attached to a single adapter.
                         Can be either an RTU device (/dev/ttyUSB0) or TCP socket (localhost:502).
                         The default adapter can be overridden per device
  -b, --baudrate int     Serial interface baud rate (default 9600)
      --comset string    Communication parameters for default adapter, either 8N1 or 8E1.
                         Only applicable if the default adapter is an RTU device (default "8N1")
  -c, --config string    Config file (default is $HOME/mbmd.yaml)
  -h, --help             Help for mbmd
      --raw              Log raw device data
      --rtu              Use RTU over TCP for default adapter.
                         Typically used with RS485 to Ethernet adapters that don't perform protocol conversion (e.g. USR-TCP232).
                         Only applicable if the default adapter is a TCP connection
  -v, --verbose          Verbose mode

Use "mbmd [command] --help" for more information about a command.
```

also, a docker-compose file would be great :)

andig, 2020-05-03:
> i couldnt get docker to start. its throwing error probably some wrong arguments/
> Error: unknown shorthand flag: 'u' in -u

The error says it all. Try:

    --api string                  REST API url. Use 127.0.0.1:8080 to limit to localhost. (default "0.0.0.0:8080")

> also, a docker-compose file would be great :)

What for?

dimagoltsman, 2020-05-03:
dont get it. what is --api?  it doesnt recognize it as well...

how should i start that docker?

dimagoltsman, 2020-05-03:
got it to work with
` docker run -p 8080:8080 --device=/dev/ttyUSB0 volkszaehler/mbmd run -a /dev/ttyUSB0 -d sdm:1`

the second "run" is missing in the readme...

closed 2020-05-03 by dimagoltsman