volkszaehler/mbmd #127: Add support for ORNO OR-WE-516 and OR-WE-517

bbiegun, 2020-05-31:
Hallo zusammen!

I'd like to add a support for ORNO OR-WE-516 and OR-WE-517 three-phase meters. They're almost identical - 517 additionally have 4 tariffs. 
I have a list of registers, lengths etc.

Unfortunately I never write a thing in Golang. I can can try, but no promises.
I reviewed the code for couple of meters and it seems very easy to adopt a new one. If I'm not mistaken you're supporting up to 2 tariffs. Would it be possible to add 2 more?

In theory I've registers for one-phase meters like OR-WE-514 or OR-WE-515 but haven't any of them to test.

andig, 2020-05-31:
Never heard of those? Please feel free to open a PR- lets start with 2 tariffs and add the rest when we get there (is there any use for 4 tariffs?). Let me know if you have specific questions.

andig, 2020-06-01:
I couldn‘t find any register documentation for this device. If you need help please post link to the relevant docs.

bbiegun, 2020-06-01:
I got it from the software they providing to read registers. Unfortunately there is no solid documentation of those.
Please allow me some time to gather them all together. 
Orno meters are used most in Poland, not sure where they're popular too.
https://www.partner.orno.pl/grafiki2/PC%20softwre170621.rar

andig, 2020-06-01:
Quick check found them on Amazon in Germany, too. Looks like an affordable replacement of the popular SDM meters.

ochykysh, 2020-06-01:
https://github.com/gituser-rk/orno-modbus-mqtt/blob/master/Register%20description%20OR-WE-514%26OR-WE-515.pdf - here is the register information.
I"m actually working on adding WE514/515 support(single phase variant)

bbiegun, 2020-06-01:
Registers for three-phase are below. I want to do a PR, but it seems, you'll do it a much faster  than me though ;)

strOBIS,szOBISName,UnitName, DataLength
0x0000,Serial number,,2
0x0002,Modbus ID,,1
0x0003,Modbus Baudrate,,1
0x0004,Software Version,,2
0x0006,Hardware Version,,2
0x0009,SO output rate ,,2
0x000B,Combined Code,,1
0x000C,HOLIDAY-WEEKEND Tariff,,1
0x000D,LCD Cycle time,s,1
0x000E,L1 Voltage,V,2
0x0010,L2 Voltage,V,2
0x0012,L3 Voltage,V,2
0x0014,Grid Frequency,Hz,2
0x0016,L1 Current,A,2
0x0018,L2 Current,A,2
0x001A,L3 Current,A,2
0x001C,Total Active Power,kW,2
0x001E,L1 Active Power,kW,2
0x0020,L2 Active Power,kW,2
0x0022,L3 Active Power,kW,2
0x0024,Total reactive power,kvar,2
0x0026,L1 reactive power,kvar,2
0x0028,L2 reactive power,kvar,2
0x002A,L3 reactive power,kvar,2
0x002C,Total Apparent Power,kVA,2
0x002E,L1 Apparent Power,kVA,2
0x0030,L2 Apparent Power,kVA,2
0x0032,L3 Apparent Power,kVA,2
0x0034,Total Power Factor,,2
0x0036,L1 Power Factor,,2
0x0038,L2 Power Factor,,2
0x003A,L3 Power Factor,,2
0x003C,DateTime,,4
0x0041,CRC code,,1
0x0100,Total Active Energy,kWh,2
0x0102,L1 Total Active Energy,kWh,2
0x0104,L2 Total Active Energy,kWh,2
0x0106,L3 Total Active Energy,kWh,2
0x0108,Forward Active Energy,kWh,2
0x010A,L1 Forward Active Energy,kWh,2
0x010C,L2 Forward Active Energy,kWh,2
0x010E,L3 Forward Active Energy,kWh,2
0x0110,Reverse Active Energy,kWh,2
0x0112,L1 Reverse Active Energy,kWh,2
0x0114,L2 Reverse Active Energy,kWh,2
0x0116,L3 Reverse Active Energy,kWh,2
0x0118,Total Reactive Energy,kvarh,2
0x011A,L1 Reactive Energy,kvarh,2
0x011C,L2 Reactive Energy,kvarh,2
0x011E,L3 Reactive Energy,kvarh,2
0x0120,Forward Reactive Energy,kvarh,2
0x0122,L1 Forward Reactive Energy,kvarh,2
0x0124,L2 Forward Reactive Energy,kvarh,2
0x0126,L3 Forward Reactive Energy,kvarh,2
0x0128,Reverse Reactive Energy,kvarh,2
0x012A,L1 Reverse Reactive Energy,kvarh,2
0x012C,L2 Reverse Reactive Energy,kvarh,2
0x012E,L3 Reverse Reactive Energy,kvarh,2
0x0130,T1 Total Active Energy,kWh,2
0x0132,T1 Forward Active Energy,kWh,2
0x0134,T1 Reverse Active Energy,kWh,2
0x0136,T1 Total Reactive Energy,kvarh,2
0x0138,T1 Forward Reactive Energy,kvarh,2
0x013A,T1 Reverse Reactive Energy,kvarh,2
0x013C,T2 Total Active Energy,kWh,2
0x013E,T2 Forward Active Energy,kWh,2
0x0140,T2 Reverse Active Energy,kWh,2
0x0142,T2 Total Reactive Energy,kvarh,2
0x0144,T2 Forward Reactive Energy,kvarh,2
0x0146,T2 Reverse Reactive Energy,kvarh,2
0x0148,T3 Total Active Energy,kWh,2
0x014A,T3 Forward Active Energy,kWh,2
0x014C,T3 Reverse Active Energy,kWh,2
0x014E,T3 Total Reactive Energy,kvarh,2
0x0150,T3 Forward Reactive Energy,kvarh,2
0x0152,T3 Reverse Reactive Energy,kvarh,2
0x0154,T4 Total Active Energy,kWh,2
0x0156,T4 Forward Active Energy,kWh,2
0x0158,T4 Reverse Active Energy,kWh,2
0x015A,T4 Total Reactive Energy,kvarh,2
0x015C,T4 Forward Reactive Energy,kvarh,2
0x015E,T4 Reverse Reactive Energy,kvarh,2
0x0300,TIME interval 1,,12
0x030C,TIME interval 2,,12
0x0318,TIME interval 3,,12
0x0324,TIME interval 4,,12
0x0330,TIME interval 5,,12
0x033C,TIME interval 6,,12
0x0348,TIME interval 7,,12
0x0354,TIME interval 8,,12
0x0360,TIME zone,,12

ochykysh, 2020-06-01:
Oh, it looks like the register addresses are completely different between 514/515 and 516/517.

andig, 2020-06-02:
Out of curiosity- how did you hear about mbmd?

closed 2020-06-02 by andig

bbiegun, 2020-06-03:
I was looking for a library similar to wmbusmeters. This one looked promising as supporting various meters, mqtt etc. I don't wanted to reinvent the wheel but to use existing framework to add support for a new meter :)

andig, 2020-06-03:
Ah, didn't know about wmbusmeters, thank you. If you're interested in meter data and visualisation, German https://volkszaehler.org project might also be of interest for you.

bbiegun, 2020-06-03:
I saw it, it's very promising. I'm starting to integrate meters in my homeassistant instance. For the analysis I was wondering about splunk instance, will look into this lib surely too