volkszaehler/mbmd #128: SolarEdge SE10K

BMOD89, 2020-05-31:
Hi, 

ich wollte mbmd für meinen SolarEdge SE10K einsetzen um die Daten an eine InfluxDB zu schicken. Leider komme ich nicht voran. 

Der inspect-Befehl gibt folgendes aus: 
<details>
  <summary>Ausgabe</summary>

```
/ # mbmd inspect -v -d SOLAREDGE:1@192.168.100.24:502
2020/05/31 17:25:47 config: creating TCP connection for 192.168.100.24:502
--------- Model 1 common ---------
Mn         SolarEdge     string
Md              SE10K    string
Opt                      string
Vr           0003.225    string
SN           7E14903B    string
DA               1.00    uint16
Pad            0x0067    pad
--------- Model 103 inverter ---------
A                    2.49    uint16
AphA                 0.81    uint16
AphB                 0.85    uint16
AphC                 0.82    uint16
A_SF                   -2    sunssf
PPVphAB            406.70    uint16
PPVphBC            409.20    uint16
PPVphCA            408.80    uint16
PhVphA             232.00    uint16
PhVphB             235.50    uint16
PhVphC             237.30    uint16
V_SF                   -1    sunssf
W                  360.00    int16
W_SF                   -1    sunssf
Hz                  49.98    uint16
Hz_SF                  -2    sunssf
VA                 584.90    int16
VA_SF                  -1    sunssf
VAr               -461.00    int16
VAr_SF                 -1    sunssf
PF                 -61.44    int16
PF_SF                  -2    sunssf
WH            17017776.00    acc32
WH_SF                   0    sunssf
DCA                  0.49    uint16
DCA_SF                 -4    sunssf
DCV                747.60    uint16
DCV_SF                 -1    sunssf
DCW                365.40    int16
DCW_SF                 -1    sunssf
TmpCab                n/a    int16
TmpSnk              40.43    int16
TmpTrns               n/a    int16
TmpOt                 n/a    int16
Tmp_SF                 -2    sunssf
St                      4    enum16
StVnd                   0    enum16
Evt1                  n/a    bitfield32
Evt2                  n/a    bitfield32
EvtVnd1        0x00000001    bitfield32
EvtVnd2               n/a    bitfield32
EvtVnd3               n/a    bitfield32
EvtVnd4        0x00000000    bitfield32
--------- Model 1 common ---------
Mn           WattNode    string
Md      WND-3Y-400-MB    string
Opt          Export+I    string
Vr                 31    string
SN            5007280    string
DA               1.00    uint16
Pad            0x00cb    pad
--------- Model 203 ac_meter ---------
A                            2.20    int16
AphA                         2.10    int16
AphB                         0.70    int16
AphC                         0.00    int16
A_SF                           -1    sunssf
PhV                        232.50    int16
PhVphA                     232.50    int16
PhVphB                     236.00    int16
PhVphC                     237.65    int16
PPV                       -247.70    int16
PhVphAB                   -249.68    int16
PhVphBC                   -246.92    int16
PhVphCA                   -246.50    int16
V_SF                           -2    sunssf
Hz                          49.99    int16
Hz_SF                          -2    sunssf
W                          -36.00    int16
WphA                       -15.00    int16
WphB                       -55.00    int16
WphC                        34.00    int16
W_SF                            0    sunssf
VA                         564.00    int16
VAphA                      281.00    int16
VAphB                      152.00    int16
VAphC                      144.00    int16
VA_SF                           0    sunssf
VAR                       -563.00    int16
VARphA                    -280.00    int16
VARphB                    -141.00    int16
VARphC                    -140.00    int16
VAR_SF                          0    sunssf
PF                           3.00    int16
PFphA                        3.70    int16
PFphB                       28.15    int16
PFphC                      -22.84    int16
PF_SF                          -2    sunssf
TotWhExp              12142959.00    acc32
TotWhExpPhA            5083589.00    acc32
TotWhExpPhB            5031494.00    acc32
TotWhExpPhC            4536532.00    acc32
TotWhImp               2051098.00    acc32
TotWhImpPhA            2118733.00    acc32
TotWhImpPhB             965564.00    acc32
TotWhImpPhC            1475456.00    acc32
TotWh_SF                        0    sunssf
TotVAhExp                     NaN    acc32
TotVAhExpPhA                  NaN    acc32
TotVAhExpPhB                  NaN    acc32
TotVAhExpPhC                  NaN    acc32
TotVAhImp                     NaN    acc32
TotVAhImpPhA                  NaN    acc32
TotVAhImpPhB                  NaN    acc32
TotVAhImpPhC                  NaN    acc32
TotVAh_SF                     n/a    sunssf
TotVArhImpQ1                  NaN    acc32
TotVArhImpQ1PhA               NaN    acc32
TotVArhImpQ1PhB               NaN    acc32
TotVArhImpQ1PhC               NaN    acc32
TotVArhImpQ2                  NaN    acc32
TotVArhImpQ2PhA               NaN    acc32
TotVArhImpQ2PhB               NaN    acc32
TotVArhImpQ2PhC               NaN    acc32
TotVArhExpQ3                  NaN    acc32
TotVArhExpQ3PhA               NaN    acc32
TotVArhExpQ3PhB               NaN    acc32
TotVArhExpQ3PhC               NaN    acc32
TotVArhExpQ4                  NaN    acc32
TotVArhExpQ4PhA               NaN    acc32
TotVArhExpQ4PhB               NaN    acc32
TotVArhExpQ4PhC               NaN    acc32
TotVArh_SF                    n/a    sunssf
Evt                    0x00000000    bitfield32
```

</details>

Soweit so gut, wenn ich jedoch den run-Befehl verwende, bekomme ich folgende Fehlermeldung: 

```
/ # mbmd run -v -d SOLAREDGE:1@192.168.100.24:502
2020/05/31 17:15:23 mbmd unknown version (unknown commit)
2020/05/31 17:15:23 config: creating TCP connection for 192.168.100.24:502
2020/05/31 17:15:23 config: declared device SOLAREDGE:1
2020/05/31 17:15:23 httpd: starting api at 0.0.0.0:8080
2020/05/31 17:15:23 initializing device SOLAREDGE1.1 failed: sunspec: multiple devices found
```
Wobei der Fehler immer wieder in einer Schleife kommt. 

Weiterhin habe ich versucht, per scan die angeschlossenen Geräte zu funden. 
Findet das Programm nichts: `2020/05/31 16:46:20 found 0 active devices:`

@andig hast du eine Idee woran es liegen könnte?

andig, 2020-05-31:
Interessant. Dein SE besteht aus 2 Devices- das habe ich bisher nirgendwo gesehen. Jetzt gibts 2 Möglichkeiten: Devices > 1 ignorieren statt Fehler. Dann hättest Du keinen Zugriff auf das `WattNode` Device. Alternativ überlegen wie sich das einbauen ließe, z.B. als mehrere Devices obwohl nur eines konfiguriert ist. Da müsste ich etwas basteln. Falls Du mir Remote Zugang (Port freigeben, ZeroTier, ssh) geben könntest meld Dich gerne per PM (cpuidle@gmx.de).

andig, 2020-05-31:
PS:

> Findet das Programm nichts: 2020/05/31 16:46:20 found 0 active devices:

Das dürfte ein Folgefehler sein.

BMOD89, 2020-06-15:
Danke für dein Feedback. Ich hatte Lust selbst etwas zu basteln. 

In #132 ist mein Änderungsvorschläg. 
Meine Umsetzung ist etwas pragmatisch. Ist es besser in cmd/run.go weitere Devices anzulegen? 

In der Weboberfläche sind nicht alle Werte enthalten. Vermutlich liegt es daran, dass mehrere Elemente mit dem gleichen Namen existieren. Ich möchte das Programm zum Befüllen einer InfluxDB verwenden. Der Test damit steht noch aus

Meine Go-Kenntnisse sind noch sehr ausbaufähig, daher gerne Änderungswünsche.

andig, 2020-06-16:
Danke für den Vorschlag. Ich kann es auf die Schnelle leider nicht testen. Schöne wäre ja wenn alle Devices dann auch einzeln auftauchen würden- das ist mit Deinem Vorschlag noch nicht der Fall, oder?

BMOD89, 2020-06-17:
Ja das ist nicht der Fall und ich gebe dir Recht es wäre schöner, wenn alle Devices auftauchen. 

In der Funktion `confHandler.CreateDeviceFromSpec` werden die Devices angelegt. An der Stelle ist jedoch noch nicht bekannt, dass sich hinter dem angegebenen Device quasi zwei Devices stecken. Ich sehe zwei Möglichkeiten das zu umgehen. Entweder man erweitert den Übergabeparameter so, dass mehrere Sub-Devices angegeben werden können oder man versucht die eigentliche Initialisierung des Devices (wo abgefragt wird wie viele Devices es tatsächlich gibt) dorhin umzuziehen. 
Oder hast du noch ne weitere Idee?

andig, 2020-06-17:
Irgendwie so... Evtl könnte SunSpec ein SubDevices interface anbieten über das sich dann weitere Geräte anlegen liessen? Dazu vllt eine CreateSubDevive Funktion über die sich dann die Nummer des Subdevices angeben liessen? Über Kommandozeile könnte man das dann aber auch nicht steuern. 

Ich glaube Dein jetziger PR funktioniert eher zufällig. Z.B. könnte sowohl der WR (dev 0) als auch ein am WR angeschlossener Modbus Zähler (Dev 1) jeweils Power liefern. In einem Device gäbe es den Wert aber auch nir 1x im UI, mqtt etc.

magpern, 2020-07-30:
> Interesting. Your SE consists of 2 devices - I have never seen that anywhere, even if the spec allows it.
> 
> Now there are 2 options: Ignore devices> 1 instead of errors. Then you would not have access to the `WattNode`device. Alternatively, consider how this could be installed, e.g. as multiple devices although only one is configured. I would have to make something there. If you could give me remote access (open port, ZeroTier, ssh) please contact me via PM ( [cpuidle@gmx.de](mailto:cpuidle@gmx.de) ).

I have a SolarEdge with an Inepro Pro380 connected.
I have a spare raspberry pi I could grant you access to if you would like to fix the solaredge problem with meter connected?

Soon, I will have access to two pro380 connected to the solaredge.

Any idea how I could give you access to it, without exposing me totally to the internet?

andig, 2020-07-31:
@magpern please pm me at cpuidle@gmx.de

thomo, 2020-08-02:
Same problem here (SE15K with SE Energy Meter)
```
racknode:/usr/local/bin# mbmd inspect -d SUNS:1@inverter.thomo.de:1502
2020/08/02 16:41:09 config: creating TCP connection for inverter.thomo.de:1502
--------- Model 1 common ---------
Mn         SolarEdge     string
Md    SE15K-RW0T0BNN4    string
Opt                      string
Vr           0004.000    string
SN           7E08C3F2    string
DA               1.00    uint16
Pad            0x0067    pad
--------- Model 103 inverter ---------
A                   46.42    uint16
AphA                15.46    uint16
AphB                15.47    uint16
AphC                15.48    uint16
A_SF                   -2    sunssf
PPVphAB            420.70    uint16
PPVphBC            421.70    uint16
PPVphCA            419.30    uint16
PhVphA             242.70    uint16
PhVphB             243.60    uint16
PhVphC             242.20    uint16
V_SF                   -1    sunssf
W                 8632.00    int16
W_SF                    0    sunssf
Hz                  50.00    uint16
Hz_SF                  -2    sunssf
VA               11308.00    int16
VA_SF                   0    sunssf
VAr              -7305.00    int16
VAr_SF                  0    sunssf
PF                 -76.32    int16
PF_SF                  -2    sunssf
WH             1188574.00    acc32
WH_SF                   0    sunssf
DCA                 11.72    uint16
DCA_SF                 -3    sunssf
DCV                747.40    uint16
DCV_SF                 -1    sunssf
DCW               8763.00    int16
DCW_SF                  0    sunssf
TmpCab                n/a    int16
TmpSnk              54.74    int16
TmpTrns               n/a    int16
TmpOt                 n/a    int16
Tmp_SF                 -2    sunssf
St                      4    enum16
StVnd                   0    enum16
Evt1                  n/a    bitfield32
Evt2                  n/a    bitfield32
EvtVnd1        0x00000000    bitfield32
EvtVnd2               n/a    bitfield32
EvtVnd3               n/a    bitfield32
EvtVnd4        0x00000000    bitfield32
--------- Model 1 common ---------
Mn          SolarEdge    string
Md   SE-MTR-3Y-400V-A    string
Opt          Export+I    string
Vr                 72    string
SN          606384567    string
DA               2.00    uint16
Pad            0x00cb    pad
--------- Model 203 ac_meter ---------
A                           42.80    int16
AphA                        13.40    int16
AphB                        14.50    int16
AphC                        14.90    int16
A_SF                           -1    sunssf
PhV                        242.71    int16
PhVphA                     242.71    int16
PhVphB                     243.13    int16
PhVphC                     240.41    int16
PPV                          0.00    int16
PhVphAB                      0.00    int16
PhVphBC                      0.00    int16
PhVphCA                      0.00    int16
V_SF                           -2    sunssf
Hz                          50.00    int16
Hz_SF                          -2    sunssf
W                         7479.00    int16
WphA                      2054.00    int16
WphB                      2705.00    int16
WphC                      2719.00    int16
W_SF                            0    sunssf
VA                       10210.00    int16
VAphA                     3182.00    int16
VAphB                     3494.00    int16
VAphC                     3566.00    int16
VA_SF                           0    sunssf
VAR                       6950.00    int16
VARphA                    2430.00    int16
VARphB                    2211.00    int16
VARphC                    2308.00    int16
VAR_SF                          0    sunssf
PF                         -72.20    int16
PFphA                      -63.49    int16
PFphB                      -77.20    int16
PFphC                      -75.90    int16
PF_SF                          -2    sunssf
TotWhExp               1012461.00    acc32
TotWhExpPhA             319508.00    acc32
TotWhExpPhB             351443.00    acc32
TotWhExpPhC             341508.00    acc32
TotWhImp                122054.00    acc32
TotWhImpPhA              60765.00    acc32
TotWhImpPhB              32664.00    acc32
TotWhImpPhC              28625.00    acc32
TotWh_SF                        0    sunssf
TotVAhExp                     NaN    acc32
TotVAhExpPhA                  NaN    acc32
TotVAhExpPhB                  NaN    acc32
TotVAhExpPhC                  NaN    acc32
TotVAhImp                     NaN    acc32
TotVAhImpPhA                  NaN    acc32
TotVAhImpPhB                  NaN    acc32
TotVAhImpPhC                  NaN    acc32
TotVAh_SF                     n/a    sunssf
TotVArhImpQ1                  NaN    acc32
TotVArhImpQ1PhA               NaN    acc32
TotVArhImpQ1PhB               NaN    acc32
TotVArhImpQ1PhC               NaN    acc32
TotVArhImpQ2                  NaN    acc32
TotVArhImpQ2PhA               NaN    acc32
TotVArhImpQ2PhB               NaN    acc32
TotVArhImpQ2PhC               NaN    acc32
TotVArhExpQ3                  NaN    acc32
TotVArhExpQ3PhA               NaN    acc32
TotVArhExpQ3PhB               NaN    acc32
TotVArhExpQ3PhC               NaN    acc32
TotVArhExpQ4                  NaN    acc32
TotVArhExpQ4PhA               NaN    acc32
TotVArhExpQ4PhB               NaN    acc32
TotVArhExpQ4PhC               NaN    acc32
TotVArh_SF                    n/a    sunssf
Evt                    0x00000000    bitfield32
```

PR #138 (2020-08-04): Allow accessing subdevices by specifying subdevice id as slave.subdevice
Fix #128, fix #135, replaces #132.

/cc @BMOD89 @magpern @grmat
closed 2020-08-04 by andig