volkszaehler/mbmd #134: SolarEdge 2000M results in panic

schenlap, 2020-07-19:
```
./mbmd run -v -d SOLAREDGE:1@10.0.0.207:502'
2020/07/19 09:00:41 mbmd unknown version (unknown commit)
2020/07/19 09:00:41 config: creating TCP connection for 10.0.0.207:502
2020/07/19 09:00:41 config: using /home/foobar/go/src/github.com/volkszaehler/mbmd/mbmd.yaml
2020/07/19 09:00:41 config: declared device SOLAREDGE:1
2020/07/19 09:00:41 httpd: starting api at 0.0.0.0:8080
2020/07/19 09:00:41 initialized device SolarEdge 1.1: {SolarEdge  SE2000M-RWK02BNN  0004.000 7C00151A}
panic: no such block

goroutine 50 [running]:
github.com/andig/gosunspec/impl.(*model).MustBlock(0xc00011a2c0, 0x1, 0x1, 0x118e900)
        /home/foobar/go/pkg/mod/github.com/andig/gosunspec@v0.0.0-20200429133549-3cf6a82fed9c/impl/model.go:34 +0x8f
github.com/volkszaehler/mbmd/meters/sunspec.(*SunSpec).Query(0xc000351260, 0xc6a840, 0xc000191ee0, 0xc000130130, 0xc000024940, 0xd, 0xc000505c10, 0x4e15fa)
        /home/foobar/go/src/github.com/volkszaehler/mbmd/meters/sunspec/sunspec.go:316 +0x5d4
github.com/volkszaehler/mbmd/server.(*Handler).queryDevice(0xc000392750, 0xc649e0, 0xc000036080, 0xc0001a8ae0, 0xc0001a8a80, 0x1, 0xc653a0, 0xc000351260)
        /home/foobar/go/src/github.com/volkszaehler/mbmd/server/handler.go:132 +0x11c
github.com/volkszaehler/mbmd/server.(*Handler).Run.func1(0xa60c01, 0xc653a0, 0xc000351260)
        /home/foobar/go/src/github.com/volkszaehler/mbmd/server/handler.go:78 +0x1d0
github.com/volkszaehler/mbmd/meters.(*Manager).All(0xc000392758, 0xc000505eb8)
        /home/foobar/go/src/github.com/volkszaehler/mbmd/meters/manager.go:41 +0x95
github.com/volkszaehler/mbmd/server.(*Handler).Run(0xc000392750, 0xc649e0, 0xc000036080, 0xc0001a8ae0, 0xc0001a8a80)
        /home/foobar/go/src/github.com/volkszaehler/mbmd/server/handler.go:49 +0x8a
github.com/volkszaehler/mbmd/server.(*QueryEngine).Run.func1(0xdf8475800, 0xc649e0, 0xc000036080, 0xc0001a8ae0, 0xc0001a8a80, 0xc0002b8000, 0xc000392750)
        /home/foobar/go/src/github.com/volkszaehler/mbmd/server/queryengine.go:93 +0xbb
created by github.com/volkszaehler/mbmd/server.(*QueryEngine).Run
        /home/foobar/go/src/github.com/volkszaehler/mbmd/server/queryengine.go:87 +0x188
```

inspect works ok

```
./mbmd inspect -d SOLAREDGE:1@10.0.0.207:502
2020/07/19 09:01:41 config: creating TCP connection for 10.0.0.207:502
--------- Model 1 common ---------
Mn         SolarEdge     string
Md   SE2000M-RWK02BNN    string
Opt                      string
Vr           0004.000    string
SN           7C00151A    string
DA               1.00    uint16
Pad            0x0065    pad
--------- Model 101 inverter ---------
A                    2.08    uint16
AphA                 2.08    uint16
AphB                  n/a    uint16
AphC                  n/a    uint16
A_SF                   -2    sunssf
PPVphAB            231.30    uint16
PPVphBC               n/a    uint16
PPVphCA               n/a    uint16
PhVphA                n/a    uint16
PhVphB                n/a    uint16
PhVphC                n/a    uint16
V_SF                   -1    sunssf
W                  376.50    int16
W_SF                   -1    sunssf
Hz                  50.00    uint16
Hz_SF                  -3    sunssf
VA                 481.40    int16
VA_SF                  -1    sunssf
VAr                299.99    int16
VAr_SF                 -2    sunssf
PF                  78.21    int16
PF_SF                  -2    sunssf
WH             2614948.00    acc32
WH_SF                   0    sunssf
DCA                  5.48    uint16
DCA_SF                 -3    sunssf
DCV                 69.80    uint16
DCV_SF                 -1    sunssf
DCW                382.20    int16
DCW_SF                 -1    sunssf
TmpCab                n/a    int16
TmpSnk              46.98    int16
TmpTrns               n/a    int16
TmpOt                 n/a    int16
Tmp_SF                 -2    sunssf
St                      4    enum16
StVnd                   0    enum16
Evt1                  n/a    bitfield32
Evt2                  n/a    bitfield32
EvtVnd1        0x00000000    bitfield32
EvtVnd2               n/a    bitfield32
EvtVnd3               n/a    bitfield32
EvtVnd4        0x00000000    bitfield32
```

andig, 2020-07-19:
Vitte aktualisieren, sollte gestern früh behoben sein.

schenlap, 2020-07-20:
Ja, danke, läuft jetzt