volkszaehler/mbmd #135: Maintain device order

grmat, 2020-07-28:
Hi,

I've set up mbmd with two SUNSPEC devices, a KOSTAL inverter (plenticore) and a smart meter device (ksem).

The corresponding device config looks like that:

```
adapters:
- device: 192.168.1.5:502
- device: 192.168.1.6:1502

devices:
- name: ksem
  type: SUNSPEC
  id: 71
  adapter: 192.168.1.5:502
- name: plenticore
  type: SUNSPEC
  id: 71
  adapter: 192.168.1.6:1502
```

The setup worked as expected, the KSEM ended up as `KOSTAL1.71`, the plenticore as `KOSTAL2.71` and `mbmd` started logging to influxdb. However, after a short downtime and restart, I've noticed that the devices got mixed up. The smart meter (KSEM) was now `KOSTAL2.71` and the inverter `KOSTAL1.71`. I've noticed this after a few hours by looking at weird graphs. After restarting `mbmd`, the order was restored again.

Corresponding stdout:

```
# second run

2020/07/28 15:27:46 mbmd unknown version (unknown commit)
2020/07/28 15:27:46 config: using /etc/mbmd/config.yaml
2020/07/28 15:27:46 config: creating TCP connection for 192.168.1.5:502
2020/07/28 15:27:46 config: creating TCP connection for 192.168.1.6:1502
2020/07/28 15:27:46 config: declared device SUNSPEC:71
2020/07/28 15:27:46 config: declared device SUNSPEC:71
2020/07/28 15:27:46 httpd: starting api at 0.0.0.0:8080
2020/07/28 15:27:46 initialized device KOSTAL2.71: {KOSTAL KOSTAL Smart Ene  1.2.1 xxx}
2020/07/28 15:27:46 sunspec: device opened partially
2020/07/28 15:27:46 initialized device KOSTAL1.71: {KOSTAL PLENTICORE plus1          01.15.04 xxx}
2020/07/28 16:36:46 received signal - stopping
2020/07/28 16:36:46 stopped

# third run

2020/07/28 16:36:50 mbmd unknown version (unknown commit)
2020/07/28 16:36:50 config: using /etc/mbmd/config.yaml
2020/07/28 16:36:50 config: creating TCP connection for 192.168.1.5:502
2020/07/28 16:36:50 config: creating TCP connection for 192.168.1.6:1502
2020/07/28 16:36:50 config: declared device SUNSPEC:71
2020/07/28 16:36:50 config: declared device SUNSPEC:71
2020/07/28 16:36:50 httpd: starting api at 0.0.0.0:8080
2020/07/28 16:36:50 initialized device KOSTAL1.71: {KOSTAL KOSTAL Smart Ene  1.2.1 xxx}
2020/07/28 16:36:50 sunspec: device opened partially
2020/07/28 16:36:50 initialized device KOSTAL2.71: {KOSTAL PLENTICORE plus1          01.15.04 xxx}
```

I think the order should be maintained, so data doesn't get messed up.

Side question: is "device partially opened" something to be worried about?

andig, 2020-07-28:
Partially opened indicates a an error due to sunspec violation by Kostal- a firmware bug that they apparently don‘t intend to fix. As for device order: I‘ve seen this before but was nit able to reproduce. Out of curiosity: what go version are you working with?

grmat, 2020-07-28:
> a firmware bug that they apparently don‘t intend to fix

OK, so you have already contacted them?

go 1.14.6

andig, 2020-07-28:
A user did. Since the problem is still there 2 years later...

andig, 2020-07-30:
The fix ware introduced here... https://github.com/volkszaehler/mbmd/pull/5

PR #138 (2020-08-04): Allow accessing subdevices by specifying subdevice id as slave.subdevice
Fix #128, fix #135, replaces #132.

/cc @BMOD89 @magpern @grmat
closed 2020-08-04 by andig