volkszaehler/mbmd #141: Documentation how to get data to Volkszaehler

StefanSchoof, 2020-08-03:
I like to get the measurements that mbmd reads into the Volkszaehler middleware. I found no documentation how this can or which is the recommended way of doing so.

Is there any documentation or examples how this can achieved?

andig, 2020-08-04:
One thing you could do is build something using node-red. As I‘m currently not sure what the most efficient way is for adapting mqtt to rest api I‘d suggest to ask on the volkszähler mailing list, see https://wiki.volkszaehler.org/contact

StefanSchoof, 2020-08-04:
Thanks did that: https://demo.volkszaehler.org/pipermail/volkszaehler-users/2020-August/014820.html

andig, 2020-08-04:
You could also give https://github.com/andig/ingress a try. I‘m not actively maintaining it at the moment and it does not have a development roadmap but it should work.

StefanSchoof, 2020-08-19:
I wrote my way in https://demo.volkszaehler.org/pipermail/volkszaehler-users/2020-August/014834.html.
Is this something that be as example in the Readme or is this a problem that nobody else has? If yes I can try to provide an PR.

andig, 2020-08-20:
Probably quite specific? Could add a PR to readme of link a description from readme.

closed 2021-03-07 by andig