volkszaehler/mbmd #142: Wrong sub device values

thomo, 2020-08-05:
version: mbmd 0.12 (0d6f45f)
SE inverter with power meter

The current data extracted from the power meter seams wrong - even on a sunny day 4492.0 A looks incredible. The web page also changes the current values between the reasonable ones (~19A) and the wrong (~4492A) on all three lines (L1/L2/L3). 

```
root@racknode:/usr/local/bin# mbmd run -a inverter.thomo.de:1502 -d SE:1.0 -d SE:1.1 -v
2020/08/05 15:09:37 mbmd 0.12 (0d6f45f)
2020/08/05 15:09:37 config: creating TCP connection for inverter.thomo.de:1502
2020/08/05 15:09:37 config: declared device SE:1.0
2020/08/05 15:09:37 config: declared device SE:1.1
2020/08/05 15:09:37 httpd: starting api at 0.0.0.0:8080
2020/08/05 15:09:38 initialized device SE1-1: {SE SolarEdge SE15K-RW0T0BNN4  0004.000 7E08C3F2 0}
2020/08/05 15:09:38 device SE1-1 L1: 0.0V 0.0A 0.0W 0.0cos | L2: 0.0V 19.5A 0.0W 0.0cos | L3: 0.0V 0.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 15:09:38 device SE1-1 L1: 244V 0.0A 0.0W 0.0cos | L2: 0.0V 19.5A 0.0W 0.0cos | L3: 0.0V 0.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 15:09:38 device SE1-1 L1: 244V 0.0A 0.0W 0.0cos | L2: 246V 19.5A 0.0W 0.0cos | L3: 0.0V 0.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 15:09:38 device SE1-1 L1: 244V 0.0A 0.0W 0.0cos | L2: 246V 19.5A 0.0W 0.0cos | L3: 0.0V 0.0A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1 L1: 244V 0.0A 0.0W 0.0cos | L2: 246V 19.5A 0.0W 0.0cos | L3: 245V 0.0A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1 L1: 244V 19.5A 0.0W 0.0cos | L2: 246V 19.5A 0.0W 0.0cos | L3: 245V 0.0A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1 L1: 244V 19.5A 0.0W 0.0cos | L2: 246V 19.5A 0.0W 0.0cos | L3: 245V 19.4A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 initialized device SE1-1.1: {SE SolarEdge SE-MTR-3Y-400V-A Export+I 72 606384567 1}
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 0.0A 0.0W 0.0cos | L2: 0.0V 0.0A 0.0W 0.0cos | L3: 0.0V 0.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 0.0A 0.0W 0.0cos | L2: 0.0V 18.2A 0.0W 0.0cos | L3: 0.0V 0.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 0.0A 0.0W 0.0cos | L2: 0.0V 18.2A 0.0W 0.0cos | L3: 0.0V 18.7A 0.0W 0.0cos | 0.0Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 0.0A 0.0W 0.0cos | L2: 0.0V 4492.0A 0.0W 0.0cos | L3: 0.0V 18.7A 0.0W 0.0cos | 0.0Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 0.0A 0.0W 0.0cos | L2: 0.0V 4492.0A 0.0W 0.0cos | L3: 0.0V 4583.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 0.0A 3071W 0.0cos | L2: 0.0V 4492.0A 0.0W 0.0cos | L3: 0.0V 4583.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 0.0A 3071W 0.0cos | L2: 0.0V 4492.0A 0.0W 0.0cos | L3: 0.0V 4583.0A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 0.0A 3071W 0.0cos | L2: 0.0V 4492.0A 3062W 0.0cos | L3: 0.0V 4583.0A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 18.6A 3071W 0.0cos | L2: 0.0V 4492.0A 3062W 0.0cos | L3: 0.0V 4583.0A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 18.6A 3071W -66.39cos | L2: 0.0V 4492.0A 3062W 0.0cos | L3: 0.0V 4583.0A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 18.6A 3071W -66.39cos | L2: 0.0V 4492.0A 3062W 0.0cos | L3: 0.0V 4583.0A 3109W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 18.6A 3071W -66.39cos | L2: 0.0V 4492.0A 3062W -67.69cos | L3: 0.0V 4583.0A 3109W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 18.6A 3071W -66.39cos | L2: 0.0V 4492.0A 3062W -67.69cos | L3: 0.0V 4583.0A 3109W -67.00cos | 50Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 18.6A 3071W -66.39cos | L2: 246V 4492.0A 3062W -67.69cos | L3: 0.0V 4583.0A 3109W -67.00cos | 50Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 18.6A 3071W -66.39cos | L2: 246V 4492.0A 3062W -67.69cos | L3: 243V 4583.0A 3109W -67.00cos | 50Hz
2020/08/05 15:09:38 device SE1-1.1 L1: 244V 4569.0A 3071W -66.39cos | L2: 246V 4492.0A 3062W -67.69cos | L3: 243V 4583.0A 3109W -67.00cos | 50Hz
2020/08/05 15:09:38 device SE1-1 L1: 244V 19.5A 0.0W 0.0cos | L2: 246V 19.5A 0.0W 0.0cos | L3: 245V 19.4A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1 L1: 243V 19.5A 0.0W 0.0cos | L2: 246V 19.5A 0.0W 0.0cos | L3: 245V 19.4A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1 L1: 243V 19.5A 0.0W 0.0cos | L2: 247V 19.5A 0.0W 0.0cos | L3: 245V 19.4A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1 L1: 243V 19.5A 0.0W 0.0cos | L2: 247V 19.5A 0.0W 0.0cos | L3: 245V 19.4A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1 L1: 243V 19.4A 0.0W 0.0cos | L2: 247V 19.5A 0.0W 0.0cos | L3: 245V 19.4A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1 L1: 243V 19.4A 0.0W 0.0cos | L2: 247V 19.5A 0.0W 0.0cos | L3: 245V 19.4A 0.0W 0.0cos | 50Hz
2020/08/05 15:09:38 device SE1-1 L1: 243V 19.4A 0.0W 0.0cos | L2: 247V 19.5A 0.0W 0.0cos | L3: 245V 19.4A 0.0W 0.0cos | 50Hz
```

(At first start of mbmd with the sub devices the web page didn't show any values, only the about sections - but I can not reproduce it.)

andig, 2020-08-05:
@thomo that sounds indeed wrong while the voltage looks good. Can you find this was not a problem with 0.11? 
Could you kindly pm me for remote access at cpuidle@gmx.de?

andig, 2020-08-05:
@thomo do you get the same results when reading only the first subdevice?

thomo, 2020-08-05:
@andig Same when only the power meter is read. 
```
root@racknode:/usr/local/bin# mbmd run -a inverter.thomo.de:1502 -d SE:1.1 -v
2020/08/05 21:30:50 mbmd 0.12 (0d6f45f)
2020/08/05 21:30:50 config: creating TCP connection for inverter.thomo.de:1502
2020/08/05 21:30:50 config: declared device SE:1.1
2020/08/05 21:30:50 httpd: starting api at 0.0.0.0:8080
2020/08/05 21:30:50 initialized device SE1-1.1: {SE SolarEdge SE-MTR-3Y-400V-A Export+I 72 606384567 1}
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 0.0A 0.0W 0.0cos | L2: 0.0V 0.0A 0.0W 0.0cos | L3: 0.0V 0.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 0.0A -191W 0.0cos | L2: 0.0V 0.0A 0.0W 0.0cos | L3: 0.0V 0.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 0.0A -191W 0.0cos | L2: 0.0V 0.0A -310W 0.0cos | L3: 0.0V 0.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 0.0A -191W 0.0cos | L2: 0.0V 403.0A -310W 0.0cos | L3: 0.0V 0.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 237.0A -191W 0.0cos | L2: 0.0V 403.0A -310W 0.0cos | L3: 0.0V 0.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 237.0A -191W 0.0cos | L2: 0.0V 403.0A -310W 0.0cos | L3: 0.0V 170.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 237.0A -191W 0.0cos | L2: 0.0V 403.0A -310W 70.59cos | L3: 0.0V 170.0A 0.0W 0.0cos | 0.0Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 237.0A -191W 0.0cos | L2: 0.0V 403.0A -310W 70.59cos | L3: 0.0V 170.0A 0.0W 0.0cos | 50Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 237.0A -191W 0.0cos | L2: 238V 403.0A -310W 70.59cos | L3: 0.0V 170.0A 0.0W 0.0cos | 50Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 237.0A -191W 0.0cos | L2: 238V 403.0A -310W 70.59cos | L3: 238V 170.0A 0.0W 0.0cos | 50Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 237.0A -191W 0.0cos | L2: 238V 403.0A -310W 70.59cos | L3: 238V 0.7A 0.0W 0.0cos | 50Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 237.0A -191W 75.19cos | L2: 238V 403.0A -310W 70.59cos | L3: 238V 0.7A 0.0W 0.0cos | 50Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 237.0A -191W 75.19cos | L2: 238V 403.0A -310W 70.59cos | L3: 238V 0.7A -73W 0.0cos | 50Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 1.0A -191W 75.19cos | L2: 238V 403.0A -310W 70.59cos | L3: 238V 0.7A -73W 0.0cos | 50Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 1.0A -191W 75.19cos | L2: 238V 1.8A -310W 70.59cos | L3: 238V 0.7A -73W 0.0cos | 50Hz
2020/08/05 21:30:50 device SE1-1.1 L1: 237V 1.0A -191W 75.19cos | L2: 238V 1.8A -310W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 1.0A -191W 75.19cos | L2: 238V 438.0A -310W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 1.0A -191W 75.19cos | L2: 238V 438.0A -354W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 1.0A -191W 75.19cos | L2: 238V 438.0A -354W 70.59cos | L3: 238V 170.0A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 1.0A -191W 75.19cos | L2: 238V 438.0A -354W 70.59cos | L3: 238V 170.0A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 235.0A -191W 75.19cos | L2: 238V 438.0A -354W 70.59cos | L3: 238V 170.0A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 235.0A -191W 75.19cos | L2: 238V 438.0A -354W 70.59cos | L3: 238V 170.0A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 235.0A -191W 75.19cos | L2: 238V 438.0A -354W 70.59cos | L3: 238V 170.0A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 235.0A -191W 75.19cos | L2: 238V 438.0A -354W 70.59cos | L3: 238V 170.0A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 235.0A -191W 75.19cos | L2: 238V 438.0A -354W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 235.0A -191W 75.19cos | L2: 238V 438.0A -354W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 235.0A -191W 75.19cos | L2: 238V 438.0A -354W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 1.0A -191W 75.19cos | L2: 238V 438.0A -354W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 1.0A -191W 75.19cos | L2: 238V 1.8A -354W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 1.0A -191W 75.19cos | L2: 238V 1.8A -354W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 1.0A -191W 75.19cos | L2: 238V 1.8A -354W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:51 device SE1-1.1 L1: 237V 1.0A -189W 75.19cos | L2: 238V 1.8A -354W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:52 device SE1-1.1 L1: 237V 1.0A -189W 75.19cos | L2: 238V 410.0A -354W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:52 device SE1-1.1 L1: 237V 1.0A -189W 75.19cos | L2: 238V 410.0A -319W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:52 device SE1-1.1 L1: 237V 234.0A -189W 75.19cos | L2: 238V 410.0A -319W 70.59cos | L3: 238V 0.7A -73W 41.10cos | 50Hz
2020/08/05 21:30:52 device SE1-1.1 L1: 237V 234.0A -189W 75.19cos | L2: 238V 410.0A -319W 70.59cos | L3: 238V 170.0A -73W 41.10cos | 50Hz
2020/08/05 21:30:52 device SE1-1.1 L1: 237V 234.0A -189W 75.19cos | L2: 238V 410.0A -319W 70.59cos | L3: 238V 170.0A -73W 41.10cos | 50Hz
[... cut ...]
2020/08/05 21:30:53 device SE1-1.1 L1: 237V 232.0A -185W 75.19cos | L2: 238V 421.0A -333W 70.59cos | L3: 238V 170.0A -72W 41.10cos | 50Hz
2020/08/05 21:30:53 device SE1-1.1 L1: 237V 232.0A -185W 75.19cos | L2: 238V 421.0A -333W 70.59cos | L3: 238V 0.7A -72W 41.10cos | 50Hz
2020/08/05 21:30:53 device SE1-1.1 L1: 237V 232.0A -185W 75.19cos | L2: 238V 421.0A -333W 70.59cos | L3: 238V 0.7A -72W 41.10cos | 50Hz
^C2020/08/05 21:30:54 received signal - stopping
```

When only the inverter is read all values seam to be ok.

andig, 2020-08-05:
@thomo could we get in Touch at cpuidle@gmx.de?

thomo, 2020-08-09:
With the last changes (https://github.com/volkszaehler/mbmd/pull/148/files) it seams to work now.

Thanks a lot.

closed 2020-08-09 by andig