volkszaehler/mbmd #144: SDM120C found in scan, but cannot collect data.

dproetel, 2020-08-05:
I try to read data from my SDM120C. with
`/mbmd scan -a /dev/ttyUSB0 -b 2400
2020/08/05 21:53:15 config: creating RTU connection for /dev/ttyUSB0 (2400baud, 8N1)
2020/08/05 21:53:15 starting bus scan on /dev/ttyUSB0
2020/08/05 21:53:15 device 1: SDM type device found, VoltageL1: 230.80
2020/08/05 21:53:15 device 2: SDM type device found, VoltageL1: 229.75
`
device 1 is the SDM120c. Device 2 is a 630 and can be read perfectly.
but
./mbmd run -a /dev/ttyUSB0  -d sdm:1 -b 2400   -v
2020/08/05 21:54:08 mbmd 0.12 (0d6f45f)
2020/08/05 21:54:08 config: creating RTU connection for /dev/ttyUSB0 (2400baud, 8N1)
2020/08/05 21:54:08 config: declared device SDM:1.0
2020/08/05 21:54:08 httpd: starting api at 0.0.0.0:8080
2020/08/05 21:54:08 initialized device SDM1-1: {SDM SDM Eastron SDM630    0}
2020/08/05 21:54:08 device SDM1-1 did not respond (1/3)
2020/08/05 21:54:09 device SDM1-1 did not respond (2/3)
2020/08/05 21:54:11 device SDM1-1 did not respond (3/3)
Says no response.

SOme parameters which I take wrong?
with 

sdm120c  -a1  -PN -S1    /dev/ttyUSB0  
Voltage: 230.50 V 

I can read at least the Voltage.

THanks in advance.

andig, 2020-08-06:
According to https://github.com/volkszaehler/mbmd/blob/master/docs/mbmd_run.md there are multiple SDM devices with different readings. Could you try SDM220 instead of SDM? The detection cannot distinguish these by looking at a single voltage only.

dproetel, 2020-08-06:
Hi,
I tried the SDM220 SDM230 and SDM72, but none of them respond.

dproetel, 2020-08-06:
If I do it with --raw the non working SDN120 shows:
`2020/08/06 07:20:51 initialized device SDM2201-1: {SDM220 SDM220 Eastron SDM220    0}
2020/08/06 07:20:51 modbus: send 01 04 01 58 00 02 f1 e4
2020/08/06 07:20:51 modbus: recv 01 04 04 3d 4c cc cd a3 6a
2020/08/06 07:20:51 modbus: send 01 04 00 4c 00 02 b0 1c
2020/08/06 07:20:51 modbus: recv 01 04 04 3d 4c cc cd a3 6a
2020/08/06 07:20:51 modbus: send 01 04 00 4e 00 02 11 dc
2020/08/06 07:20:52 modbus: recv 
2020/08/06 07:20:52 device SDM2201-1 did not respond (1/3)`

At least the SDM120C answers. Can I provide more data?
Using the This installation runs MBMD version 0.12 (compiled with go1.13) Version

andig, 2020-08-06:
@dproetel could you please format your markdown so it becomes readable? Done that for you above ;)
How is your SDM attached? How do you call mbmd? Are the comset parameters correct?

andig, 2020-08-06:
As far as I can see 120 is a superset of 220, so the 220 definition should work.

dproetel, 2020-08-06:
HI,
thanks for the formatting. The SDM is attached via a USB Modbus adapter to my Raspberry. With my SDM630 it worked well.  I use:
`mbmd scan -a /dev/ttyUSB0   -b 2400   -v  --raw  `
for the scanning. with 
```2020/08/06 08:27:09 config: creating RTU connection for /dev/ttyUSB0 (2400baud, 8N1)
2020/08/06 08:27:09 starting bus scan on /dev/ttyUSB0
2020/08/06 08:27:09 modbus: send 01 03 00 23 00 01 75 c0
2020/08/06 08:27:09 modbus: recv 01 03 02 00 00 b8 44
2020/08/06 08:27:09 modbus: send 01 03 0b d4 00 02 86 17
2020/08/06 08:27:09 modbus: recv 01 03 04 00 00 00 00 fe f3
2020/08/06 08:27:09 modbus: send 01 03 50 02 00 02 74 cb
2020/08/06 08:27:09 modbus: recv 
2020/08/06 08:27:09 modbus: send 01 03 4a 38 00 02 53 de
2020/08/06 08:27:10 modbus: recv 01 03 04 00 00 00 00 fa 33
2020/08/06 08:27:10 modbus: send 01 03 00 0e 00 02 a5 c8
2020/08/06 08:27:10 modbus: recv 
2020/08/06 08:27:10 modbus: send 01 04 00 00 00 02 71 cb
2020/08/06 08:27:10 modbus: recv 01 04 04 43 65 cc cd 6a 8a
2020/08/06 08:27:10 device 1: SDM type device found, VoltageL1: 229.80
```

die Abfrage starte ich mit:
```
mbmd run  -a /dev/ttyUSB0  -d sdm220:1 -b 2400   -v  --raw 
2020/08/06 08:36:13 mbmd 0.12 (0d6f45f)
2020/08/06 08:36:13 config: creating RTU connection for /dev/ttyUSB0 (2400baud, 8N1)
2020/08/06 08:36:13 config: declared device SDM220:1.0
2020/08/06 08:36:13 httpd: starting api at 0.0.0.0:8080
2020/08/06 08:36:13 initialized device SDM2201-1: {SDM220 SDM220 Eastron SDM220    0}
2020/08/06 08:36:13 modbus: send 01 04 00 00 00 02 71 cb
2020/08/06 08:36:13 modbus: recv 01 04 04 43 66 19 9a 85 e4
2020/08/06 08:36:13 modbus: send 01 04 00 06 00 02 91 ca
2020/08/06 08:36:13 modbus: recv 01 04 04 00 00 00 00 fb fc
2020/08/06 08:36:13 device SDM2201-1 did not respond (1/3)
2020/08/06 08:36:14 modbus: send 01 04 00 06 00 02 91 ca
2020/08/06 08:36:14 modbus: recv 01 04 04 00 00 00 00 fb 84
2020/08/06 08:36:14 modbus: send 01 04 00 48 00 02 f1 dd
2020/08/06 08:36:15 modbus: recv 
2020/08/06 08:36:15 device SDM2201-1 did not respond (2/3)
2020/08/06 08:36:16 modbus: send 01 04 00 48 00 02 f1 dd
2020/08/06 08:36:16 modbus: recv 01 04 04 00 00 00 00 fb 84
2020/08/06 08:36:16 modbus: send 01 04 00 4a 00 02 50 1d
2020/08/06 08:36:16 modbus: recv 01 04 04 00 00 00 00 fb fc
2020/08/06 08:36:16 device SDM2201-1 did not respond (3/3)
```
Ich sehe also das der SDM120 antwortet.

andig, 2020-08-06:
@dproetel could you check if c4d965bd1eac3bd5a854c08d8905f2d8ecd23dbd still works for you? Does 51eb643c23c63ea1d9599b96b9ba0308516d1494 work?

andig, 2020-08-06:
Could you please also use 009d3fe8c8efbd2b6f56b2f30f5906db76bc6f60 which adds error logging?

dproetel, 2020-08-07:
`c4d965b` doesnt work. 
51eb643  also not
with 009d3fe make build freezes at 
```go build -v -ldflags '-X "github.com/volkszaehler/mbmd/server.Version=009d3fe" -X "github.com/volkszaehler/mbmd/server.Commit=009d3fe"'
github.com/influxdata/line-protocol```

andig, 2020-08-07:
> with 009d3fe make build freezes at

Could you please retry 11c627cda760270028d817115420d82b9ee3cde8? It does compile for me:

```
❯ make                
rm -rf dist/
go install github.com/alvaroloes/enumer
go install github.com/mjibson/esc
Generating embedded assets
go generate ./...
golangci-lint run
Running testsuite
go test ./...
?       github.com/volkszaehler/mbmd    [no test files]
?       github.com/volkszaehler/mbmd/cmd        [no test files]
?       github.com/volkszaehler/mbmd/meters     [no test files]
ok      github.com/volkszaehler/mbmd/meters/rs485       (cached)
?       github.com/volkszaehler/mbmd/meters/sunspec     [no test files]
ok      github.com/volkszaehler/mbmd/server     (cached)
Version: 11c627c 2020-08-07_10:00:37
go build -v -ldflags '-X "github.com/volkszaehler/mbmd/server.Version=11c627c" -X "github.com/volkszaehler/mbmd/server.Commit=11c627c"'
```

If the older commits don't work I can only repeat my suggestion to git bisect or PM at cpuidle@gmx.de for physical device access for testing.

dproetel, 2020-08-07:
now the 009d3fe
<https://github.com/volkszaehler/mbmd/commit/009d3fe8c8efbd2b6f56b2f30f5906db76bc6f60>
works, more swap space was needed.
```` ./mbmd run  -a /dev/ttyUSB0  -d sdm220:1 -b 2400   -v  --raw

2020/08/07 11:17:40 mbmd 009d3fe (009d3fe)

2020/08/07 11:17:40 config: creating RTU connection for /dev/ttyUSB0
(2400baud, 8N1)

2020/08/07 11:17:40 config: declared device SDM220:1.0

2020/08/07 11:17:40 httpd: starting api at 0.0.0.0:8080

2020/08/07 11:17:40 initialized device SDM2201-1: {SDM220 SDM220 Eastron
SDM220    0}

2020/08/07 11:17:40 modbus: send 01 04 00 48 00 02 f1 dd

2020/08/07 11:17:40 modbus: recv 01 04 04 41 16 5a 1d f4 d5

2020/08/07 11:17:40 modbus: send 01 04 00 4a 00 02 50 1d

2020/08/07 11:17:40 modbus: recv 01 04 04 00 00 00 00 fb 84

2020/08/07 11:17:40 modbus: send 01 04 01 56 00 02 90 27

2020/08/07 11:17:41 modbus: recv 01 04 04 41 16 5a 1d f4 d5

2020/08/07 11:17:41 modbus: send 01 04 01 58 00 02 f1 e4

2020/08/07 11:17:41 modbus: recv 01 04 04 3f 82 b0 21 e2 e0

2020/08/07 11:17:41 device SDM2201-1 did not respond (1/3): read failed:
modbus: response crc '57570' does not match expected '24802'

2020/08/07 11:17:42 modbus: send 01 04 01 58 00 02 f1 e4

2020/08/07 11:17:42 modbus: recv 01 04 04 3f 82 b0 21 e2 60

2020/08/07 11:17:42 modbus: send 01 04 00 4c 00 02 b0 1c

2020/08/07 11:17:42 modbus: recv 01 04 04 3d cc cc cd a2 82

2020/08/07 11:17:42 modbus: send 01 04 00 4e 00 02 11 dc

2020/08/07 11:17:42 modbus: recv 01 04 04 3f 6b c6 a8 d4 52

2020/08/07 11:17:42 modbus: send 01 04 00 00 00 02 71 cb

2020/08/07 11:17:42 modbus: recv 01 04 04 43 64 99 9a 45 e4

2020/08/07 11:17:42 modbus: send 01 04 00 06 00 02 91 ca

2020/08/07 11:17:43 modbus: recv 01 04 04 00 00 00 00 fb fc

2020/08/07 11:17:43 device SDM2201-1 did not respond (2/3): read failed:
modbus: response crc '64763' does not match expected '34043'

```

Am Fr., 7. Aug. 2020 um 12:05 Uhr schrieb andig <notifications@github.com>:

> with 009d3fe
> <https://github.com/volkszaehler/mbmd/commit/009d3fe8c8efbd2b6f56b2f30f5906db76bc6f60>
> make build freezes at
>
> Could you please retry 11c627c
> <https://github.com/volkszaehler/mbmd/commit/11c627cda760270028d817115420d82b9ee3cde8>?
> It does compile for me:
>
> ❯ make
>
> rm -rf dist/
>
> go install github.com/alvaroloes/enumer
>
> go install github.com/mjibson/esc
>
> Generating embedded assets
>
> go generate ./...
>
> golangci-lint run
>
> Running testsuite
>
> go test ./...
>
> ?       github.com/volkszaehler/mbmd    [no test files]
>
> ?       github.com/volkszaehler/mbmd/cmd        [no test files]
>
> ?       github.com/volkszaehler/mbmd/meters     [no test files]
>
> ok      github.com/volkszaehler/mbmd/meters/rs485       (cached)
>
> ?       github.com/volkszaehler/mbmd/meters/sunspec     [no test files]
>
> ok      github.com/volkszaehler/mbmd/server     (cached)
>
> Version: 11c627c 2020-08-07_10:00:37
>
> go build -v -ldflags '-X "github.com/volkszaehler/mbmd/server.Version=11c627c" -X "github.com/volkszaehler/mbmd/server.Commit=11c627c"'
>
>
> If the older commits don't work I can only repeat my suggestion to git
> bisect or PM at cpuidle@gmx.de for physical device access for testing.
>
> —
> You are receiving this because you were mentioned.
> Reply to this email directly, view it on GitHub
> <https://github.com/volkszaehler/mbmd/issues/144#issuecomment-670440166>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/APMIS6UJBHBCFHNPMHMFKELR7PGWPANCNFSM4PV4OMIA>
> .
>

andig, 2020-08-07:
> read failed:

    modbus: response crc '64763' does not match expected '34043'

Not helpful but conclusive. The meter violates the spec. I'd suggest to contact Eastron as I have no idea what CRC algorithm they are using- it is apparently wrong...

dproetel, 2020-08-07:
Thanks,
I read somewhere that the 120C uses a "different" protocol. Maybe thats the reason. Because a lot of communication works fine, like the first 3 requests and also the scanning request. 

I will replace the SDM with annother model, the 230. Maybe that works. Is there somewhere in the code the possibiliy to configure the requests:
```
2020/08/07 11:17:42 modbus: send 01 04 00 06 00 02 91 ca
2020/08/07 11:17:43 modbus: recv 01 04 04 00 00 00 00 fb fc
````
thanks Daniel

andig, 2020-08-07:
We can support something else if needed, but to do that I must understand what that "different" protocol is- couldn't find any hints...

MartinJ-de, 2022-06-23:
I found this thread when searching for the mbmd problems with my SDM120 meters. The SDM72DM-V2 on the same bus works flawlessly, the three SM120 meters one rail lower don't. I see the same "timeout" errors in the mbmd output. However, the raw output shows several successful requests to these meters and then a timeout occurs. I don't see a error like the "modbus: response crc '64763' does not match expected '34043'" error message from above.

Is the patch with this extended error message in the latest build or do I have to do anything to see possible crc errors? I won't completely dismiss the possibility that I really have communication errors on the line...

Martin

PS. Thank you for your excellent work so far ;-)

MartinJ-de, 2022-06-23:
Another observation: I'm using a Raspi 3B+ with the latest OS-Version "Bullseye 64bit" and a RS485-hat. Installed a golang-environment (go version go1.18.3 linux/arm64) and compiled mbmd. There are four meters on the Modbus (1m long, each end terminated with 120 Ohm): SDM72DM-V2 and 3 x SDM120.

The SDM72 worked out-of-the-box. Works perfectly, even with no termination and a long cable, no problems at all. The SDM120 didn't work at first and on-and-off after a few tweaks (short cable, termination on both side),

BUT: if I add "--raw" at the end of the CL

./mbmd run  -a /dev/ttyS0 --comset 8N1 -b 9600 -d sdm:10,sdm:21,sdm:22,sdm:23 --raw

it works without obvious problems. There is (or is not, depends) a on-time error when registering the meter, after that the requests flow freely. For me as a long time software developer this screams "timing problem". It seems that the follow-up requests from the mbmd after a modbus message are too soon after the previous request for the small processor in the SM120. I will install the system on an old (slow...) Raspi 1B and check if this will change things.

Martin

timtimsson, 2022-10-15:
Hello,
I use 1x630 + 2x 120C with https://github.com/gonium/gosdm630 no problems…
First run with mbmd without errors - 2nd loop with error:
2022/10/15 23:41:06 modbus: send 03 04 00 34 00 02 31 e7
2022/10/15 23:41:07 modbus: recv 
2022/10/15 23:41:07 device SDM1.3 did not respond (1/3): read failed: serial: timeout

Tim

andig, 2022-10-16:
did you try settings a timeout?

timtimsson, 2022-10-16:
> did you try settings a timeout?
`./mbmd run -a /dev/serial/by-id/usb-Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller_011AB4B9-if00-port0 --comset 8N1 -b 9600 -d SDM:1,SDM230:2,SDM230:3 —raw
`
How can I do that?

DerTod2, 2023-05-23:
Hi there. 
I use one sdm630 and one sdm120M. The 630 is no problem only the 120 is not readable.

With mbdm with the following command (or only sdm:2) i'll get this response:
`
mbmd -a /dev/serial0 run -d sdm220:2 -b 9600 --api 0.0.0.0:8082 --raw -v
2023/05/23 19:06:24 mbmd 0.13 (fa6b574)
2023/05/23 19:06:24 config: creating RTU connection for /dev/serial0 (9600baud, 8N1)
2023/05/23 19:06:24 config: declared device SDM220:2.0
2023/05/23 19:06:24 httpd: starting api at 0.0.0.0:8082
2023/05/23 19:06:24 initialized device SDM2201.2: {SDM220 SDM220 Eastron SDM220    0}
2023/05/23 19:06:24 modbus: send 02 04 00 4c 00 02 b0 2f
2023/05/23 19:06:25 modbus: recv
2023/05/23 19:06:25 device SDM2201.2 did not respond (1/3): read failed: serial: timeout
2023/05/23 19:06:26 modbus: send 02 04 00 4c 00 02 b0 2f
2023/05/23 19:06:26 modbus: recv 02 04 04 00 00 00 00 c8 84
2023/05/23 19:06:26 modbus: send 02 04 00 4e 00 02 11 ef
2023/05/23 19:06:26 modbus: recv
2023/05/23 19:06:26 device SDM2201.2 did not respond (2/3): read failed: serial: timeout
2023/05/23 19:06:27 modbus: send 02 04 00 4e 00 02 11 ef
2023/05/23 19:06:27 modbus: recv
2023/05/23 19:06:27 device SDM2201.2 did not respond (3/3): read failed: serial: timeout
2023/05/23 19:06:28 device SDM2201.2 is offline
^C2023/05/23 19:06:30 received signal - stopping
2023/05/23 19:06:30 stopped

`

With sdm120c the power is without any problem readable:
`
sdm120c /dev/serial0 -a 2 -b 9600 -p -d 1 -x
20230523-19:08:34.211974: sdm120c(1972) debug_flag = 1
20230523-19:08:34.212289: sdm120c(1972) optind = 10, argc = 10, c = x, optarg = (null)
20230523-19:08:34.212564: sdm120c(1972) trace_flag = 1, count_param = 1
20230523-19:08:34.212750: sdm120c(1972) cmdline="sdm120c /dev/serial0 -a 2 -b 9600 -p -d 1 -x"
20230523-19:08:34.212959: sdm120c(1972) szttyDevice: /dev/serial0
20230523-19:08:34.213140: sdm120c(1972) devLCKfile: </var/lock/LCK..serial0>
20230523-19:08:34.213319: sdm120c(1972) devLCKfileNew: </var/lock/LCK..serial0.1972>
20230523-19:08:34.213499: sdm120c(1972) PID: 1972
20230523-19:08:34.213760: sdm120c(1972) Attempting to get lock on Serial Port /dev/serial0...
20230523-19:08:34.213995: sdm120c(1972) Acquiring shared lock on /var/lock/LCK..serial0...
20230523-19:08:34.214209: sdm120c(1972) Shared lock on /var/lock/LCK..serial0 acquired...
20230523-19:08:34.214511: sdm120c(1972) Checking for lock
20230523-19:08:34.214872: sdm120c(1972) errno=0, bRead=2 PID=1972 LckPID=1972
20230523-19:08:34.215044: sdm120c(1972) Checking process 1972 (sdm120c) for lock
20230523-19:08:34.215218: sdm120c(1972) PID: 1972 COMMAND: "sdm120c" LckPID: 1972 LckCOMMAND: "sdm120c" LckPIDcommand "sdm120c" = me
20230523-19:08:34.215322: sdm120c(1972) Appears we got the lock.
20230523-19:08:34.215414: sdm120c(1972) resp_timeout=200000us
20230523-19:08:34.215516: sdm120c(1972) Libmodbus context open (9600E1)
20230523-19:08:34.215610: sdm120c(1972) Byte timeout disabled.
20230523-19:08:34.215702: sdm120c(1972) New response timeout: 0s, 200000us
Opening /dev/serial0 at 9600 bauds (E, 8, 1)
20230523-19:08:34.215999: sdm120c(1972) 1/1. Register Address 30013 [000C]
[02][04][00][0C][00][02][B1][FB]
Waiting for a confirmation...
<02><04><04><43><CB><0C><CD><68><6B>
20230523-19:08:34.247044: sdm120c(1972) Read time: 30935us
20230523-19:08:34.247130: sdm120c(1972) reg[0/1]=17355 (0x43CB)
20230523-19:08:34.247207: sdm120c(1972) reg[1/1]=3277 (0xCCD)
Power: 406.10 W
20230523-19:08:34.247407: sdm120c(1972) devLCKfile: </var/lock/LCK..serial0>
20230523-19:08:34.247489: sdm120c(1972) devLCKfileNew: </var/lock/LCK..serial0.1972>
20230523-19:08:34.247565: sdm120c(1972) Clearing Serial Port Lock (1972)...
20230523-19:08:34.247673: sdm120c(1972) Acquiring exclusive lock on /var/lock/LCK..serial0...
20230523-19:08:34.247775: sdm120c(1972) Exclusive lock on /var/lock/LCK..serial0 acquired (0) Success...
20230523-19:08:34.247945: sdm120c(1972) cmdLen=7
20230523-19:08:34.248027: sdm120c(1972) cmdLen=7 COMMAND is not null
20230523-19:08:34.248130: sdm120c(1972) errno=0, bRead=2 LckPID=1972 PID=1972 COMMAND='sdm120c'
20230523-19:08:34.248214: sdm120c(1972) errno=0, bRead=-1 LckPID=1972 PID=0 COMMAND=''
20230523-19:08:34.248438: sdm120c(1972) Clearing Serial Port Lock done
OK
`

Does this help?

andig, 2026-09-30:
The original CRC mismatch is the meter sending non-standard checksums, which mbmd cannot work around. There is now a dedicated `SDM120` type (#377) and a `--timeout` flag for slow devices; please open a new issue with raw logs if problems persist on a current build.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig