volkszaehler/mbmd #151: iEM3000 Fehler

wing84, 2020-08-11:
panic: runtime error: index out of range [7] with length 4

goroutine 52 [running]:

encoding/binary.bigEndian.Uint64(...)

/home/travis/.gimme/versions/go1.13.linux.amd64/src/encoding/binary/binary.go:124

github.com/volkszaehler/mbmd/meters/rs485.RTUInt64ToFloat64(0x1aea103, 0x4, 0xfd, 0x0, 0x19da700)

/home/travis/gopath/src/github.com/volkszaehler/mbmd/meters/rs485/transform.go:76 +0x88

github.com/volkszaehler/mbmd/meters/rs485.MakeScaledTransform.func1(0x1aea103, 0x4, 0xfd, 0x4, 0xfd)

/home/travis/gopath/src/github.com/volkszaehler/mbmd/meters/rs485/transform.go:83 +0x3c

github.com/volkszaehler/mbmd/meters/rs485.(*RS485).QueryOp(0x199af40, 0x756b20, 0x187ae40, 0xdc60003, 0x2, 0x31, 0x1ac8150, 0x0, 0x0, 0x0, ...)

/home/travis/gopath/src/github.com/volkszaehler/mbmd/meters/rs485/rs485.go:101 +0xd4

github.com/volkszaehler/mbmd/meters/rs485.(*RS485).Query(0x199af40, 0x756b20, 0x187ae40, 0xa, 0x1ac6048, 0xa, 0x0, 0x6)

/home/travis/gopath/src/github.com/volkszaehler/mbmd/meters/rs485/rs485.go:137 +0xe4

github.com/volkszaehler/mbmd/server.(*Handler).queryDevice(0x187aea0, 0x753060, 0x199b1a0, 0x186ce80, 0x186ce40, 0x753403, 0x753460, 0x199af40)

/home/travis/gopath/src/github.com/volkszaehler/mbmd/server/handler.go:136 +0xd8

github.com/volkszaehler/mbmd/server.(*Handler).Run.func1(0x19ea003, 0x753460, 0x199af40)

/home/travis/gopath/src/github.com/volkszaehler/mbmd/server/handler.go:84 +0x14c

github.com/volkszaehler/mbmd/meters.(*Manager).All(0x199aea0, 0x1a70f40)

/home/travis/gopath/src/github.com/volkszaehler/mbmd/meters/manager.go:42 +0x6c

github.com/volkszaehler/mbmd/server.(*Handler).Run(0x187aea0, 0x753060, 0x199b1a0, 0x186ce80, 0x186ce40)

/home/travis/gopath/src/github.com/volkszaehler/mbmd/server/handler.go:55 +0x60

github.com/volkszaehler/mbmd/server.(*QueryEngine).Run.func1(0x3b9aca00, 0x0, 0x753060, 0x199b1a0, 0x186ce80, 0x186ce40, 0x19d6010, 0x187aea0)

/home/travis/gopath/src/github.com/volkszaehler/mbmd/server/queryengine.go:95 +0x80

created by github.com/volkszaehler/mbmd/server.(*QueryEngine).Run

/home/travis/gopath/src/github.com/volkszaehler/mbmd/server/queryengine.go:89 +0x128

andig, 2020-08-11:
Seems the reading is not an int64. Are you aware which value that affects? If unsure- could you grant testing access to the meter or have a live vscode debug session (please pm me at cpuidle@gmx.de)?

andig, 2020-08-11:
Please check if #152 solves the problem. Are the readings correct?

wing84, 2020-08-13:
now i can start, but i get very strange values
https://forum.geizhals.at/files/57237/Image%201.jpg
https://forum.geizhals.at/files/57237/Image%202.jpg

andig, 2020-08-13:
Mhhm, might be big endianninstead of little endian?

PR #152 (2020-08-12): Fix int64 read length
Fix #151