volkszaehler/mbmd #153: Configuration file is used continuously if specified once

thomo, 2020-08-12:
I create a config file and run mbmd with the config param (`mbmd run -c /etc/mbmd.yaml`)  - ok.
After I stop the program (CTRL-C) and rerun it without the config param it keeps going to read the config. 

```
pi@racknode:/usr/local/bin $ mbmd run -c /etc/mbmd.yaml
2020/08/12 21:48:52 mbmd 0.13 (fa6b574)
2020/08/12 21:48:52 config: using /etc/mbmd.yaml
2020/08/12 21:48:52 config: creating TCP connection for inverter.thomo.de:1502
2020/08/12 21:48:52 config: using default adapter inverter.thomo.de:1502 for device {SE 1 0 inverter }
2020/08/12 21:48:52 config: using default adapter inverter.thomo.de:1502 for device {SE 1 1 powermeter }
2020/08/12 21:48:52 config: declared device SE:1.0
2020/08/12 21:48:52 config: declared device SE:1.1
2020/08/12 21:48:52 httpd: starting api at 0.0.0.0:8080
2020/08/12 21:48:52 initialized device SE1.1: {SE SolarEdge SE15K-RW0T0BNN4  0004.000 7E08C3F2 0}
2020/08/12 21:48:53 initialized device SE1.1.1: {SE SolarEdge SE-MTR-3Y-400V-A Export+I 73 606384567 1}
^C2020/08/12 21:48:53 received signal - stopping
2020/08/12 21:48:53 stopped
pi@racknode:/usr/local/bin $ mbmd run
2020/08/12 21:49:00 mbmd 0.13 (fa6b574)
2020/08/12 21:49:00 config: using /etc/mbmd.yaml
2020/08/12 21:49:00 config: creating TCP connection for inverter.thomo.de:1502
2020/08/12 21:49:00 config: using default adapter inverter.thomo.de:1502 for device {SE 1 0 inverter }
2020/08/12 21:49:00 config: using default adapter inverter.thomo.de:1502 for device {SE 1 1 powermeter }
2020/08/12 21:49:00 config: declared device SE:1.0
2020/08/12 21:49:00 config: declared device SE:1.1
2020/08/12 21:49:00 httpd: starting api at 0.0.0.0:8080
2020/08/12 21:49:01 initialized device SE1.1: {SE SolarEdge SE15K-RW0T0BNN4  0004.000 7E08C3F2 0}
^C2020/08/12 21:49:01 received signal - stopping
2020/08/12 21:49:01 initialized device SE1.1.1: {SE SolarEdge SE-MTR-3Y-400V-A Export+I 73 606384567 1}
2020/08/12 21:49:01 stopped
```

Does mbmd somewhere store the information about the config or is it a feature of golang? Or is it magic?

andig, 2020-08-12:
It is https://github.com/volkszaehler/mbmd/blob/master/cmd/root.go#L98 as in every other unix peogram.

closed 2020-08-12 by andig

thomo, 2020-08-12:
Ok - would be good if this is also mentioned in the docu.