volkszaehler/mbmd #155: Orno WE514 Errors

dosordie, 2020-08-13:
Hi, have a problem with the Orno counter. Every second request to the device fails. However, all values ​​are displayed correctly.
How could I isolate the problem?

`{"StartTime":"2020-08-11T21:16:29.595013831+02:00","UpTime":170780.75667915,"Goroutines":47,"Memory":{"Alloc":5208816,"HeapAlloc":5208816},"Meters":[{"Device":"DZG1.18","Type":"DZG","Online":true,"Requests":21349,"RequestsPerMinute":7.500607811139889,"Errors":1,"ErrorsPerMinute":0.0003513329809892683},{"Device":"ORNO1p1.1","Type":"ORNO1p","Online":true,"Requests":42696,"RequestsPerMinute":15.000337266489238,"Errors":21348,"ErrorsPerMinute":7.500168633244619}]}`

andig, 2020-08-13:
No idea. Bus not terminated? Too many devices?

dosordie, 2020-08-13:
There are only those two devices. Bus is terminated.
But the cable to the Orno is not optimal. Maybe it's because of that.
I was just amazed that almost every second request fails.
Everything works, so not that bad :)

andig, 2020-08-13:
Might be it needs more delay to recover after slave id change. Currently its only 40 or 100ms if I remember correctly. You would need to modify the source to verify that.

dosordie, 2020-08-31:
Changing the source code is not my strength;)
can you adjust that in the next version as a test?

andig, 2020-08-31:
You can change it here: https://github.com/volkszaehler/mbmd/blob/master/meters/rtu.go#L74. How us your orno connected btw? Could you share your config?

roednix, 2021-01-18:
this alone won't help, the delay is required after changing the ID, quick patch...:
```
if b.prevID != 0 && deviceID != b.prevID {
		time.Sleep(time.Duration(100) * time.Millisecond)
		b.prevID = deviceID
	}
b.Handler.SetSlave(deviceID)
       // 200ms works fine in my setup
time.Sleep(time.Duration(200) * time.Millisecond)
```
The correct approach IMHO would be to change Slave and delay afterwards only on condition of different slave IDs (which works on my setup, however I have no idea if it breaks others, only ORNO to test here):
```
if b.prevID != 0 && deviceID != b.prevID {
	b.prevID = deviceID
	b.Handler.SetSlave(deviceID)
        time.Sleep(time.Duration(200) * time.Millisecond)
}
```

andig, 2021-01-18:
Could you submit a pr?

ochykysh, 2021-09-12:
Hi,

I tried connecting two ORNO WE-514 to one bus. I get timeout errors when mbmd tries switching bus ID. Bus is property terminated, shielded and about 20cm long.

`2021/09/12 22:18:43 device ORNO1p1.1 did not respond (1/3): read failed: serial: timeout
2021/09/12 22:18:44 modbus: send 01 03 01 31 00 01 d4 39
2021/09/12 22:18:44 modbus: recv 01 03 02 5a 66 02 ce
2021/09/12 22:18:44 modbus: send 01 03 01 30 00 01 85 f9
2021/09/12 22:18:44 modbus: recv 01 03 02 13 89 74 d2
2021/09/12 22:18:44 modbus: send 01 03 01 39 00 02 15 fa
2021/09/12 22:18:44 modbus: recv 01 03 04 00 00 05 c1 38 f3
2021/09/12 22:18:44 modbus: send 01 03 01 40 00 02 c4 23
2021/09/12 22:18:44 modbus: recv 01 03 04 00 00 01 47 bb 91
2021/09/12 22:18:44 modbus: send 01 03 01 48 00 02 45 e1
2021/09/12 22:18:44 modbus: recv 01 03 04 00 00 00 62 7b da
2021/09/12 22:18:44 modbus: send 01 03 01 50 00 02 c5 e6
2021/09/12 22:18:44 modbus: recv 01 03 04 00 00 01 55 3b 9c
2021/09/12 22:18:44 modbus: send 01 03 01 58 00 01 04 25
2021/09/12 22:18:44 modbus: recv 01 03 02 03 bf f9 04
2021/09/12 22:18:44 modbus: send 01 03 a0 00 00 02 e6 0b
2021/09/12 22:18:44 modbus: recv 01 03 04 00 03 05 30 09 77
2021/09/12 22:18:44 modbus: send 01 03 a0 1e 00 02 86 0d
2021/09/12 22:18:44 modbus: recv 01 03 04 00 01 8f cc cf 96
2021/09/12 22:18:44 modbus: send 1f 03 01 31 00 01 d7 87
2021/09/12 22:18:45 modbus: recv
2021/09/12 22:18:45 device ORNO1p1.31 did not respond (1/3): read failed: serial: timeout
2021/09/12 22:18:46 modbus: send 1f 03 01 31 00 01 d7 87
2021/09/12 22:18:46 modbus: recv 1f 03 02 59 d8 2a 4c
2021/09/12 22:18:46 modbus: send 1f 03 01 30 00 01 86 47
2021/09/12 22:18:46 modbus: recv 1f 03 02 13 8a 9c d1
2021/09/12 22:18:46 modbus: send 1f 03 01 39 00 02 16 44
2021/09/12 22:18:46 modbus: recv 1f 03 04 00 00 01 bb 45 d1
2021/09/12 22:18:46 modbus: send 1f 03 01 40 00 02 c7 9d
2021/09/12 22:18:46 modbus: recv 1f 03 04 00 00 00 29 c5 ec
2021/09/12 22:18:46 modbus: send 1f 03 01 48 00 02 46 5f
2021/09/12 22:18:46 modbus: recv 1f 03 04 00 00 00 5e 85 ca
2021/09/12 22:18:46 modbus: send 1f 03 01 50 00 02 c6 58
2021/09/12 22:18:46 modbus: recv 1f 03 04 00 00 00 66 84 18
2021/09/12 22:18:46 modbus: send 1f 03 01 58 00 01 07 9b
2021/09/12 22:18:46 modbus: recv 1f 03 02 01 97 50 78
2021/09/12 22:18:46 modbus: send 1f 03 a0 00 00 02 e5 b5
2021/09/12 22:18:46 modbus: recv 1f 03 04 00 03 04 0e 77 36
2021/09/12 22:18:46 modbus: send 1f 03 a0 1e 00 02 85 b3
2021/09/12 22:18:46 modbus: recv 1f 03 04 00 02 0a 5c a3 6b
2021/09/12 22:18:46 modbus: send 01 03 01 31 00 01 d4 39
2021/09/12 22:18:47 modbus: recv
2021/09/12 22:18:47 device ORNO1p1.1 did not respond (1/3): read failed: serial: timeout
2021/09/12 22:18:48 modbus: send 01 03 01 31 00 01 d4 39
2021/09/12 22:18:48 modbus: recv 01 03 02 5a 85 43 47
2021/09/12 22:18:48 modbus: send 01 03 01 30 00 01 85 f9
2021/09/12 22:18:48 modbus: recv 01 03 02 13 8a 34 d3
2021/09/12 22:18:48 modbus: send 01 03 01 39 00 02 15 fa
2021/09/12 22:18:48 modbus: recv 01 03 04 00 00 05 c8 f8 f5
2021/09/12 22:18:48 modbus: send 01 03 01 40 00 02 c4 23
2021/09/12 22:18:48 modbus: recv 01 03 04 00 00 01 49 3a 55
2021/09/12 22:18:48 modbus: send 01 03 01 48 00 02 45 e1
2021/09/12 22:18:48 modbus: recv 01 03 04 00 00 00 62 7b da
2021/09/12 22:18:48 modbus: send 01 03 01 50 00 02 c5 e6
2021/09/12 22:18:48 modbus: recv 01 03 04 00 00 01 57 ba 5d
2021/09/12 22:18:48 modbus: send 01 03 01 58 00 01 04 25
2021/09/12 22:18:48 modbus: recv 01 03 02 03 bf f9 04
2021/09/12 22:18:48 modbus: send 01 03 a0 00 00 02 e6 0b
2021/09/12 22:18:48 modbus: recv 01 03 04 00 03 05 30 09 77
2021/09/12 22:18:48 modbus: send 01 03 a0 1e 00 02 86 0d
2021/09/12 22:18:48 modbus: recv 01 03 04 00 01 8f cc cf 96
2021/09/12 22:18:48 modbus: send 1f 03 01 31 00 01 d7 87
2021/09/12 22:18:49 modbus: recv
`
@roednix I can test your PR, if you want me to.

andig, 2022-04-29:
Thank you @RichieB2B

closed 2022-04-29 by andig