volkszaehler/mbmd #163: could we supply aliases for meter types?

jellevictoor, 2020-08-27:
i had mbmd running for rather a long time but i was facing issues with connecting my SDM120, i had it configured as an SDM, but only after a long time i found out that i needed to configure it as a SDM220.
I know the SDM 120 is a very popular meter, i want to update the documenation to make this more clear, but i don't really know where i should do this.

andig, 2020-09-01:
Good idea. I thought that the 120 is not working at all (https://github.com/volkszaehler/mbmd/issues/144)?

jellevictoor, 2020-09-03:
the 120 works when i configured it as SDM220, i do get some drops, but overal it works. i have 1 120 and 1 630 on 1 bus. i will add more in the future

closed 2021-03-07 by andig