volkszaehler/mbmd #168: Data readings on Kostal Inverter incorrect

zulautholger, 2020-08-30:
I'm trying to read Data from my Kostal Plenticore Plus 10.

When I use inspect I get different models with different results. 
Model 103 reads the right values e.g. 2780553.00 for WH.
Model 113 results in values e.g. -2.4630083e-06 for WH.

When I open the API in a browser, it unfortunatly reads the values for Model 113.

How can i change the model or how can i get mbmd to deliver the correct values?

andig, 2020-08-31:
Could you share the inspect output?

zulautholger, 2020-08-31:
[kostal_inspect.txt](https://github.com/volkszaehler/mbmd/files/5149155/kostal_inspect.txt)

andig, 2020-08-31:
@zulautholger could you email me at cpuidle@gmx.de for remote access to the inverter? It would be great if I could fix the broken float readings. I assume this is more an encoding issue.

closed 2021-03-07 by andig