volkszaehler/mbmd #178: Missing Registers on HTML page for SMA products

DerTod2, 2020-11-06:
Maybe im to stupid or don't understand the configuration correctly but the html page shows only specific data.

I have two SMA products installed, one "SMA Tripower 8.0" Solar Inverter and one "SMA Sunny Island 4.4M-13" Battery Inverter.
The webpage shows following entries for the battery:
- Charge State
- Battery Voltage
- Voltage
- Current
- Power
- Export
- Frequency

 which is all i need.

The webpage shows following entries for the solar inverter:
- Current (for all 3 phases)
- Export 

and that is the problem ^^

With the inspect commend all registers are displayed but with run only the listed above. I need the current generation and some more data, how can i add the entries to the deamon? 

Using the current master branch self compiled.
Thanks for any advice :)

andig, 2020-11-06:
Please show inspect output wirh required registers. We can add more.

DerTod2, 2020-11-06:
Full Inspect for Solar Inverter:

>  mbmd inspect -a 192.168.188.31:502 -d SMA:126
2020/11/06 21:42:12 config: creating TCP connection for 192.168.188.31:502
--------- Model 1 common ---------
Mn                SMA    string
Md     Solar Inverter    string
Opt              9347    string
Vr           3.10.11.    string
SN         30076xxxxx    string
DA                n/a    uint16
Pad            0x8000    pad
--------- Model 11  ---------
Spd               100.00    uint16
CfgSt             0x0003    bitfield16
St                     1    enum16
MAC    00:40:??:??:??:??    eui48
Nam                         string
Ctl                  n/a    bitfield16
FrcSpd               n/a    uint16
--------- Model 12  ---------
Nam                          string
CfgSt                   1    enum16
ChgSt              0x0000    bitfield16
Cap                0x0005    bitfield16
Cfg                     1    enum16
Ctl                     0    enum16
Addr             192.168.    string
Msk              255.255.    string
Gw               192.168.    string
DNS1             192.168.    string
DNS2                         string
NTP1                         string
NTP2                         string
DomNam                       string
HostNam                      string
Pad                0x8000    pad
--------- Model 103 inverter ---------
A                     n/a    uint16
AphA              3276.80    uint16
AphB              3276.80    uint16
AphC              3276.80    uint16
A_SF                   -1    sunssf
PPVphAB               n/a    uint16
PPVphBC               n/a    uint16
PPVphCA               n/a    uint16
PhVphA                n/a    uint16
PhVphB                n/a    uint16
PhVphC                n/a    uint16
V_SF                   -1    sunssf
W                     n/a    int16
W_SF                    1    sunssf
Hz                    n/a    uint16
Hz_SF                  -2    sunssf
VA                    n/a    int16
VA_SF                   1    sunssf
VAr                   n/a    int16
VAr_SF                  1    sunssf
PF                    n/a    int16
PF_SF                  -3    sunssf
WH             1374820.00    acc32
WH_SF                   1    sunssf
DCA                   n/a    uint16
DCA_SF                n/a    sunssf
DCV                   n/a    uint16
DCV_SF                n/a    sunssf
DCW                   n/a    int16
DCW_SF                  1    sunssf
TmpCab                n/a    int16
TmpSnk                n/a    int16
[... 510 lines cut ...]
V13                    n/a    uint16
Tms14                  n/a    uint16
V14                    n/a    uint16
Tms15                  n/a    uint16
V15                    n/a    uint16
Tms16                  n/a    uint16
V16                    n/a    uint16
Tms17                  n/a    uint16
V17                    n/a    uint16
Tms18                  n/a    uint16
V18                    n/a    uint16
Tms19                  n/a    uint16
V19                    n/a    uint16
Tms20                  n/a    uint16
V20                    n/a    uint16
CrvNam                        string
ReadOnly                 0    enum16

For Total Pac:
The register should be 40200 (named W, one register) for SunSpec or 30775 (named GridMs.TotW, two registers) for SMA but can't correctly confirm because it is night here :) 

For Pac for each phase is not available in SunSpec only for SMA registers :/
Good would also be the daily production but can't find the SunSpec register only for SMA :(

When it is day here i'll reply with the inspect/data and really needed registers.

andig, 2020-11-06:
Wenn SunSpec nicht ausreicht kannst Du immer nich ein eigenes Mapping für SMA anfertigen- mir fehlt dafür leider die Zeit :/

DerTod2, 2020-11-06:
Wenn ich wüsste wie, ich habe die komplette Mapping Tabelle von SMA vorliegen. Gibt es eine Beispiel-Datei wo Mappings bzw. die Register-Adressen drinstehen die genutzt werden dann kann ich schauen das ich das hinkriege ^^ Oder kommt das von irgendeiner Abhängikeit?

andig, 2020-11-07:
Die Mappings findest Du alle in https://github.com/volkszaehler/mbmd/tree/master/meters/rs485. Jeweils Register auf generischen Wert.

andig, 2020-12-21:
Hat sich das Thema erledigt? Was kann hier noch getan werden?

DerTod2, 2020-12-21:
Habe das vergessen da ich es zu umständlich fand und dann doch nun direkt per Modbus auf meine Wechselrichter zugreife.

Daher hat sich das soweit erledigt :)

closed 2020-12-21 by DerTod2