volkszaehler/mbmd #179: AMP not totalling

matthijsberg, 2020-11-25:
HI, 

First of all thank you very much for building this. Using this to a INEPRO 380 meter on a Unipi device, works like a charm! 

I am reading power production from my solar panels. In the web page all values give me a total value, but AMP is not, see the screenshot. Is this to be expected, or not counting because it's a negative?

thanks again! 

[image]

andig, 2020-12-21:
Think I wasn't sure if it's physically correct to add these, but I guess we could do.

andig, 2026-09-30:
Sorry for letting this sit for so long without a resolution. The request is valid, but nobody has picked it up in the meantime and there are no plans to implement it from the maintainer side, so I am closing this to keep the tracker honest. If it is still relevant, a PR is very welcome and we will happily reopen.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig