volkszaehler/mbmd #180: Sunspec via TCP stops sending data

fjgalesloot, 2020-11-25:
I am using two separate mbmd instances (docker containers). One is reading a few Modbus meters, the other one is connecting to my Solaredge inverter using the Sunspec protocol via TCP/IP. I use the MQTT protocol to get the data where I want in my domotica system (node-red --> domoticz)

The instance that reads the inverter stops sending data regularly. I cannot find a pattern in when it stops. As a workaround I restart the container several times a day (during dark hours) but even that is not enough to get reliable readings every day.

Maybe the TCP connection drops for some reason. Is there a way to reconnect if that happens? I am willing to try and solve this myself, but as I never used Go, I would really appreciate some pointers where to start looking in the code.

andig, 2020-11-25:
Normally, the modbus lib should reconnect itself. Do you have a container logfile? As for go- give it a try, it's a really amazing language!

fjgalesloot, 2020-11-26:
See the log below. There are the log entries after the reboot scheduled at 03:00 UTC. Somewhere between 03:02:09 and 07:34:31 there were no messages published to the MQTT broker. I rebooted the container at 07:34 this morning.

I see now that I am not running the latest version. I will upgrade right away, 

> 2020/11/25 03:00:12 mbmd 0.10 (e83718d)
> 2020/11/25 03:00:12 config: creating TCP connection for x.x.x.35:1502
> 2020/11/25 03:00:12 config: declared device SUNS:1
> 2020/11/25 03:00:12 mqtt: connecting mbmd_solaredge at [tcp://x.x.x.240:1883]
> 2020/11/25 03:00:12 httpd: starting api at 0.0.0.0:8080
> 2020/11/25 03:00:12 initialized device SolarEdge 1.1: {SolarEdge  SE15K-RW000BNN4  0004.000 xxxxxx}
> 2020/11/25 03:00:12 mqtt: connecting mbmd_solaredge-solaredge 1-1 at [tcp://x.x.x.240:1883]
> 2020/11/25 03:00:12 updates available - please upgrade to 0.13.0
> 2020/11/25 21:30:12 mbmd 0.10 (e83718d)
> 2020/11/25 21:30:12 config: creating TCP connection for x.x.x.35:1502
> 2020/11/25 21:30:12 config: declared device SUNS:1
> 2020/11/25 21:30:12 mqtt: connecting mbmd_solaredge at [tcp://x.x.x.240:1883]
> 2020/11/25 21:30:12 httpd: starting api at 0.0.0.0:8080
> 2020/11/25 21:30:12 initialized device SolarEdge 1.1: {SolarEdge  SE15K-RW000BNN4  0004.000 xxxxxx}
> 2020/11/25 21:30:12 mqtt: connecting mbmd_solaredge-solaredge 1-1 at [tcp://x.x.x.240:1883]
> 2020/11/25 21:30:12 updates available - please upgrade to 0.13.0
> 2020/11/26 03:00:12 mbmd 0.10 (e83718d)
> 2020/11/26 03:00:12 config: creating TCP connection for x.x.x.35:1502
> 2020/11/26 03:00:12 config: declared device SUNS:1
> 2020/11/26 03:00:12 mqtt: connecting mbmd_solaredge at [tcp://x.x.x.240:1883]
> 2020/11/26 03:00:12 httpd: starting api at 0.0.0.0:8080
> 2020/11/26 03:00:12 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:12 updates available - please upgrade to 0.13.0
> 2020/11/26 03:00:15 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:18 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:21 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:24 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:27 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:30 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:33 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:36 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:39 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:42 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:45 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:48 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:51 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:54 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:00:57 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:00 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:03 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:06 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:09 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:12 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:15 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:18 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:21 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:24 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:27 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:30 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:33 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:36 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:39 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:42 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:45 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
> 2020/11/26 03:01:48 initializing device SUNS1.1 failed: dial tcp x.x.x.35:1502: connect: connection refused
[... cut ...]
> 2020/11/26 07:34:31 config: creating TCP connection for x.x.x.35:1502
> 2020/11/26 07:34:31 config: declared device SUNS:1
> 2020/11/26 07:34:31 mqtt: connecting mbmd_solaredge at [tcp://x.x.x.240:1883]
> 2020/11/26 07:34:31 httpd: starting api at 0.0.0.0:8080
> 2020/11/26 07:34:31 initialized device SolarEdge 1.1: {SolarEdge  SE15K-RW000BNN4  0004.000 xxxxxx}
> 2020/11/26 07:34:31 mqtt: connecting mbmd_solaredge-solaredge 1-1 at [tcp://x.x.x.240:1883]
> 2020/11/26 07:34:31 updates available - please upgrade to 0.13.0

andig, 2020-11-26:
Looks like:
- inverter not reachable but finally contacted
- mqtt connection hanging

You could put some more debugging into the mqtt connect. It looks a bit as if mqtt didn't start and hence go routines deadlocked, potentially waiting for a message consumer. No idea really, you'd need (through logging) find out where the stop actually occurs. Maybe pprof live data could help but I've never used it.

fjgalesloot, 2020-11-26:
Don't know why the inverter is not reachable in the 3-o'clock reboot. I don't see any hickups in MQTT on other platforms, and I think the MQTT connection from the mbmd container is working fine, right unit it is not responding any more. 

The logs look a bit different now I installed the new version this morning:

> 2020/11/26 08:15:57 mbmd 0.13 (fa6b574)
> 2020/11/26 08:15:57 config: creating TCP connection for x.x.x.35:1502
> 2020/11/26 08:15:57 config: declared device SUNS:1.0
> 2020/11/26 08:15:57 mqtt: connecting mbmd_solaredge at [tcp://x.x.x.240:1883]
> 2020/11/26 08:15:57 httpd: starting api at 0.0.0.0:8080
> 2020/11/26 08:15:57 initialized device SUNS1.1: {SUNS SolarEdge SE15K-xxxx  0004.000 xxxx 0}
> 2020/11/26 08:15:57 mqtt: connecting mbmd_solaredge-suns1-1 at [tcp://x.x.x.240:1883]
> 2020/11/26 08:35:43 device SUNS1.1 did not respond (1/3): read tcp 172.16.0.3:54934->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:35:45 device SUNS1.1 did not respond (2/3): read tcp 172.16.0.3:54934->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:35:47 device SUNS1.1 did not respond (3/3): read tcp 172.16.0.3:54934->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:35:48 device SUNS1.1 is offline
> 2020/11/26 08:35:50 device SUNS1.1 is offline - reactivating
> 2020/11/26 08:35:51 device SUNS1.1 did not respond (1/3): dial tcp x.x.x.35:1502: i/o timeout
> 2020/11/26 08:35:53 device SUNS1.1 did not respond (2/3): dial tcp x.x.x.35:1502: i/o timeout
> 2020/11/26 08:35:55 device SUNS1.1 did not respond (3/3): dial tcp x.x.x.35:1502: i/o timeout
> 2020/11/26 08:35:56 device SUNS1.1 is offline
> 2020/11/26 08:35:58 device SUNS1.1 is offline - reactivating
> 2020/11/26 08:35:59 device SUNS1.1 did not respond (1/3): dial tcp x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:01 device SUNS1.1 did not respond (2/3): dial tcp x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:03 device SUNS1.1 did not respond (3/3): dial tcp x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:04 device SUNS1.1 is offline
> 2020/11/26 08:36:06 device SUNS1.1 is offline - reactivating
> 2020/11/26 08:36:24 device SUNS1.1 did not respond (1/3): read tcp 172.16.0.3:54948->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:26 device SUNS1.1 did not respond (2/3): read tcp 172.16.0.3:54948->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:35 device SUNS1.1 did not respond (1/3): read tcp 172.16.0.3:54948->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:37 device SUNS1.1 did not respond (2/3): read tcp 172.16.0.3:54948->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:39 device SUNS1.1 did not respond (1/3): read tcp 172.16.0.3:54948->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:41 device SUNS1.1 did not respond (1/3): read tcp 172.16.0.3:54948->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:45 device SUNS1.1 did not respond (1/3): read tcp 172.16.0.3:54948->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:47 device SUNS1.1 did not respond (2/3): read tcp 172.16.0.3:54948->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:49 device SUNS1.1 did not respond (1/3): read tcp 172.16.0.3:54948->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:51 device SUNS1.1 did not respond (2/3): read tcp 172.16.0.3:54948->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:54 device SUNS1.1 did not respond (3/3): read tcp 172.16.0.3:54948->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:36:55 device SUNS1.1 is offline
> 2020/11/26 08:36:57 device SUNS1.1 is offline - reactivating
> 2020/11/26 08:37:00 device SUNS1.1 did not respond (1/3): read tcp 172.16.0.3:54950->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:37:09 device SUNS1.1 did not respond (1/3): read tcp 172.16.0.3:54950->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:37:11 device SUNS1.1 did not respond (2/3): read tcp 172.16.0.3:54950->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:37:13 device SUNS1.1 did not respond (3/3): read tcp 172.16.0.3:54950->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:37:14 device SUNS1.1 is offline
> 2020/11/26 08:37:16 device SUNS1.1 is offline - reactivating
> 2020/11/26 08:37:17 device SUNS1.1 did not respond (1/3): dial tcp x.x.x.35:1502: i/o timeout
> 2020/11/26 08:37:31 device SUNS1.1 did not respond (1/3): read tcp 172.16.0.3:54954->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:37:34 device SUNS1.1 did not respond (1/3): read tcp 172.16.0.3:54954->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:37:36 device SUNS1.1 did not respond (2/3): read tcp 172.16.0.3:54954->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:37:45 device SUNS1.1 did not respond (1/3): read tcp 172.16.0.3:54954->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:37:47 device SUNS1.1 did not respond (2/3): read tcp 172.16.0.3:54954->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:37:49 device SUNS1.1 did not respond (3/3): read tcp 172.16.0.3:54954->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:37:50 device SUNS1.1 is offline
> 2020/11/26 08:37:52 device SUNS1.1 is offline - reactivating
> 2020/11/26 08:38:00 device SUNS1.1 did not respond (1/3): read tcp 172.16.0.3:54956->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:38:02 device SUNS1.1 did not respond (2/3): read tcp 172.16.0.3:54956->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:38:04 device SUNS1.1 did not respond (3/3): read tcp 172.16.0.3:54956->x.x.x.35:1502: i/o timeout
> 2020/11/26 08:38:05 device SUNS1.1 is offline
> 2020/11/26 08:38:07 device SUNS1.1 is offline - reactivating

No logs after that. I'll look into the connection with the inverter. It seems something is going on there.  

Does the newer version handle I/O timeouts and dropouts better perhaps?

fjgalesloot, 2020-12-03:
I can confirm updating to version 0.13 from 0.10 fixed the issue. When the inverter times out, the connection gets reactivated by mbmd again.

closed 2020-12-03 by fjgalesloot