volkszaehler/mbmd #193: Prometheus support

humenius, 2021-02-10:
Hey everyone!

I'd like to implement Prometheus support for our [project "LibreWB"](https://github.com/LibreWB). You can consider it as an alternative to InfluxDB.

If things work out well, I'll create a pull request.

andig, 2021-02-13:
Sounds good. Please let me know if you need support! The. Ain task would be mapping the plethora of measurements to Prometheus metric types, the rest should be staightforward?

humenius, 2021-02-13:
I guess. Currently 
- I added a router for `/metrics` in the existing HTTP daemon and a YAML property `api.prometheus` (maybe move it to `prometheus` as standalone) that can be set if Prometheus support is needed. 
- Added a "Prometheusd" struct which takes care of registring Prometheus metrics and also provides a `Run` method for snip runners.

I decided to create `Prometheusd` even though Prometheus Go library provides a global default registry. But I think it does make sense to have a "single instance of responsibility" for Prometheus stuff. Technically it's not really a daemon so it may be renamed at some point.

humenius, 2021-02-26:
I'm slowly getting close to finish. I have a "proposal" for refactoring Measurement structs (for a better context, check out commit https://github.com/volkszaehler/mbmd/pull/195/commits/19e54e9ac007bed9b6eae5a78db0831ffc6e2d7c.): 
* I refactored `Measurement` to be more strongly typed.
* Units are now typed as well with `enumer` and struct `Unit`.
* I implemented a optional pattern which allows flexible creation of future Measurement structs, without adding too unnecessary clutter like filling parameters for factory function with empty/nil values.
* Required `measurementOptions` for creating a `Measurement` are:
  * `WithDescription(string)` 
  * `WithUnit(string)`
  * `WithMetricType(MetricType)`

Moreover, for `Measurement` specifically, I implemented a dynamic creation of Prometheus metrics.
* At its current state, Prometheus metric names and help text for `Measurement` are derrived from a `Measurement`'s description and unit. 
  1. `Measurement`'s description is transformed to lower case and each whitespace character is replaced with an underscore character
  2. Description and the `Measurement`'s `Unit`'s plural form are concatenated with underscores.
      * If the `Unit`'s plural form is not present, we simply take the alternative representation (for Prometheus) and append an `s`.
      * The default plural form can be overridden by simply setting a custom plural form for a `Unit`
  3. In order to distinguish measurement metrics, `measurement` is prepended.
  4. For help texts:
      * If a `Unit` is present: `Measurement of DESCRIPTION in UNIT`
      * Else: `Measurement of DESCRIPTION`
  5. Possible name at Prometheus metric creation: `mbmd_measurement_l1_power_watts`
* If you don't like these "auto-generated" names and help texts, you can simply override them by using the `measurementOptions` (however, make sure to follow the usual Prometheus naming conventions):
  * `WithPrometheusName(string)`
  * `WithPrometheusDescription(string)`

The last steps would be to check where metrics can be embedded as well and to test if the result of `/metrics` HTTP API is good for us. :)

andig, 2026-09-30:
Sorry for letting this sit for so long without a resolution. The request is valid, but nobody has picked it up in the meantime and there are no plans to implement it from the maintainer side, so I am closing this to keep the tracker honest. If it is still relevant, a PR is very welcome and we will happily reopen.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig