volkszaehler/mbmd #194: How to get "power" in an http-request?

xsven80x, 2021-02-13:
Hi,

I need to geht the value of "power" for my ABB-meter in a http-request. Actually installed mbmd Version 0.13. 

I need the value for openWB Wallbox und Tecalor/Stiebel heat-pump to enable sg-ready.

Is it possible to create a http-request to get defined values (in my case "power")?

Thank you and

greetings
  Sven

andig, 2021-02-13:
Does https://github.com/volkszaehler/mbmd#rest-api not work for you?

xsven80x, 2021-02-13:
No, sorry... 

/api/last/{ID} offers me all the values, but I need only the value "power"... I don't know how to separate the values...

andig, 2021-02-23:
Not sure _why_ you need that as you can simply pick power from the json, but that's not available. Sounds related to https://github.com/volkszaehler/mbmd/issues/52.

closed 2021-03-07 by andig