volkszaehler/mbmd #197: SDM 630 v2 MID

atryda22, 2021-03-18:
Hello

First of all, thanks for a great job :-)
I am using your project to read two meters: Orno WE517 and SDM 630 .
My SDM meter is a version called SDM 630 V2 Modbus.
He has many more registers than in your project (exactly 86).
I tried to change the sdm.go file myself, but as you probably guessed it, no effect ;-)
But the good news is that I haven't broken anything, so your project is immune to such manipulation ;-)
There are also several other variations of this meter, such as MT with multiple tariffs.
Would there be a chance to expand the project with such versions?

greetings
atryda22

andig, 2021-04-27:
Would be happy to accept a pr.

lbort, 2022-01-02:
I have the same meter, and already started working with it a bit. I am new to go however, and some input would be good. 

My status is:

- I have a working hardware setup, meter connected to 3 phases of house, modbus connected to raspberry pi via TCP.
- I also have a working go environment, and managed to compile the current version. 
- Changing sdm.go also partly worked. I tried to guess some variable names, and that worked: 
```
ApparentPowerL1: 0x0012, // apparent power L1
ApparentPowerL2: 0x0014, // apparent power L2
ApparentPowerL3: 0x0016, // apparent power L3
ReactivePowerL1: 0x0018, // reactive power L1
ReactivePowerL2: 0x001A, // reactive power L2
ReactivePowerL3: 0x001C, // reactive power L3 
```

I realized that those names are defined somehow in meters/measurements.go, and are shown in assets/index.html, but adding completely new variables failed. What I tried was: 

1. add line `PhaseAngleL1:  0x0024,` in sdm.go
2. add line `PhaseAngleL1` and `PhaseAngleL1:     {"Phase Angle L1", "°"},` in measurements.go (just below the existing lines for "PhaseAngle",  l. 142 and l. 238 respectively)
3. change line 363 in index.html to `<td class="col-2" v-if="pop(m.PhaseAngleL1)">${ m.PhaseAngleL1 }</td>
						<td class="col-2" v-else>&mdash;</td>`

There are no compile errors, but also no numbers are plotted in live view. I suppose it is somehow related to the "enumer" thing mentionend in measurement.go line 23, and the measurement_enumer.go file. This is something I did not understand so far. Could someone help me with adding completely new variables? If it works, I will implement that nicely for all registers and can provide a PR.

andig, 2026-09-30:
Sorry for letting this sit for so long without a resolution. Nobody has picked up the implementation in the meantime and there are no plans to work on it from the maintainer side, so I am closing this to keep the tracker honest. If the device is still relevant to you, a PR adding it (the existing meters/rs485/*.go files serve as templates) is very welcome and we will happily reopen.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig