volkszaehler/mbmd #199: No output for model sunspec 213

BenGig, 2021-04-16:
With PR #191 support for meter float model sunspec 213 was introduced. I have a Fronius Smart Meter 63A which reports as such a model.

With `mbmd inspect`, I get all measurement correctly:

```
> docker run --rm -p 8080:8080 volkszaehler/mbmd  inspect -d SUNS:240@symo:502
2021/04/16 21:51:34 config: creating TCP connection for symo:502
--------- Model 1 common ---------
Mn            Fronius    string
Md    Smart Meter 63A    string
Opt          3.16.7-1    string
Vr                2.9    string
SN           18410005    string
DA             240.00    uint16
Pad            0x00d5    pad
--------- Model 213 ac_meter ---------
A                            3.05    float32
AphA                        0.929    float32
AphB                         1.48    float32
AphC                        0.641    float32
PhV                     244.23334    float32
PhVphA                      243.7    float32
PhVphB                      243.7    float32
PhVphC                      245.3    float32
PPV                     423.03333    float32
...
```

With `mbmd run`, connection is established, the device is again detected correctly, but no values are reported:

```
> docker run --rm -v $(pwd)/mbmd.yml:/etc/mbmd.yaml -p 8080:8080 volkszaehler/mbmd run -v
2021/04/16 22:17:33 mbmd 0.13 (fa6b574)
2021/04/16 22:17:33 config: using /etc/mbmd.yaml
2021/04/16 22:17:33 config: creating TCP connection for symo:502
2021/04/16 22:17:33 config: declared device SUNSPEC:240.0
2021/04/16 22:17:33 httpd: starting api at 0.0.0.0:8080
2021/04/16 22:17:35 initialized device SUNSPEC1.240: {SUNSPEC Fronius Smart Meter 63A 3.16.7-1 2.9 18410005 0}
^C2021/04/16 22:21:45 received signal - stopping
2021/04/16 22:21:45 stopped
```
Output stops until interrupted with CTRL-C.

andig, 2021-07-24:
@BenGig could you provide remote access for debugging?

BenGig, 2021-07-24:
Do you have the option to use Wireguard? I could give you a config, and provide a devel environment in a VM.

andig, 2021-07-24:
Wireguard would work. Access to the inverter would be good- why the VM?

BenGig, 2021-07-25:
Nevermind the VM, the first idea is rarely the best. May I use your mail address from your profile, for the details?

andig, 2021-07-25:
Sure, cpuidle@gmx.de

BenGig, 2021-07-26:
key: EeYRANsxmEKaS8uwyOD6xdeI

andig, 2021-07-28:
Bei mir sieht das so aus:

[image]

Was fehlt Dir denn jetzt konkret?

andig, 2021-07-30:
ping @BenGig

BenGig, 2021-07-30:
Sorry, Abgabetermin@Work

Browserproblem. 

Ich hab nacheinander meine Standards Brave und Waterfox versucht, dann Edge und Firefox. Nur FF hat die Daten angezeigt. Bei den anderen: der statische Teil der Webseite lädt, bleibt dann auf "Measurements: loading...".

BenGig, 2021-07-30:
Korrektur: auch Firefox zeigt nichts an; Noscript war noch aktiv. Was ich gesehen hab waren die Platzhalter {}

andig, 2021-07-30:
Kann ich nicht reproduzieren. Geht mit Safari und Chrome wunderbar.

BenGig, 2021-07-30:
JS deaktiviert:
[image]

JS aktiv:
[image]

BenGig, 2021-07-30:
Seltsam ist, dass auch MQTT zwar /mbdb/status:connected meldet, aber sonst nichts, kein /mbmd/<unique id>/*. 

mbmd
  version     : 0.13
  commit      : fa6b574
  go version  : go1.13
  go compiler : gc
  platform    : linux/amd64

andig, 2021-07-31:
> mbdb/status:connected meldet, aber sonst nichts, kein /mbmd//*.

Also root bei mqtt ist `mbmd`, nicht `/mbmd`. Ebenso ist Wildcard `mbmd/#` und nicht `/mbmd//*`.

BenGig, 2021-07-31:
War nur schnell hingeschrieben; so sieht es aus, mehr als "connected" kommt nicht:

[image]

snh, 2023-01-16:
Figured it out, the `latest` Docker image is 2 years old, since that is when the last release of `mbmd` appears to have been cut, so predates https://github.com/volkszaehler/mbmd/pull/191.

Building the Docker image from the latest version of the `master` branch resolved the issue for me:

```
docker build -t volkszaehler/mbmd:devel https://github.com/volkszaehler/mbmd.git
```

And then using `volkszaehler/mbmd:devel` as the image in the `docker run` invocation.

@andig Any plans for a new release soon, as I see a lot of improvements have been made since 0.13?

/cc @bengig

snh, 2023-01-16:
Curious, `docker build -t volkszaehler/mbmd:devel https://github.com/volkszaehler/mbmd.git#59dcc46d62d4d0169e430223c949810112ea26f3`, which is the merge commit for https://github.com/volkszaehler/mbmd/pull/191, works fine for `FRONIUS1.240` with both `inspect` and `run`.

So there seems to be a regression between when https://github.com/volkszaehler/mbmd/pull/191 was merged and now. I'll see if I can narrow in on it the relevant PR.

andig, 2023-01-16:
I guess I should automate the release process...

snh, 2023-01-16:
I think I've narrowed the regression down to https://github.com/volkszaehler/mbmd/pull/265, as the immediately proceeding commit of 7251dd80a3fb8cd49a3bb50c43207a03d0f7ffbd is fine, with a1401c1f7c7dab43910453c790fc9973e5274832 being the commit where the `i/o timeout` error crops up.

Any ideas @tobima what may be causing this?

tobima, 2023-01-17:
Looks like RTU and TCP had different values for the timeout before #265:
https://github.com/volkszaehler/mbmd/blob/870b3b92b81ec66ec4265fa2589851fb701f6fc7/meters/tcp.go#L21
It was 1s TCP and 300ms RTU. https://github.com/volkszaehler/mbmd/blob/870b3b92b81ec66ec4265fa2589851fb701f6fc7/meters/rtu.go#L36

The PR #265 was initial only about RTU and included later TCP, too. https://github.com/volkszaehler/mbmd/pull/265#issuecomment-1164712909

snh, 2023-01-17:
Thanks @tobima, I couldn't find any other reports of issues post this PR, so I suspect 300ms is still plenty for most TCP cases. I suspect part of the reason it is an issue here is that I'm talking to a Fronius Inverter using Modbus TCP which is then having to talk RTU to the connected Smart Meter, so a bit of latency is introduced by this double hop/translation.

tobima, 2023-01-17:
Issue #277  was caused by the regression after this PR, too. The issue was closed after the reporter used the timeout parameter.

I see here an issue with the documentation. Perhaps we should add a troubleshouting section to the Readme.md mentioning the timeout parameter to make this parameter more visible.

andig, 2026-09-30:
The original report was caused by the two-year-old `latest` Docker image predating #191; the later regression was the 300 ms default Modbus timeout from #265, which is too short for meters read through a Fronius inverter gateway. Using `--timeout 1s` (or `timeout: 1s` in the config) resolves it, as confirmed here and in #277. Closing as answered; a README note about `--timeout` would be a welcome PR.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig