volkszaehler/mbmd #200: Precision loss in total export energy (SunSpec Model 101)

kereis, 2021-04-19:
mbmd uses the SunSpec Model 101 for reading total export energy in Wh.

For SunSpec registers, Model 101 and Model 122 can be used. However, it looks like that the total export value in Model 101 is slightly unprecise. 

Here you can find an example. You can simply check it out (https://github.com/LibreWB/mbmd/commit/ca23429878711eb0195bf98e254017a7eaa00c35) and run `mbmd scan -c <CONFIG_FILE> -a <YOUR_ADAPTER>`. 

https://github.com/LibreWB/mbmd/blob/ca23429878711eb0195bf98e254017a7eaa00c35/meters/sunspec/sunspec.go#L191
```golang
actWh, actWhErr := client.ReadHoldingRegisters(40302, 4)
wh, whErr := client.ReadHoldingRegisters(40209, 2)

if actWhErr == nil {
    fmt.Printf("ActWh > %d\n", binary.BigEndian.Uint64(actWh))
} else {
    fmt.Println(fmt.Errorf("actWhErr > %s", whErr))
}

if whErr == nil {
    fmt.Printf("WH > %d\n", binary.BigEndian.Uint32(wh))
} else {
    fmt.Println(fmt.Errorf("whErr > %s", whErr))
}
```

What we do here is reading from the SunSpec register directly via ModBus client. Total Export Wh in Model 101 is 32-bit, in Model 122 64-bit. Actually, this shouldn't make a big difference as both total export values should fit into each sizes. However, it looks like that the 32-bit value is rounded up to the next decimal place. Example values are:
```
ActWh > 23665022
WH > 2366502
```

It might not be a huge problem for people who use mbmd casually, but in terms of monitoring, one decimal place could make a difference - at least for our current implementation of Prometheus (#195).

I've already chatted with @andig about this "issue". We should maybe look into this and think about maybe adding a model mapping entry for Model 122 or replace it with 101.

I don't know if this affects any other values, but we noticed the value difference in total export energy.

andig, 2021-04-19:
Hat `WH` einen Scale Factor? Evtl. dividieren wir hier einfach blöd? Oder kommt es wirklich *falsch* vom Bus?

kereis, 2021-04-19:
Das sind Werte, die direkt vom Register rauskommen - ich habe da nichts umgerechnet und auch kein Scaling Factor angewandt.
Im SMA-Register kommt das gleiche wie bei ActWh raus.

andig, 2026-09-30:
The one-digit difference is the model 101 scale factor: `WH` is 2366502 with `WH_SF = 1`, i.e. 23665020 Wh, which mbmd applies via `ScaledValue()`. The remaining 10 Wh resolution is what the 32-bit register offers, so this is not a precision bug in mbmd. Mapping the acc64 `ActWh` of model 122 would be the only way to get 1 Wh resolution; feel free to open a PR or a new issue if that is still needed.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig