volkszaehler/mbmd #201: can't read Modbus Meter

Chris591, 2021-04-27:
Hi,

Ich komme von EVCC und wollte mehr daten aus meinem Meter auslesen und speichern und da wurde mir MBMD empfohlen.
Leider bekomme ich hier aber diesen Output mit den gleichen Settings von Docker und auch der USB Dongle ist gleich konfiguriert.
EVCC kann ich den Meter ordentlich auslesen.

mit einem Scan findet er nichts.

hab es auch mit einem anderen USB Adapter probiert - wieder das gleiche EVCC läuft - MBMD die gleichen Probleme.

Gruß,
Chris

> 2021/04/27 08:51:41 mbmd 0.13 (fa6b574),
> 2021/04/27 08:51:41 config: using /etc/mbmd.yaml,
> 2021/04/27 08:51:41 config: creating RTU connection for /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 (9600baud, 8N1),
> 2021/04/27 08:51:41 config: declared device SDM:1.0,
> 2021/04/27 08:51:41 httpd: starting api at 0.0.0.0:8080,
> 2021/04/27 08:51:41 mqtt: connecting mbmd at [tcp://10.201.1.10:1883],
> 2021/04/27 08:51:43 initialized device SDM1.1: {SDM SDM Eastron SDM630    0},
> 2021/04/27 08:51:43 device SDM1.1 did not respond (1/3): read failed: permission denied,
> 2021/04/27 08:51:44 device SDM1.1 did not respond (2/3): read failed: permission denied,
> 2021/04/27 08:51:45 device SDM1.1 did not respond (3/3): read failed: permission denied,
> 2021/04/27 08:51:46 device SDM1.1 is offline,
> 2021/04/27 08:51:48 device SDM1.1 is offline - reactivating,
> 2021/04/27 08:51:48 device SDM1.1 did not respond (1/3): read failed: permission denied,
> 2021/04/27 08:51:49 device SDM1.1 did not respond (2/3): read failed: permission denied,
> 2021/04/27 08:51:50 device SDM1.1 did not respond (3/3): read failed: permission denied,
> 2021/04/27 08:51:51 device SDM1.1 is offline,
> 2021/04/27 08:51:53 device SDM1.1 is offline - reactivating,
> ....

andig, 2021-04-27:
Also im logfile steht permission denied. Ich würde sagen das ist das Problem.

Chris591, 2021-04-27:
Ja aber wieso und wo kann ich schauen?
EVCC läuft auf dem gleichem Raspberry PI mit dem gleichen settings vom Docker ohne Probleme.
Ist es das er nicht auf den ModBus zugreifen kann oder das er nicht auf den Meter auf alle werte zugreifen kann?
Was macht MBMD anders?

MBMD config:
```
# adapters are referenced by device
adapters:
- device: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
  baudrate: 9600
  comset: 8N1 # "8E1" needs be quoted as string or will error

# list of devices
devices:
- name: grid
  type: sdm
  id: 1
  adapter: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

```

> /usr/share/mbmd $ docker run --rm -v $(pwd)/mbmd.yaml:/etc/mbmd.yaml -v /dev:/dev -p 8080:8080 --device /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 volkszaehler/mbmd:latest run --config /etc/mbmd.yaml
> 2021/04/27 12:39:15 mbmd 0.13 (fa6b574)
> 2021/04/27 12:39:15 config: using /etc/mbmd.yaml
> 2021/04/27 12:39:15 config: creating RTU connection for /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 (9600baud, 8N1)
> 2021/04/27 12:39:15 config: declared device SDM:1.0
> 2021/04/27 12:39:15 httpd: starting api at 0.0.0.0:8080
> 2021/04/27 12:39:15 mqtt: connecting mbmd at [tcp://10.201.1.10:1883]
> 2021/04/27 12:39:17 initialized device SDM1.1: {SDM SDM Eastron SDM630    0}
> 2021/04/27 12:39:17 device SDM1.1 did not respond (1/3): read failed: permission denied
> 2021/04/27 12:39:18 device SDM1.1 did not respond (2/3): read failed: permission denied
> 2021/04/27 12:39:19 device SDM1.1 did not respond (3/3): read failed: permission denied
> 

EVCC config:
```
meters:
- name: grid
  type: modbus
  model: sdm
  device: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
  baudrate: 9600
  comset: 8N1
  id: 1
```

> /usr/share/evcc $ docker run --rm -v $(pwd)/evcc.yaml:/etc/evcc.yaml -v /dev:/dev -p 7070:7070 --device /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 --entrypoint evcc andig/evcc:latest meter
> Power: 583W
>

andig, 2021-04-27:
Kann ich Dir leider nicht sagen, aber die Fehlermeldung erscheint eindeutig. Ich würde das mal ohne Docker „nativ“ ausprobieren.

Chris591, 2021-04-27:
ja nativ läuft es - danke

closed 2021-04-27 by andig

trippled1993, 2021-06-24:
Der Vollständigkeit halber, falls noch jemand hierüber stolpert:

In meinem Fall hatte der User im Container nicht die Berechtigung meinen USB-RS485-Adapter unter `/dev/ttyUSB0` zu öffnen.
Da `/dev/ttyUSB0` standardmäßig der Gruppe `dialout` gehört, könnte z.B. der User im Container zu dieser Gruppe hinzugefügt werden.
Durch den Parameter `--group-add dialout` im docker run Befehl wurde das Problem in meinem Fall gelöst.
Pendant dazu in docker-compose wäre:
```
    group_add: 
      - dialout
```

ChristianSteffens, 2023-01-22:
Danke für den Tip - hat mir auch geholfen.