volkszaehler/mbmd #209: Howto build Docker-Image?

mottobug, 2021-05-26:
Hello,

the supplied image on docker-hub doesn't work for me because of permission issues while accessing /dev/ttyUSB0. This image does a switch to an different uid (appuser?).

I'd like to build my own docker-image but i'm getting a problem after another. The supplied dockerfile in the project doesn't build sucessfully:

go build -v -tags=release -ldflags='-X "github.com/volkszaehler/mbmd/server.Version=a1c92a0" -X "github.com/volkszaehler/mbmd/server.Commit=a1c92a0"'
build github.com/volkszaehler/mbmd: cannot load embed: malformed module path "embed": missing dot in first path element
make: *** [Makefile:30: build] Error 1

I've tried different tags and different go versions, but without success.

The supplied Dockerfile in the project is different from the one build on hub.docker.com. Wheres the original Dockerfile from hub.docker.com?

Sincerely, Thomas

andig, 2021-05-26:
Use go 1.16

mottobug, 2021-05-26:
I've changed "FROM golang:1.13-alpine as builder" to 1.16:

Step 1/16 : FROM golang:1.16-alpine as builder
 ---> 28c3a23afc0f
Step 2/16 : RUN apk update && apk add --no-cache git ca-certificates tzdata alpine-sdk && update-ca-certificates
 ---> Running in fccd89c1950b
fetch https://dl-cdn.alpinelinux.org/alpine/v3.13/main/armv7/APKINDEX.tar.gz
fetch https://dl-cdn.alpinelinux.org/alpine/v3.13/community/armv7/APKINDEX.tar.gz
ERROR: https://dl-cdn.alpinelinux.org/alpine/v3.13/main: temporary error (try again later)
WARNING: Ignoring https://dl-cdn.alpinelinux.org/alpine/v3.13/main: No such file or directory
ERROR: https://dl-cdn.alpinelinux.org/alpine/v3.13/community: temporary error (try again later)
WARNING: Ignoring https://dl-cdn.alpinelinux.org/alpine/v3.13/community: No such file or directory

mottobug, 2021-05-26:
I've changed to golang:1.16-alpine3.12, this also doesn't work:
go install github.com/mjibson/esc
no required module provides package github.com/mjibson/esc; to add it:
	go get github.com/mjibson/esc
make: *** [Makefile:19: install] Error 1
The command '/bin/sh -c make install' returned a non-zero code: 2

andig, 2021-05-26:
Fixed in https://github.com/volkszaehler/mbmd/commit/e1fec7232ed792afbfd8d33631e45a127e37ab24

mottobug, 2021-05-31:
Now its building successfully. Last issue, where's the Dockerfile of the images pushed at hub.docker.com?

andig, 2021-05-31:
It's https://github.com/volkszaehler/mbmd/blob/master/tmpl.Dockerfile

closed 2021-05-31 by andig