volkszaehler/mbmd #214: Web server issue on build

hacor, 2021-06-30:
Hello 

I made a fork of mbmd to add several extra addresses to the [SDM220 meter](https://github.com/hacor/mbmd/blob/master/meters/rs485/sdm220.go#L25)
The build seems to work fine, but in the newly generated `mbmd` binary the web server seems to fail. I get a 404 error on all css and javascript files.

Maybe I did something wrong.
I hope someone out here can help.
If this works I create a PR.

All the best!

Hacor

andig, 2021-06-30:
Did you `make` the assets?

hacor, 2021-06-30:
The build is realised by running the following docker command:

```bash
docker run --rm -e GOOS=linux -e GOARCH=arm -e GOARM=7 -v ${PWD}:/app --workdir=/app golang:1.16 /bin/bash -c "make install && make build"
```
I'm not really sure if this covers your question? I just followed the `README.md` build info

Regards!

H

andig, 2021-06-30:
Do you have the same problem building without docker?

hacor, 2021-06-30:
I never used go before, so I didn't really install it on my local machine.
I will give it a try!

hacor, 2021-06-30:
Also with the local installed Go 1.16 the https assets return a 404. See screenshot...

[image]

hacor, 2021-06-30:
When reading more and more it indeed seems the assets are not `build` into the release.
Do you have any instructions on this?

Thanks

hacor, 2021-07-02:
@andig By investigating the problem I found out that the master branch Makefile doesn't contain an `assets` option.
A `make assets` command doesn't work, but I see this changed in a [commit](https://github.com/volkszaehler/mbmd/commit/aa0e98212cfb2ba9bc1327ad7526a2ff35faeaa0#diff-76ed074a9305c04054cdebb9e9aad2d818052b07091de1f20cad0bbac34ffb52L25) from may 25.
Can you please suggest how to build the assets now with Go 1.16, or which branch/tag to use for local development? 
I know this is a complete noob question, but after successfully building my adaptations I can create the PR

Thanks!

andig, 2021-07-02:
Sorry for the confusion, some leftovers in the code, fixed with https://github.com/volkszaehler/mbmd/pull/215

closed 2021-07-02 by andig