volkszaehler/mbmd #222: ORNO WE-517 not working

bilbolodz, 2021-08-31:
I've ORNO WE-517 using simple USB to RS485 converter. Meter is not detected (and not working) but hardware connections seems to be working.
Output from custom python script: ornowe51x.py
`testing @  /dev/ttyUSB0
meter ok
Energy MODBUS: /dev/ttyUSB0

volatge L1 ...  226.0
volatge L2 ...  223.7
`
but:

`pi@raspberrypi:~/ORNO_OR-WE-51x_Modbus $ mbmd run -a /dev/ttyUSB0 -d orno3p:1 --raw
2021/08/31 21:29:21 mbmd 0.13 (fa6b574)
2021/08/31 21:29:21 config: creating RTU connection for /dev/ttyUSB0 (9600baud, 8N1)
2021/08/31 21:29:21 config: declared device ORNO3p:1.0
2021/08/31 21:29:21 httpd: starting api at 0.0.0.0:8080
2021/08/31 21:29:21 initialized device ORNO3p1.1: {ORNO3p ORNO3p ORNO WE-516 & WE-517    0}
2021/08/31 21:29:21 modbus: send 01 03 00 14 00 02 84 0f
2021/08/31 21:29:22 modbus: recv 
2021/08/31 21:29:22 device ORNO3p1.1 did not respond (1/3): read failed: serial: timeout
2021/08/31 21:29:23 modbus: send 01 03 00 14 00 02 84 0f
2021/08/31 21:29:23 modbus: recv 
2021/08/31 21:29:23 device ORNO3p1.1 did not respond (2/3): read failed: serial: timeout
2021/08/31 21:29:24 modbus: send 01 03 00 14 00 02 84 0f
2021/08/31 21:29:24 modbus: recv 
2021/08/31 21:29:24 device ORNO3p1.1 did not respond (3/3): read failed: serial: timeout
2021/08/31 21:29:25 device ORNO3p1.1 is offline
2021/08/31 21:29:26 device ORNO3p1.1 is offline - reactivating
2021/08/31 21:29:26 modbus: send 01 03 00 14 00 02 84 0f
2021/08/31 21:29:27 modbus: recv 
2021/08/31 21:29:27 device ORNO3p1.1 did not respond (1/3): read failed: serial: timeout
^C2021/08/31 21:29:27 received signal - stopping
2021/08/31 21:29:27 stopped
`

andig, 2021-08-31:
Hard to read without formatting. Sure only lne program is accessing the bus?

bilbolodz, 2021-09-01:
Yes only only one program, one device on bus. I've reformated output, generally: no response from meter.
`mbmd run -a /dev/ttyUSB0 -d orno3p:1 --raw
2021/08/31 21:29:21 mbmd 0.13 (fa6b574)
2021/08/31 21:29:21 config: creating RTU connection for /dev/ttyUSB0 (9600baud, 8N1)
2021/08/31 21:29:21 config: declared device ORNO3p:1.0
2021/08/31 21:29:21 httpd: starting api at 0.0.0.0:8080
2021/08/31 21:29:21 initialized device ORNO3p1.1: {ORNO3p ORNO3p ORNO WE-516 & WE-517 0}
2021/08/31 21:29:21 modbus: send 01 03 00 14 00 02 84 0f 2021/08/31
21:29:22 modbus: recv 2021/08/31
21:29:22 device ORNO3p1.1 did not respond (1/3): read failed: serial: timeout
2021/08/31 21:29:23 modbus: send 01 03 00 14 00 02 84 0f
2021/08/31 21:29:23 modbus: recv
2021/08/31 21:29:23 device ORNO3p1.1 did not respond (2/3): read failed: serial: timeout
2021/08/31 21:29:24 modbus: send 01 03 00 14 00 02 84 0f
2021/08/31 21:29:24 modbus: recv 2021/08/31
21:29:24 device ORNO3p1.1 did not respond (3/3): read failed: serial: timeout
2021/08/31 21:29:25 device ORNO3p1.1 is offline
2021/08/31 21:29:26 device ORNO3p1.1 is offline - reactivating
2021/08/31 21:29:26 modbus: send 01 03 00 14 00 02 84 0f
2021/08/31 21:29:27 modbus: recv 2021/08/31
21:29:27 device ORNO3p1.1 did not respond (1/3): read failed: serial: timeout ^C
2021/08/31 21:29:27 received signal - stopping 2021/08/31 21:29:27 stopped `

andig, 2021-09-01:
Might be good to compare the modbus traffic with the python script. I don‘t have this meter for testing so I can only guess…

andig, 2021-09-01:
/cc @bbiegun @ochykysh

ochykysh, 2021-09-01:
Hi @bilbolodz,

can you attach that script, please? Maybe the script modified the bus speed and mbmd expects the default settings.

bbiegun, 2021-09-01:
Hi @bilbolodz,

set com to 8E1 with `--comset 8E1`.

bilbolodz, 2021-09-01:
[https://drive.google.com/file/d/1qfypWMCsjrB861YvQiYkYt_w0Pqf1bc-/view?usp=sharing]

bilbolodz, 2021-09-01:
> --comset 8E1

Working

closed 2021-09-01 by bilbolodz

bilbolodz, 2021-09-01:
It's indeed mentioned in documentation:

> By default, the meter communicates using 9600 8E1

but maybe it's worth to write it's straight: "add --comset 8E1 parameter"???

ochykysh, 2021-09-01:
@bilbolodz feel free to create a PR :)