volkszaehler/mbmd #233: Siemens PAC 2200

kirk-loeten, 2021-11-09:
I like to integrate the PAC2200. Is it possible to write a configuration file in the rs485 folder with the addresses of the manual? The device has a modus tcp interface.

[Siemens PAC 2200 Manual](https://cache.industry.siemens.com/dl/dl-media/835/109746835/att_1058513/v3/EMAN_L1V30415167A-04_de-DE/de-DE/de-DE/anhang/modbus/index.html#96582261003
)

I will do this with your support.

andig, 2021-11-09:
Think so- just give it a try!

kirk-loeten, 2021-11-13:
OK. I try to modify a config file with the Modus addresses of the PAC. I think the development is better on a Linux machine. Later i like to use it on a Raspberry Pi or can i do the build on a PI too?

closed 2022-11-14 by kirk-loeten