volkszaehler/mbmd #242: cache writings

isarrider, 2022-01-17:
Hi,

I think it might be nice if mbmd could also cache the Database writes,
in cas it isnt availible... (like vzlogger can: "max_buffer_size": 864000,)
Thoughts?

Thx,
Alex

andig, 2022-01-17:
Which DB are you referring to? Influx?

isarrider, 2022-01-17:
yap...

andig, 2022-01-17:
The influx client already has the abilities to retry: https://pkg.go.dev/github.com/influxdata/influxdb-client-go/v2#Options.MaxRetries. Happy to take a PR exploiting it.

isarrider, 2022-01-19:
the last time I coded sth "real" is 15 years ago or so... ;)
Happy to paypal sb 10€ - guess it takes less then 30min
(maybe some computer science student is interested).

BR,
Alex

isarrider, 2026-04-19:
@andig I tried it with Claude:
https://github.com/isarrider/mbmd/tree/Caching
thoughts?
(Funfact - Claude: "I now have everything I need. I can see the current mbmd server/influx.go is using influxdb1-client (from PR #9) but the repo has since migrated to influxdb-client-go/v2 (v2.13.0 per go.sum). The current file must look similar to evcc's (a sibling project by the same author andig".)

andig, 2026-09-30:
Wontfix, see https://github.com/volkszaehler/mbmd/pull/406#issuecomment-5916315854

closed 2026-09-30 by andig