volkszaehler/mbmd #249: feature request: working as proxy

SebuZet, 2022-03-25:
Hi, first of all I want to say that program is great!
I've an enhancement proposal:
What do you think to add a "proxy" functionality. It means that we can connect eg. SDM630 meter, get data from it and emulate abother device eg. Inepro PRO380.
Thanks to thsi apraoch user can connect unsuported devices together (eg. SolarEdge inverter with SDM630 meter :))
What do you think?

Macrisu, 2022-04-25:
That would be awesome.
PV integrators are a closed wallet garden.

I have a Fronius Inverter and now I cannot find any Fonius Smartmeter o stock :(

andig, 2022-04-25:
s/wallet/walled/ ;)

Macrisu, 2022-04-25:
Wallet sink!

andig, 2023-01-31:
@SebuZet as part of https://github.com/evcc-io/evcc we have recently added a Modbus Proxy (i.e. combination of client and transparent server). You could use that to plugin any translation. It might be beyond scope of mbmd but you could still pull in evcc and mbmd as dependencies.

andig, 2026-09-30:
Translating one meter into the register layout of another device is outside the scope of mbmd. evcc ships a Modbus proxy (transparent client/server) that can be used for that kind of setup, see the comment above. Closing.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig