volkszaehler/mbmd #250: GM4008 support

bbiegun, 2022-03-26:
I recently purchased [GM4008](https://www.gotronik.pl/gm4008-modul-pomiarowy-pradu-0-24ma-ethernet-z-izolacja-p-4357.html). It is an 8-channel current acquisition module. I'm planning to use it with current transformers. It communicates over TCP Modbus.

Would you be interested in me adding support for it in mbmd?

andig, 2022-03-26:
Not sure if it fits the structure? Feel free to try!

bbiegun, 2022-03-28:
I can report `Current` only. I do see couple of caveats though.

1. It has 8 inputs -> there would need to be some option to configure which input you're interested in or one meter would generate 8 'sub-meters'
2. It can provide readings between 0-24 mA. Usually, it'd be connected to some transformer. Ideally, I would like to calculate it inside mbmd. Problem would be to provide which transformer you're using (20A/25mA, 100A/25mA, etc).

Do you think it's worth it? It'd probably require some changes in core code too. Seeking your view on it.
Alternatively, I do have a pyhton wrapper pretty much finished already.

bbiegun, 2022-04-16:
Forget. I was living under impression it measures AC current, while it does only DC :facepalm:

closed 2022-04-16 by bbiegun