volkszaehler/mbmd #252: Sinotimer DDS024MR smart meter support?

Macrisu, 2022-04-26:
I've found this Chinese producer: **Sinotimer**, with RS485 counter enabled devices:
DDS6619-017, DDS6619-016, DDS578MR, DDS578R , **DDS024MR**

For 3 phase 400V, I like [DDS024MR](https://www.aliexpress.com/item/33024113617.html)

Modbus address manual
[MODBUS.Address_EN.pdf](https://github.com/volkszaehler/mbmd/files/8560476/MODBUS.Address_EN.pdf)

Maybe useful links
https://github.com/arendst/Tasmota/issues/6325
https://gist.github.com/groupsky/d8379dafd1b202de787e28ee9b5770dc

andig, 2022-04-29:
Happy to take a PR

andig, 2026-09-30:
Sorry for letting this sit for so long without a resolution. Nobody has picked up the implementation in the meantime and there are no plans to work on it from the maintainer side, so I am closing this to keep the tracker honest. If the device is still relevant to you, a PR adding it (the existing meters/rs485/*.go files serve as templates) is very welcome and we will happily reopen.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig