volkszaehler/mbmd #255: Feature request: JSON support

RichieB2B, 2022-04-28:
`mbmd` is really cool and almost exactly what I was looking for. Currently it is publishing each measurement in it's own MQTT topic like this:
```
mbmd/orno3p1-1/ReactivePower 7.000
mbmd/orno3p1-1/ApparentPower 8.000
mbmd/orno3p1-1/Voltage/L1 234.500
mbmd/orno3p1-1/Frequency 50.000
mbmd/orno3p1-1/Current/L1 0.000
mbmd/orno3p1-1/Cosphi/L1 0.000
mbmd/orno3p1-1/Voltage/L2 233.200
mbmd/orno3p1-1/Current/L2 0.000
mbmd/orno3p1-1/Cosphi/L2 0.000
mbmd/orno3p1-1/Current/L3 0.040
mbmd/orno3p1-1/Voltage/L3 235.400
mbmd/orno3p1-1/Cosphi 0.420
mbmd/orno3p1-1/Cosphi/L3 0.440
mbmd/orno3p1-1/Sum/L1 164.530
mbmd/orno3p1-1/Sum 495.800
mbmd/orno3p1-1/Sum/L3 166.930
mbmd/orno3p1-1/Sum/L2 164.340
etc
```

It would be great if `mbmd` has an option to publish all measurements in a JSON message in a single topic like this:
```
(newlines added for readability)
mbmd/orno3p1-1 {
  "Power": {
    "Reactive": 7,
    "Apparent": 8
  },
  "Voltage": {
    "L1": 234.5,
    "L2": 233.2,
    "L3": 235.4
  },
  "Frequency": 50,
  "Current": {
    "L1": 0,
    "L2": 0,
    "L3": 0.04
  },
  "Cosphi": {
    "L1": 0,
    "L2": 0,
    "L3": 0.44,
    "Total": 0.42
  },
  "Sum": {
    "L1": 164.53,
    "L2": 164.34,
    "L3": 166.93,
    "Total": 495.8
  }
}
```
This matches more with other uses of MQTT that I have seen. Also, it would mean tools like [mqtt2prometheus](https://github.com/hikhvar/mqtt2prometheus) would also work with the MQTT output.

andig, 2022-04-28:
Happy to take a PR. Should either be an option or just published additionally.

RichieB2B, 2022-04-28:
I would if I had any experience with writing Go which I unfortunately don't. The required code is quite similar to [allDevicesHandler() in http.go](https://github.com/volkszaehler/mbmd/blob/master/server/http.go#L59) which probably could be called from [Run() in mqtt.go](https://github.com/volkszaehler/mbmd/blob/master/server/mqtt.go#L128)

andig, 2026-09-30:
Sorry for letting this sit for so long without a resolution. The request is valid, but nobody has picked it up in the meantime and there are no plans to implement it from the maintainer side, so I am closing this to keep the tracker honest. If it is still relevant, a PR is very welcome and we will happily reopen.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig