volkszaehler/mbmd #273: ORNO WE-516 not working

mkmt, 2022-08-31:
I have ORNO WE-516 configured on address 1.
I also have two ORNO 504 configured as 2 and 3.

No matter how I connect them (just one, or two or three) WE-516 doesn't work.
Same RS485 adapter connected to windows computer and using orno software works on all devices, so hardware is OK, connection is OK.

mbmd can't communicate with WE-516.
Here is command I'am using:

mbmd run -a /dev/tty.usb-1.3 -d ORNO3P:1,ORNO1P504:2,ORNO1P504:3

here is output:

2022/08/31 04:06:46 device ORNO3P1.1 did not respond (1/3): read failed: serial: timeout
2022/08/31 04:06:47 device ORNO3P1.1 did not respond (2/3): read failed: serial: timeout
2022/08/31 04:06:49 device ORNO3P1.1 did not respond (3/3): read failed: serial: timeout

What can I do to make it work?

andig, 2022-08-31:
Maybe @bilbolodz or @ochykysh would know?

ochykysh, 2022-08-31:
Looks like you are missing the "--comset 8E1" part.

bilbolodz, 2022-08-31:
Sorry no idea

mkmt, 2022-08-31:
@ochykysh meaning I should compile with this option?

mkmt, 2022-08-31:
@ochykysh with this option ORNO3P works fine but ORNO1P504 doesn't work.
Any way to make both run at the same time?

ochykysh, 2022-08-31:
It depends how your meters are configured. You can set the speed and mode with the windows software. Check that all of them use the same settings(with different IDs). Mbmd only supports setting speed and mode on the port level, not per device.

mkmt, 2022-08-31:
@ochykysh can set speed not the parity in windows.

mkmt, 2022-08-31:
Was able to make 516 working almost.
On web interface all details are all right, on mqtt on the other hand it looks like that:

mbmd/orno3p1-1/ReactiveImport/L3 0.060
homie/orno3p1-1/meter/cosphi/$name Cosphi
homie/orno3p1-1/meter/cosphi/$unit (null)
homie/orno3p1-1/meter/reactiveimport 35.800
homie/orno3p1-1/meter/cosphi/$datatype float
homie/orno3p1-1/meter/cosphil1/$name L1 Cosphi
homie/orno3p1-1/meter/cosphil1/$unit (null)
homie/orno3p1-1/meter/cosphil1/$datatype float
homie/orno3p1-1/meter/cosphil2/$name L2 Cosphi
homie/orno3p1-1/meter/cosphil2/$unit (null)
homie/orno3p1-1/meter/cosphil2/$datatype float
homie/orno3p1-1/meter/cosphil3/$name L3 Cosphi
homie/orno3p1-1/meter/cosphil3/$unit (null)
homie/orno3p1-1/meter/cosphil3/$datatype float
homie/orno3p1-1/meter/currentl1/$name L1 Current
homie/orno3p1-1/meter/currentl1/$unit A
homie/orno3p1-1/meter/currentl1/$datatype float
homie/orno3p1-1/meter/currentl2/$name L2 Current
homie/orno3p1-1/meter/currentl2/$unit A
homie/orno3p1-1/meter/currentl2/$datatype float
homie/orno3p1-1/meter/currentl3/$name L3 Current
homie/orno3p1-1/meter/currentl3/$unit A
homie/orno3p1-1/meter/currentl3/$datatype float
homie/orno3p1-1/meter/export/$name Total Export
homie/orno3p1-1/meter/export/$unit kWh
homie/orno3p1-1/meter/export/$datatype float
homie/orno3p1-1/meter/exportl1/$name L1 Export
homie/orno3p1-1/meter/exportl1/$unit kWh
homie/orno3p1-1/meter/exportl1/$datatype float
homie/orno3p1-1/meter/exportl2/$name L2 Export
homie/orno3p1-1/meter/exportl2/$unit kWh
homie/orno3p1-1/meter/exportl2/$datatype float
homie/orno3p1-1/meter/exportl3/$name L3 Export
homie/orno3p1-1/meter/exportl3/$unit kWh
homie/orno3p1-1/meter/exportl3/$datatype float
homie/orno3p1-1/meter/frequency/$name Frequency
homie/orno3p1-1/meter/frequency/$unit Hz
homie/orno3p1-1/meter/frequency/$datatype float
homie/orno3p1-1/meter/import/$name Total Import
homie/orno3p1-1/meter/import/$unit kWh
homie/orno3p1-1/meter/import/$datatype float
homie/orno3p1-1/meter/importl1/$name L1 Import
homie/orno3p1-1/meter/importl1/$unit kWh
homie/orno3p1-1/meter/importl1/$datatype float
homie/orno3p1-1/meter/importl2/$name L2 Import
homie/orno3p1-1/meter/importl2/$unit kWh
homie/orno3p1-1/meter/importl2/$datatype float
homie/orno3p1-1/meter/importl3/$name L3 Import
homie/orno3p1-1/meter/importl3/$unit kWh
homie/orno3p1-1/meter/importl3/$datatype float
homie/orno3p1-1/meter/reactiveimport/$name Reactive Import
homie/orno3p1-1/meter/reactiveimport/$unit kvarh

ochykysh, 2022-09-25:
@mkmt yes you can, see this - https://kb.plugable.com/serial-adapter/how-to-change-the-com-port-for-a-usb-serial-adapter-on-windows-7,-8,-81,-and-10

ochykysh, 2022-09-25:
@mkmt just connect them to two different USB-RS485 adapters, so you will have different port settings. Or replace the WE504 with a WE514

andig, 2023-01-31:
Seems this might have been resolved, closing.

closed 2023-01-31 by andig